chrisl8/ArloBot #5: Error when running StartRobot.sh

remarvel, 2015-09-25:
Hi Chis,

First, I can't thank you enough for putting all of this together! You've given all of us a huge jumpstart on creating our own arlobots.

I'm getting an error "Error: Cannot find module 'roslib'" when I try to run the StartRobot.sh script.
[image]

I am running Ubuntu 14.04 and I just did a fresh install a couple of days ago. I used your install script to install ROS and your packages.

Any idea what I'm doing wrong?

chrisl8, 2015-09-30:
My first guess is some sort of problem with your environment.
Is the line 'source ~/catkin_ws/devel/setup.bash' in your ~/.bashrc file? It should have been put there by the setup.
What happens when you run 'roscd roslib'?

remarvel, 2015-10-03:
I apologize in advance if I'm missing something obvious. I should note that I  am new to Linux and ROS so this issue is probably something I am doing wrong.

I decided to just start over with a fresh install of Lubuntu then I ran your script to install everything again. 

When I double-click on the ArloBot icon on the desktop it brings up a terminal window and then brings up Firefox and I get an error in Firefox "unable to connect"

Then when I just open a new terminal and run the StartRobot.sh I am still getting the same error.

I'm sure that you're right about it being my environment, but being new to both linux and ROS I'm not sure what I'm doing wrong.

Here's a screenshot of what I get when I run StartRobot.sh and then roscd roslib:

[image]

Thanks again for not only taking the time to put this together, but also to help newbs like me!

chrisl8, 2015-10-04:
I'm glad that you reinstalled, because I fixed a lot of issues. I see I still have an issue though.

Would it be possible for you to run the install script again,
```
bash <(wget -qO- --no-cache https://raw.githubusercontent.com/chrisl8/ArloBot/master/setup.sh)
```
and send me all of the output?
You should be able to copy and paste it from the terminal session. Just use your mouse to select it, and right click to get the menu to "Copy"

I would like to see what the output is to see if it shows why it isn't installing all of the packages.

After you do that, what might fix the issue is to run these commands:
```
cd ~/catkin_ws/src/Metatron/node/
npm install
```
If you do run those commands, please send me the output if there is any,

remarvel, 2015-10-04:
Ok, here is the output from running the install script again:

harvy@HarvyBot:~$ bash <(wget -qO- --no-cache https://raw.githubusercontent.com/chrisl8/ArloBot/master/setup.sh)

SETTING UP ROBOT OPERATING SYSTEM FOR YOUR ARLOBOT!
---------------------------------------------------
You will be asked for your password for running commands as root!

[Checking the Ubuntu version]
Ubuntu trusty found

[Updating & upgrading all existing Ubuntu packages]
[sudo] password for harvy: 
Ign http://security.ubuntu.com trusty-security InRelease
Ign http://us.archive.ubuntu.com trusty InRelease                              
Hit http://security.ubuntu.com trusty-security Release.gpg                     
Hit http://packages.ros.org trusty InRelease                                   
Ign http://us.archive.ubuntu.com trusty-updates InRelease                      
Hit http://security.ubuntu.com trusty-security Release                         
Ign http://us.archive.ubuntu.com trusty-backports InRelease                    
Ign http://ppa.launchpad.net trusty InRelease                                  
Hit http://us.archive.ubuntu.com trusty Release.gpg                     
Ign http://extras.ubuntu.com trusty InRelease                                  
Get:1 http://us.archive.ubuntu.com trusty-updates Release.gpg [933 B]          
Hit http://packages.ros.org trusty/main amd64 Packages                         
Hit http://security.ubuntu.com trusty-security/main Sources                    
Hit http://us.archive.ubuntu.com trusty-backports Release.gpg                  
Hit http://ppa.launchpad.net trusty Release.gpg                                
Hit http://packages.ros.org trusty/main i386 Packages                          
Hit http://security.ubuntu.com trusty-security/restricted Sources              
Hit http://extras.ubuntu.com trusty Release.gpg                                
Hit http://us.archive.ubuntu.com trusty Release                                
Hit http://security.ubuntu.com trusty-security/universe Sources                
Get:2 http://us.archive.ubuntu.com trusty-updates Release [63.5 kB]            
Hit http://security.ubuntu.com trusty-security/multiverse Sources              
Hit http://ppa.launchpad.net trusty Release                                    
Hit http://extras.ubuntu.com trusty Release                                    
Hit http://security.ubuntu.com trusty-security/main amd64 Packages             
Hit http://us.archive.ubuntu.com trusty-backports Release                      
Hit http://security.ubuntu.com trusty-security/restricted amd64 Packages       
Hit http://us.archive.ubuntu.com trusty/main Sources                           
Hit http://security.ubuntu.com trusty-security/universe amd64 Packages         
Hit http://ppa.launchpad.net trusty/main amd64 Packages                        
Hit http://security.ubuntu.com trusty-security/multiverse amd64 Packages       
Hit http://us.archive.ubuntu.com trusty/restricted Sources                     
Hit http://extras.ubuntu.com trusty/main Sources                               
Hit http://security.ubuntu.com trusty-security/main i386 Packages              
Hit http://us.archive.ubuntu.com trusty/universe Sources                       
Hit http://security.ubuntu.com trusty-security/restricted i386 Packages        
Hit http://us.archive.ubuntu.com trusty/multiverse Sources                     
Hit http://ppa.launchpad.net trusty/main i386 Packages                         
Hit http://extras.ubuntu.com trusty/main amd64 Packages                        
Hit http://security.ubuntu.com trusty-security/universe i386 Packages          
Hit http://us.archive.ubuntu.com trusty/main amd64 Packages                    
Ign http://packages.ros.org trusty/main Translation-en_US                      
Hit http://security.ubuntu.com trusty-security/multiverse i386 Packages        
Hit http://us.archive.ubuntu.com trusty/restricted amd64 Packages              
Hit http://us.archive.ubuntu.com trusty/universe amd64 Packages                
Hit http://security.ubuntu.com trusty-security/main Translation-en             
Ign http://packages.ros.org trusty/main Translation-en                         
Hit http://ppa.launchpad.net trusty/main Translation-en                        
Hit http://extras.ubuntu.com trusty/main i386 Packages                         
Hit http://security.ubuntu.com trusty-security/multiverse Translation-en       
Hit http://us.archive.ubuntu.com trusty/multiverse amd64 Packages              
Hit http://us.archive.ubuntu.com trusty/main i386 Packages                     
Hit http://security.ubuntu.com trusty-security/restricted Translation-en       
Hit http://us.archive.ubuntu.com trusty/restricted i386 Packages               
Hit http://security.ubuntu.com trusty-security/universe Translation-en         
Hit http://us.archive.ubuntu.com trusty/universe i386 Packages                 
Hit http://us.archive.ubuntu.com trusty/multiverse i386 Packages    
[... 507 lines cut ...]
npm ERR! 
npm ERR! Failed at the canvas@1.2.9 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the canvas package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls canvas
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
[... cut ...]
Already up-to-date.

Anytime you want to update ArloBot code from the web you can run this same script again. It will pull down and compile new code without wiping out custom configs in ~/.arlarbot. I run this script myself almost every day.

-----------------------------------
ALL DONE! REBOOT AND START TESTING!
I have a list of tests here: cat /home/harvy/catkin_ws/src/ArloBot/manualTests.txt
Look at README.md for testing ideas.
See here for your next step: http://ekpyroticfrood.net/?p=165

harvy@HarvyBot:~$

remarvel, 2015-10-04:
I also tried installing npm with your commands and it said to use 'sudo apt-get install npm' so I did that as well. Here's the output from running the commands you gave me:

[image]

chrisl8, 2015-10-04:
Thank you for the output. For some reason the roslib package isn't building. I will have to erase my install and try it to see what is wrong.

Let me work on the install script a bit and see if I can find out why it is failing to install roslib.

If you want to play with ROS itself you can look at the documentation some more and go into `~/catkin_ws/src/Metatron/scripts/` and run `start-arlobot-only.sh` and/or `start-metatron.sh`

chrisl8, 2015-10-04:
Also, don't install npm with apt-get, that will cause problems.

chrisl8, 2015-10-04:
*I found it!* One missing dependency to make roslib compile. I guess I had it on my system already.
I added it to the install script, so can you try it again?

```
bash <(wget -qO- --no-cache https://raw.githubusercontent.com/chrisl8/ArloBot/master/setup.sh)
```

It should complete without error now.
After it is done, if it has errors, send me the output,
other wise be sure to log out and back in, and then startRobot.sh should work
```
~/catkin_ws/src/Metatron/startRobot.sh
```

remarvel, 2015-10-04:
I think we're getting closer. When I run startRobot.sh now it does bring up the web page, but here is the output in the terminal:

harvy@HarvyBot:~$ ~/catkin_ws/src/Metatron/startRobot.sh
Now using node v0.12.7 (npm v2.11.3)
Go to: http://HarvyBot:8080/
Arlobot > Press Enter to get a prompt.
Run 'help()' for a list of options.
Use Ctrl+d to shut down robot.
Process Error:
{ [Error: listen EADDRINUSE] code: 'EADDRINUSE', errno: 'EADDRINUSE', syscall: 'listen' }
Error: listen EADDRINUSE
    at exports._errnoException (util.js:746:11)
    at Server._listen2 (net.js:1156:14)
    at listen (net.js:1182:10)
    at Server.listen (net.js:1267:5)
    at EventEmitter.listen (/home/harvy/catkin_ws/src/Metatron/node/node_modules/express/lib/application.js:617:24)
    at Object.start (/home/harvy/catkin_ws/src/Metatron/node/webserver.js:194:25)
    at Object.<anonymous> (/home/harvy/catkin_ws/src/Metatron/node/index.js:116:11)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
Main process exit, calling killROS
Running kill_ros.sh . . .
Process Error:
{ [Error: listen EADDRINUSE] code: 'EADDRINUSE', errno: 'EADDRINUSE', syscall: 'listen' }
Error: listen EADDRINUSE
    at exports._errnoException (util.js:746:11)
    at Server._listen2 (net.js:1156:14)
    at listen (net.js:1182:10)
    at Server.listen (net.js:1267:5)
    at Object.<anonymous> (/home/harvy/catkin_ws/src/Metatron/node/index.js:855:4)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
Main process exit, calling killROS
Shutdown:Killing everything, please wait...
Shutdown:Everything Killed.
Shutdown:Turning off all relays
lexicon english_poslex not defined

Shutdown:Error: Need to state which relay
kill_ros.sh closed with code 1
Shutdown complete.
harvy@HarvyBot:~$

chrisl8, 2015-10-04:
This error: { [Error: listen EADDRINUSE] code: 'EADDRINUSE', errno: 'EADDRINUSE', syscall: 'listen' }
Indicates that the Web port is already in use.
You said the site works though? 
My first guess is that it is somehow running twice.
I hate to say this, but did you try rebooting and running it again?

remarvel, 2015-10-04:
You're right. Not sure how that happened, but after rebooting it works! This is really amazing!! Thank you!! I only have one more question (for now), how do I get the teleop web page to work? I can bring up ROS on the main page, but when I go to the remote control page it is not connected? Here are some screenshots. I'm sure it's something obvious that I'm missing.

[image]

[image]

chrisl8, 2015-10-04:
Let me know if the issue with it running twice keeps happening. I've never had it happen, so I haven't built any code check for it.

Regarding the remote control page, I think the page isn't finishing loading because of there being no relay board. Someone else had this issue also and I need to fix it.
If you can, press F12 and see if it shows any errors. You could post them here.

If you want to try fixing it open up `~/catkin_ws/src/Metatron/node/public/remote-control.js` and see if you can work out what parts need to be eliminated or changed, although the final solution probably needs to affect more files.
In another issue chronoglass said that he:
"I've commented out the references to the device in 
node/public/remote-control.js
lines 1162 -1166
lines 1205 - 1209"
I think, though, that that might completely break the page so that it doesn't work at all, instead of just ignoring the relay.
So focus on any lines that talk about subscribing to the "relay" services.

When you make changes to ` ~/catkin_ws/src/Metatron/node/public/remote-control.js` you don't have to restart the entire program, just refresh the remote control page.
Also, you can use that page from another computer in the house.

I will try to solve this later this week, but I doubt that I will get to it today.

remarvel, 2015-10-04:
Thanks a lot Chris! I haven't had any more trouble with it starting up twice. I've just about got the remote control page to come up without any errors by commenting out most anything to do with the relays. Still can't get the teleop to work, but I think I have a different problem there because I can't get the keyboard teleop to work either. I'll keep plugging away at it.

Thanks again for all the help and patience!!

chrisl8, 2015-10-05:
You should be able to run this in one terminal window:
`~/catkin_ws/src/Metatron/scripts/start-arlobot-only.sh`
and then this in another:
`~/catkin_ws/src/Metatron/scripts/keyboard-teleop.sh`
and drive the robot around.

Until that works, I wouldn' try anything else. :)

chrisl8, 2015-10-06:
I fixed the system so that the web remote control should work without modification even if you have no USB Relay board.

If you have made changes to your files at all it will probably work best to just delete the entire `~/catkin_ws` folder and run the setup again if you want to start fresh and see if it works.
Otherwise you have to know how to use git to overwrite your changes.

If you have a chance to test this, please let me know, otherwise I'll close this in a day or two.

Thanks!

remarvel, 2015-10-06:
That did the trick! The remote control page comes up without errors and no need to modify. I deleted my catkin_ws folder before installing to make sure.

Still haven't gotten the teleop working, but I'm pretty sure that's an issue on my end.

Thanks again for all the help and taking the time to put all of this together to share with others!!

closed 2015-10-06 by remarvel