chrisl8/ArloBot #6: nodejs package problems (NaN, contextify, ngrok, log.io)

chronoglass, 2015-10-01:
a number of the nodejs packages seem to have gotten themselves broken and require the latest NaN.js.. fixed by going into the node folder ~/catkin_ws/src/Metatron/node/
nvm stable
npm install nan
then npm install

Also found I had to find every instance of contextify mentioned (even in the node modules folders of the node modules installed) and change to the latest version (for log.io I had to actually install contextify globally)

ngrok needed to be updated to 0.2.2 to allow it to build as well as to get log.io working (i think.. it was really late, and my notes got a bit.. er, scrawley there at the end)

figured I'd share where I am so far in case anyone else runs into it. Knowing how nodejs packages work.. they probably will.

current issue seems to be after I get the startRobot.sh to go, I start ros (the running log states ROS successfully started, but the upper left indicator sticks at "startROS starting up . . .") remote control screen states it's not connected, and the log streamer keeps cycling that it is resetting a relay and waiting for the PAB to "settle down" though I has no relay. heh, will post another issue if/when I track that one down.

thanks by the way, for sharing all of your work on this so far, I'd still be muddling around the ROS tutorials deciding if I wanted to start getting a bot together if I hadn't ran across this.

chrisl8, 2015-10-02:
Thanks. Could this be because of using node 4.0?
I just changed the code today to use 0.12.7 instead of "stable" because I found it doesn't work with 4.0, but if this can make 4.0 work then I'll make the changes and update.

chronoglass, 2015-10-02:
using 4.1.1 according to nvm, really, this is just one of the fun quirks of using nodejs and npm i've found. it's like linux 10 years ago. haha

chrisl8, 2015-10-02:
Good to know. My guess is that this is an issue with the node version.
If you run the setup again now it will install node 0.12.7 and use it.
The entire idea of npm and nvm is to be able to develop in and deploy a consistent version so you can be confident things will work. However I broke that by having it use "stable" instead of a specific version.

That said, with your notes here I can probably get it up to 4.1.1 :+1:

chrisl8, 2015-10-02:
I pulled ngrok out. I'm not using it anymore. It was only there for use with Twilio for incoming SMS messages to the robot, but it requires a paid account now to be stable, so I'm not going to use it anymore. It never worked "out of the box" anyway, so I'll leave Twilio to a blog post later if other people want to try it.
That fixes the ngrok issue. Next for log.io

chronoglass, 2015-10-02:
i'll give it shot in an hour or so

chronoglass, 2015-10-02:
saw no errors rerunning the install script this time. almost everything fired up!
log streamer won't start (no error, just click and nothing, will investigate)
new error after turning on ROS run (error:parameter [/movebase] is not set)
and I'll have to re-comment the toggle relay stuff
and a weird bit of stuff coming up in the status window of the main page.. will track down and open a new issue for anything I find.. i think this bugger is done!

chrisl8, 2015-10-02:
Thanks, I made some updates based on what you told me here, so this was very helpful. I'll close this down now.
Please do open new issues with anything else you find.

closed 2015-10-02 by chrisl8