chrisl8/ArloBot #7: usb_relay

chronoglass, 2015-10-02:
seems that the config files I can see that are asking about it all say false, but the code is still trying to fire the thing up and hanging.

I've commented out the references to the device in 
node/public/remote-control.js
lines 1162 -1166
lines 1205 - 1209

that seems to make the web server side at least happy enough to claim it has a connection to ROS (it's still arguing with me about actually sending any commands or the like, baby steps!)

chronoglass, 2015-10-02:
really can't decide if it's related or not but this is what my log.io looks like
just this, over and over

arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,402: Serial Data Gateway starting . . .
rosout application_server 1443761617.491101980 INFO [propellerbot_node.py:709(startSerialPort) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] Serial Data Gateway started.
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,490: Serial Data Gateway started.
rosout application_server 1443761617.509819030 INFO [propellerbot_node.py:874(watchDog) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] Watchdog Timeout Reset initiated
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,509: Watchdog Timeout Reset initiated
rosout application_server 1443761617.512759923 INFO [propellerbot_node.py:257(_reset_serial_connection) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] Serial Data Gateway stopping . . .
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,512: Serial Data Gateway stopping . . .
rosout application_server 1443761617.518564939 INFO [SerialDataGateway.py:44(Stop) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] Stopping serial gateway
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,518: Stopping serial gateway
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,631: Serial Data Gateway stopped.
arlobot-6 application_server [
arlobot-6 application_server [rosout][INFO] 2015-10-01 21:53:37,632: 5 second pause to let Activity Board settle after serial port reset . . .
rosout application_server 1443761617.632139921 INFO [propellerbot_node.py:262(_reset_serial_connection) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] Serial Data Gateway stopped.
rosout application_server 1443761617.632736921 INFO [propellerbot_node.py:263(_reset_serial_connection) [topics: /arlobot_safety/safetyStatus, /arlobot/pirState, /serial, /rosout, /infrared_scan, /tf, /odom, /arlo_status, /ultrasonic_scan, /mobile_base/commands/velocity] 5 second pause to let Activity Board settle after serial port reset . . .

chrisl8, 2015-10-02:
"Watchdog Timeout Reset initiated" means that the serial communication is not working. It is timing out waiting for data form the Propeller board and then trying again.
So yes, the kernel issue would cause this, but if you went to -63 it isn't the Kernel.

Did you use SimpleIDE to install the code on the Propeller board?
In ~/catkin_ws/src/Metatron/scripts/ there is a script called direct2PropSerialTest.sh that you can use to test communication with your Propeller board, make sure it works.
There is also a file in the same location called direct2PropSerialTestLines.txt with some test lines to run. You don't need to run all of them, in fact, if you just get the "i     0" lines back from the script then it should all work, but the lines help with future testing.

chronoglass, 2015-10-02:
direct2PropSerialTest.sh has been my current mountain, it returns the "i 0" until i try a test line, then sometimes it returns garbage, other times it just hangs until the world freezes over. Interestingly enough, if I restart the script, it sends data still "i 0" nothing new sadly

I've narrowed it down to the point where in your code it hits the "drive_speed(0, 0)" function for initing the motors.

I've verified by running my own code that should have made the bot run me over that as soon as I call drive_speed(0, 0).. i get script hangage.. here's the part i'm not getting.. I CAN control the wheels via the calibration script.. sooo, yeah. haha going to start poking around the parallax forums as well.

chrisl8, 2015-10-02:
Before you built and installed the code on the Propeller board did you edit the settings in ~/.arlobot/per_robot_settings_for_propeller_c_code.h and did it use them?
If those settings are wrong it might work right up until you initialize it with some text, and then it will try to access some hardware you don't have and hang.

Did you try sending text to it inside of SimpleIDE? I think that it uses a completely different serial interface driver, because it worked even with the broken kernel.

You might look here at the full duplex serial code: http://learn.parallax.com/propeller-c-simple-protocols/full-duplex-serial

Finally, if you are sure it is using the settings from ~/.arlobot/per_robot_settings_for_propeller_c_code.h and that they are right, just get into SimpleIDE and start tearing apart the code and running it in the local terminal inside of SimpleIDE,
The first thing I would do is just change the code so that when you send it something it just replies with some generic text like "Hello world" and loops back to the "i 0" part, just to prove that you can receive data from it.
Then you can just kind of work down the code, stopping it at different places with a line from the code to you and watch it in the SimpleIDE terminal. You can even just put an output line after every line of code and eventually you will find what line is causing it to stall.

chrisl8, 2015-10-02:
P.S. You can even just change it so that it "initializes" itself in the code, for testing, so you don't have to send it a line. I think there is even a commented out line in the code to do that for when I need to test things.

chronoglass, 2015-10-02:
I've been poking around in there, that's how I've narrowed it down to the moment it calls drive_speed(0, 0); the first time.. and verified that by trying to call it myself on it's own (set it to run me over and kill me mode, haha) and got no joy.. you'd think given the media, robots would be more willing to turn on and kill their masters given the specific order to do so...

I've got a post up on parallax asking about that function specifically.. and might go poking in the library to see why the calibration makes the wheels go, but the actual abdrive library decides it wants nothing to do with me. it must be calling the motors differently, or doing something sketchy before it does.

if I offload the init to a cog, I can get past it.. but i'm having doubts that the cog is actually doing anything other than hanging and letting the program continue while it does so.

chrisl8, 2015-10-02:
Wow, you are really figuring this out! Good work!
It is almost like maybe the Arlodrive library is not working. Someone else told me it was broken, but I didn't get enough information from them to follow up.

What version of SimpleIDE are you using? I've noticed that the latest version didn't work for me. I think I found that only version 0.9.66 works for me, which I referenced in the blog article.

If we can narrow it down and prove that the Arlodrive library is broken maybe we can get Parallax to update it. I might even give them a call if we have a clear way to reproduce the failure.

Questions:
What version of SimpleIDE are you using?
Where did you download the ArloDrive library from?

chronoglass, 2015-10-02:
answers:
arlo library, from you as well as from their post (tried both)
simpleide 1.0.1 rc (linux)
1.0.2 rc (osx)

chrisl8, 2015-10-02:
Try it with SimpleIDE 0.9.66 You might have to Google to find the .deb package file for it.

chronoglass, 2015-10-02:
holy crackerjacks.. i got response.. for a moment.. it stopped pretty quickly, but progress!
for future googlers

https://d9d46cb6fc558ba1db5c3aa51f1eb3a56e713404.googledrive.com/host/0B8ruEl5BL0dfZzZfdHRiX2pYNm8/

is where i obtained it (might download em all and mirror on my own google drive)

chrisl8, 2015-10-02:
I had notes in my blog post (http://ekpyroticfrood.net/?p=165) about which version of SimpleIDE, but I just updated it with some red text and more links, so hopefully it will be more obvious.
I'll add notes to the github Arlobot readme too so everyone can see.

Meanwhile maybe we can get Parallax to fix it in your forum thread.

Regarding SimpleIDE, make sure to fully remove the broken version, including the folders it puts in your home directory. I think even if you downgrade SimpleIDE the "learn" folder form the new version may remain and break arlodrive.h

Starting fresh never hurts, just keep track of any changes you made that fixed things. :)

chronoglass, 2015-10-02:
so yeah.. it was unrelated to the usbrelay.. want me to separate into two issues "new" and link to each?

chrisl8, 2015-10-02:
Regarding the USB relay,

Commenting out lines 1162 -1166 should help, but
if you comment out lines 1205 - 1209 I THINK you are just basically breaking the ROS connection completely. So that will hide lots of errors.

But yes, if you want to create new issues that would be fine. Nobody is going to yell at us if we use issues wrong. That is one benefit of all of this, we get to experiment and learn. :)

chrisl8, 2015-10-02:
Ok, I changed the name of the issue and now I'll close it, here is the fix for now, which I have added to my blog post and to the Arlobot Readme.MD file:

### DO NOT USE THE LATEST VERSION OF SimpleIDE! ###
I think Parallax should fix this, but for now, the current version of SimpleIDE does NOT work with Parallax’s own Arlodrive library.
Instead use version 0.9.66, which you can find here:
[https://d9d46cb6fc558ba1db5c3aa51f1eb3a56e713404.googledrive.com/host/0B8ruEl5BL0dfZzZfdHRiX2pYNm8/](https://d9d46cb6fc558ba1db5c3aa51f1eb3a56e713404.googledrive.com/host/0B8ruEl5BL0dfZzZfdHRiX2pYNm8/)

Or just run this to grab and install the working version:
```
wget https://googledrive.com/host/0B8ruEl5BL0dfZzZfdHRiX2pYNm8/simpleide-0.9.66-amd64-debian-linux.deb
sudo dpkg -i simpleide-0.9.66-amd64-debian-linux.deb
```

We also have this Parallax forum thread open asking them to help us fix the latest version, so check it out for updates: http://forums.parallax.com/discussion/162351/arlo-hardware-activity-bot-software-drive-speed-0-0-ubuntu-and-no-happy-sunset

closed 2015-10-02 by chrisl8