chrisl8/ArloBot #8: kinect with arlobot

chronoglass, 2015-10-16:
So i've tracked down three things thus far that need to happen in the setup script to allow kinect functionality to work.

#1 it seems the tlp tlp-rdw addition knocks it offline
#2 the unstable branch of https://github.com/avin2/SensorKinect needs to be installed
#3 the env variable TURTLEBOT_3D_SENSOR needs to be set to kinect

im looking at making an add to the setup script to do an lsusb and grep out the entries for the sensors to decide how to proceed, i've got a kinect4w and a regular kinect so i can find those, but no primesense. I can send the entries to you, or you wanna send me what lsusb looks like on your end?

k4win (which near as I can tell won't work) looks like:
Bus 002 Device 005: ID 045e:02bf Microsoft Corp. 
Bus 002 Device 004: ID 045e:02be Microsoft Corp. 
Bus 002 Device 003: ID 045e:02c2 Microsoft Corp.

(have to put my 360 one back together, then i'll get you what that one looks like, dropped it and broke the support neck.. woops)

chronoglass, 2015-10-16:
i take it back, the k4w appears to work!
it looks like you should add this line to /etc/udev/rules.d/55-primesense-usb.rules (after installing the driver from avin2)
SUBSYSTEM=="usb", ATTR{idProduct}=="02c2", ATTR{idVendor}=="045e", MODE:="0666", OWNER:="root", GROUP:="video"

chrisl8, 2015-10-16:
This is the lsusb output for the ASUS Xtion:
`Bus 002 Device 023: ID 1d27:0601 ASUS`

If we can sort it out so that either can work and the startup sequence is automatic based on a setting in the `~/.arlobot/personalDataForBehavior.json` file that would be perfect.

The tlp tlp-rdw addition was purely for the purpose of conserving laptop battery power. If you are sure that the Kinect works without it, but not with it, I'll remove it from the setup.

Adding another repository is easy if it can just go in with the others. Even if everyone doesn't use it, it is fine to just have everyone download it.

When does the env variable have to be set? I think it gets set now in some of the launch scripts, so we might need different launch scripts.
If the existing launch scripts work can work with nothing more than setting TURTLEBOT_3D_SENSOR to "kinect", that could be done in `start-arlobot-only.sh`  and `start-metatron.sh`.

chronoglass, 2015-10-16:
i've got it set now as my bashrc (though i havn't quite gotten as far as clicking the arlobot icon quote yet, still in CLI land)
it just needs to be set before any of the turtlebot stuff fires up

i havn't yet researched WHY the two packages cause the kinect to stop, but after they are installed (on the first reboot) kinect goes down to a 1 item listing in lsusb (but you do get a new usb hub, instead of the mics and camera being attached to them)
uninstall em, reboot and the kinect is back and happy.

seems the kinect may also require a calibration step.. which ain't exactly fun.. though right now I'm at the point of trying to cut up the power cord and cross my fingers to run it off the 12v line.. hahaha

chrisl8, 2015-10-16:
Good enough for me. 1. I removed tlp tlp-rdw from the setup script.
Now on to #2 and #3 :smile:

chrisl8, 2016-12-17:
It has been over a year. If you want to reopen this and keep working on it I'm happy to. Otherwise I'll assume we've moved on.

closed 2016-12-17 by chrisl8