chrisl8/ArloBot #11: Unexpected behaviour, when putting the C code on the Parallax board.

cristian-frincu, 2016-05-27:
Robot would start moving in circles, all by itself, with no command being sent.
The readings from the PING)) sensors are not updating.
- I can see the sensors are working and readings are being taken because they flash the LED
- They work and are wired correctly because with my own code they work
- I am assuming it is a problem with the setup file I am supposed to comment/uncomment
- I have 4 ping sensors, 3 in the front 1 in the back. plugged in the pins 8(back),9(left),10(center),11(right)   , I have configured this in the  file, but it does not seem to work.

The haltdistance and slowing down distance I have set both to {10,10,10,10}
Is this the right way to fill the array, even though my ping)) are connected to pins 8-11, or would I need to fill it: 
{0,0,0,0,0,0,0,0,10,10,10,10}

??

Any help would be welcomed

chrisl8, 2016-05-31:
See if this works for you. Compare it to the original to see what I changed and tweak it more as you need to:
https://gist.github.com/chrisl8/108c4037c49269f57396444c40097576

cristian-frincu, 2016-05-31:
Thank you, that has solved my problem!
Thank you for the great program you have put together

closed 2016-05-31 by cristian-frincu