chrisl8/ArloBot #12: from HB25 to DHB-10

jonpol01, 2016-11-14:
Hi, I have a working environment on HB25. then, when I change to DHB-10 board my current indigo ROS failed to move my robot.

cbarrett27, 2016-11-14:
I'm getting ready to order my motor controller(s) for my Arlo bot and wanting to know which to purchase.
Anyone have the DHB-10 working?
thanks in advance

chrisl8, 2016-11-15:
@jonpol01 Can you give us some more details? Perhaps some log output from using the directToSerial test scripts?

@cbarrett27 Yes, the DHB-10 is working now with my code. I haven't fully tested it yet, but it is functional. I would recommend getting the DHB-10 as it is the path forward.

jonpol01, 2016-11-16:
@chrisl8 Sorry our robot is in a different environment, but I can definitely say we can move the motors properly through directToSerial test. If there is any part i should look for in the log output can you tell me specifically.
Unresponsive motor only occurs when we run it through arlobot from your teleop script. Though numbers/data (supposedly motor speed prop.) flies back to terminal, the motor is idle.

chrisl8, 2016-11-16:
The output of the command "rostopic echo serial" before, during and after you attempt to move the robot would help.

jonpol01, 2016-11-21:
Hi, im sorry for a late reply
btw here it is, through arlobot_teleop

before
in: o   0.000 0.000 0.000 0.000 0.000 0.000 {"p0":25,"p1":28,"p2":180,"p3":113}
in: o   0.000 0.000 0.000 0.000 0.000 0.000 {"p0":25,"p1":28,"p2":180,"p3":113}
out: s,0.0,0.0

during
in: o   0.000 0.000 0.000 0.000 0.000 0.000 {"p0":25,"p1":28,"p2":180,"p3":113}
in: o   1        1        0       200    39      0      4.69   4.69  0  0
out: s,0.0,0.0

after is somewhat the same with "before"
it seems like command from teleop properly come through. I really looking forward what you can come up with this.

chrisl8, 2016-12-02:
The constant "out: s,0.0,0.0" is telling the Propeller board to shut the motors off and be perfectly still.
Normally when ROS is running you will not get any of these lines. Whatever ROS node is sending the s,0.0,0.0 is keeping the robot still.

You should see positive and negative numbers for motion.

The command:
rostopic echo /cmd_vel_mux/active 
will tell you what ROS node is active and keeping the robot still.

Usually you only get constant 0.0 output if you have the keyboard teleop running.
If a joystick is online it can do this too.

jonpol01, 2016-12-02:
Thank you for replying.

Somehow  I managed a temporary fix to it, by setting "monitorACconnection" to false made the motors roll. 
I am on prototyping stage and the bot is hooked to a dc adapter sitting on a table, and setting it by default to true can give a lot of headache.

again thank you, I am looking forward for more support.

chrisl8, 2016-12-02:
Excellent point! 

I will change the default to false, and review the configuration to see if there are any other defaultsuggestions that may leave a new robot stuck.

closed 2016-12-05 by jonpol01