chrisl8/ArloBot #13: 2D Nav Goal accuracy and speed

jonpol01, 2016-12-05:
hi,

Env:
kinect (done &worked)
saved map

My issue:
After saving a map, I used this script one by one
  on ssh:
   ./start-robot.sh
   ./load-map.sh map
  workstation:
    roslaunch arlobot_rviz_launchers view_navigation.launch --screen

The map loads but the position is not accurate, and when I do a 2D Nav Goal well It moves slow and spins until it gets to his placed goal on rviz window.

looking forward for a response

chrisl8, 2016-12-15:
When you build a map the "0" location is the point where you started to build the map.
When you load the map it will always assume the robot is at that same position.
In RViz you can set the robot's current position using the '2D Pose Estimate' 

I hope I have understood your question properly and that this helps.

chrisl8, 2016-12-17:
I too found some issues with control and mapping accuracy this week while doing some testing.
I've fixed several issues today and pushed the new code up to github.
Please give the new code a try and let me know if it helps.

The changes include new C code for the propeller board, so you will have to run SimpleIDE and update the code on the Propeller board to get the fixes.

chrisl8, 2016-12-19:
FYI: The issue I am seeing now with my DHB-10 code is that the robot always rotates past the target position and then has to rotate back again to correct the over-rotation.

It works, but it isn't ideal, so I'm still tinkering.

It makes working maps though and is able to navigate them.

jonpol01, 2016-12-20:
Hi thank you on your reply,
good job on your work. I am now testing your new code merged with mine and it is working fine as of now.
Though i'll give you a feedback if any means come up.

chrisl8, 2018-03-31:
I spent a lot of time tweaking things, and I think it is working well enough now.

I have also had some input and even pull requests from some others, which I hope improved the reliability of the navigation.

closed 2018-03-31 by chrisl8