chrisl8/ArloBot #15: Joystick control latency and timeout problem

henryclever, 2017-03-05:
Hi Chris,

Hope you are doing well.  I have a couple questions about the Propeller code.

First, there is a delay of about 0.5 - 1.0 seconds when I use the joystick to drive the base.  To see where this may be cropping up, I edited the following line which is around 590 in the ROSInterfaceForArlobotWithDHB-10.c:

//dprint(term, "o\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t", X, Y, Heading, gyroHeading, V, Omega);
dprint(term, "o\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t", X, Y, Heading, CommandedVelocity, V, Omega);

Thus, the velocity from the DHB-10 feedback, V, is compared to the CommandedVelocity at each time step on the ROS interface by adding a rospylog line to one of the scripts.  There is definitely a lag going on there--and I'd like to see what I do to fix it.  

The next issue is that when I am driving forward on my joystick, the timeoutCounter frequently becomes higher than the ROStimeout timer and it throws a clearTwistRequest(), resulting in an inconsistent halting of the wheels for 2-10 seconds.  This does not happen when I am driving backwards, but it does happen when I send rotate commands, where one wheel or the other is driving forward.  I tried commenting out the clearTwistRequest(), but instead of halting the robot becomes uncontrollable for that time period.  I think the problem is occurring in the serial buffer, and perhaps some characters throw more errors than others.  Do you have suggestions on this before I dig really deep on my own?

Thank you !

Henry C.

chrisl8, 2017-03-16:
Glad to see your interest!

Unfortunately I don't have an answer. I am aware that the code using the DHB-10 controller has issues.
The odometry from it seems to be impeccable, but the control input is problematic.
My best guess is that it is either just slow by nature, or that it has some sort of "ramping" built into it which forces it to respond slowly.
That doesn't explain why it seems to still say it is moving when it is stopped.

Your notes actually helped explain more than I knew before.

I keep dabbling with it, but I don't have a lot of time lately to focus on it.
So if you can sort anything out that would be wonderful.
It may be that the Propeller code needs a complete overhaul, or maybe there is just some logic flaw in my setup that is hampering it.

henryclever, 2017-03-21:
Hi Chris,

Thanks for letting me know!! I will keep you posted if I find a solution.

Warmth,
Henry C.

closed 2018-03-31 by chrisl8