chrisl8/ArloBot #16: Catkin_make returning errors

petherman, 2017-08-20:
I am receiving 'could not find a package configuration file provided by "PACKAGE_NAME" with any of the the following names' errors.

I am using Ubuntu 14.04 LTS on VMWare Workstation 12 Pro and ROS Indigo.

I have tried manually installing the packages but eventually i get to a package called "ceres" and can't progress any further.

I have also tried removing and reinstalling ROS Indigo and have not seen any results.

https://snag.gy/H4eEvW.jpg

chrisl8, 2017-08-31:
Sorry, I am at a loss here.

Do you get these errors when you just go through the ROS Install setup here:
http://wiki.ros.org/indigo/Installation/Ubuntu

Also, I'm moving to Kinetic on Ubutu 16.04, which is now the default branch.
Maybe you could try that?

chrisl8, 2018-02-24:
The solution is to add two more packages before running the install or cmake:

```
    sudo apt install ros-kinetic-geodesy 
    sudo apt install libceres-dev
```

I will go ahead and add those two packages to the install now.

closed 2018-02-24 by chrisl8