chrisl8/ArloBot #18: .arlobot hidden folder does not exist

swinericvuong, 2017-08-31:
Hi,
When i try to launch: nic@ubuntu:~/catkin_ws$ roslaunch arlobot_bringup minimal.launch --screen

This error occurs:
`error loading <rosparam> tag: 
	file does not exist [/home/nic/.arlobot/arlobot.yaml]
XML is <rosparam command="load" file="$(env HOME)/.arlobot/arlobot.yaml"/>
The traceback for the exception was written to the log file`

Ive checked the hidden files and .arlobot does not exist. All other folders are created.

chrisl8, 2017-08-31:
Something is wrong with the setup script. :(

Can you try running it again and see if it has any errors about creating the .arlobot folder?

You could try making the folder first and see if that helps:
mkdir ~/.arlobot

swinericvuong, 2017-09-12:
@chrisl8 I installed a fresh copy of Ubuntu 16.04 LTS and ran your quick setup script. After installing and restarting i tried going into cd catkin_ws and running catkin_make. and it gives this error.

https://snag.gy/nczEBv.jpg

I saw this error during one of the steps in installation.

chrisl8, 2018-03-31:
I believe that some dependencies I was relying on got out of date.
I've since removed them and changed my code. It appears to compile correctly now.

closed 2018-03-31 by chrisl8