chrisl8/ArloBot #23: Error at [Grabbing dependencies for node packages]

sypronamp, 2017-10-28:
I get this error when reaching [Grabbing dependencies for node packages] I tested that on 2 laptops with the same error while using quick install method 

```
bash <(wget -qO- --no-cache https://raw.githubusercontent.com/chrisl8/ArloBot/kinetic/setup-kinetic.sh)
```

```
[Initializing the required Node version]
This is the version I'm developing with now, so stick with it.
v8.4.0 is already installed.
Now using node v8.4.0 (npm v5.3.0)
stable -> 8.4.0 (-> v8.4.0)
default -> 8.4.0 (-> v8.4.0)

[Grabbing dependencies for node packages]
npm WARN deprecated jquery@1.8.3: Versions of the jquery npm package older than 1.9.0 are patched versions that don't work in web browsers. Please upgrade to >=1.11.0.
npm WARN deprecated connect@2.7.2: connect 2.x series is deprecated
/home/ros/.nvm/versions/node/v8.4.0/bin/log.io-server -> /home/ros/.nvm/versions/node/v8.4.0/lib/node_modules/log.io/bin/log.io-server
/home/ros/.nvm/versions/node/v8.4.0/bin/log.io-harvester -> /home/ros/.nvm/versions/node/v8.4.0/lib/node_modules/log.io/bin/log.io-harvester

> contextify@0.1.15 install /home/ros/.nvm/versions/node/v8.4.0/lib/node_modules/log.io/node_modules/contextify
> node-gyp rebuild

make: Entering directory '/home/ros/.nvm/versions/node/v8.4.0/lib/node_modules/log.io/node_modules/contextify/build'
  CXX(target) Release/obj.target/contextify/src/contextify.o
../src/contextify.cc: In static member function ‘static v8::Local<v8::Context> ContextWrap::createV8Context(v8::Local<v8::Object>)’:
../src/contextify.cc:131:68: warning: ‘v8::Local<v8::Object> v8::Function::NewInstance() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
         Local<Object> wrapper = Nan::New(constructor)->NewInstance();
                                                                    ^
In file included from /home/ros/.node-gyp/8.4.0/include/node/v8.h:26:0,
                 from /home/ros/.node-gyp/8.4.0/include/node/node.h:63,
                 from ../src/contextify.cc:1:
/home/ros/.node-gyp/8.4.0/include/node/v8.h:3792:52: note: declared here
   V8_DEPRECATED("Use maybe version", Local<Object> NewInstance() const);
                                                    ^
/home/ros/.node-gyp/8.4.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
../src/contextify.cc:150:16: error: ‘class v8::ObjectTemplate’ has no member named ‘SetAccessCheckCallbacks’
         otmpl->SetAccessCheckCallbacks(GlobalPropertyNamedAccessCheck,
                ^
../src/contextify.cc: In static member function ‘static void ContextWrap::GlobalPropertyGetter(v8::Local<v8::String>, const Nan::PropertyCallbackInfo<v8::Value>&)’:
../src/contextify.cc:182:80: warning: ‘v8::Local<v8::Value> v8::Object::GetRealNamedProperty(v8::Local<v8::String>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
         Local<Value> rv = Nan::New(ctx->sandbox)->GetRealNamedProperty(property);
                                                                                ^
In file included from /home/ros/.node-gyp/8.4.0/include/node/v8.h:26:0,
                 from /home/ros/.node-gyp/8.4.0/include/node/node.h:63,
                 from ../src/contextify.cc:1:
/home/ros/.node-gyp/8.4.0/include/node/v8.h:3345:30: note: declared here
                 Local<Value> GetRealNamedProperty(Local<String> key));
                              ^
/home/ros/.node-gyp/8.4.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
../src/contextify.cc: In static member function ‘static void ContextWrap::GlobalPropertyQuery(v8::Local<v8::String>, const Nan::PropertyCallbackInfo<v8::Integer>&)’:
../src/contextify.cc:209:67: warning: ‘v8::Local<v8::Value> v8::Object::GetRealNamedProperty(v8::Local<v8::String>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
         if (!Nan::New(ctx->sandbox)->GetRealNamedProperty(property).IsEmpty() ||
                                                                   ^
In file included from /home/ros/.node-gyp/8.4.0/include/node/v8.h:26:0,
                 from /home/ros/.node-gyp/8.4.0/include/node/node.h:63,
                 from ../src/contextify.cc:1:
/home/ros/.node-gyp/8.4.0/include/node/v8.h:3345:30: note: declared here
                 Local<Value> GetRealNamedProperty(Local<String> key));
                              ^
/home/ros/.node-gyp/8.4.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
../src/contextify.cc:210:71: warning: ‘v8::Local<v8::Value> v8::Object::GetRealNamedProperty(v8::Local<v8::String>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
             !Nan::New(ctx->proxyGlobal)->GetRealNamedProperty(property).IsEmpty()) {
                                                                       ^
[... 57 lines cut ...]

[Setting up Parallax SimpleIDE for putting code on Activity Board.]
--2017-10-28 22:18:30--  https://web.archive.org/web/20161005174013/http://downloads.parallax.com/plx/software/side/101rc1/simple-ide_1-0-1-rc1_amd64.deb
Resolving web.archive.org (web.archive.org)... 207.241.225.186
Connecting to web.archive.org (web.archive.org)|207.241.225.186|:443... connected.
Unable to establish SSL connection.
dpkg: error processing archive /tmp/simple-ide_1-0-1-rc1_amd64.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/simple-ide_1-0-1-rc1_amd64.deb
rm: cannot remove '/tmp/simple-ide_1-0-1-rc1_amd64.deb': No such file or directory

[You must Update your SimpleIDE Learn Folder using the instructions here!]
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/%{ <-- HERE (.*?)}/ at /usr/bin/print line 528.
Error: no such file ":\n\033[0;32mhttp://learn.parallax.com/tutorials/language/propeller-c/propeller-c-set-simpleide/update-your-learn-folder\033[0m\n"

Anytime you want to update ArloBot code from the web you can run this same script again. It will pull down and compile new code without wiping out custom configs in ~/.arlarbot. I run this script myself almost every day.

-----------------------------------
ALL DONE! REBOOT AND START TESTING!
I have a list of tests here: cat /home/ros/catkin_ws/src/ArloBot/manualTests.txt
Look at README.md for testing ideas.
See here for your next step: http://ekpyroticfrood.net/?p=165

```

chrisl8, 2018-03-31:
All of this should be sorted and corrected now.

closed 2018-03-31 by chrisl8

ahmediv, 2018-04-01:
Thank you for your great effort