chrisl8/ArloBot #35: Possible error with serial communication.

geistmate, 2018-06-17:
Hi Chris,

I just followed all your directions to install everything. When I'm trying to get it to start, I'm running into errors. First, when I try the direct2PropSerialTest.sh, it won't send me a stream of i  0; (however it does when I earlier loaded the Simple IDE code into EEPROM). 

I get the return as
```
Activity Board should be sending a stead stream of:
i     0
i     0
etc.
Paste this line into the terminal to get it to start sending sensor data:
For HB-25:
d,0.403000,0.006760,0,0,0,0,0,0.0,0.0,0.0
For DHB-10:
d,0.403000,0.00338,0,0,0,0,0,0.0,0.0,0.0
usage: miniterm.py [-h] [--parity {N,E,O,S,M}] [--rtscts] [--xonxoff]
                   [--rts RTS] [--dtr DTR] [--ask] [-e] [--encoding CODEC]
                   [-f NAME] [--eol {CR,LF,CRLF}] [--raw] [--exit-char NUM]
                   [--menu-char NUM] [-q] [--develop]
                   [port] [baudrate]
miniterm.py: error: argument baudrate: invalid int value: 'Not'

```

Later, when I try to run roslaunch arlobot_bringup minimal.launch --screen to begin teleop, I get an this continuous stream of restart errors until I terminate control C. 

```
ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [7759]
process[arlobot-2]: started with pid [7760]
process[mobile_base_nodelet_manager-3]: started with pid [7761]
process[cmd_vel_mux-4]: started with pid [7769]
process[arlobot_safety-5]: started with pid [7776]
process[arlobot_usbrelay-6]: started with pid [7789]
[ INFO] [1529273155.805748677]: Loading nodelet /cmd_vel_mux of type yocs_cmd_vel_mux/CmdVelMuxNodelet to manager mobile_base_nodelet_manager with the following remappings:
[ INFO] [1529273155.805819798]: /cmd_vel_mux/output -> /mobile_base/commands/velocity
[ INFO] [1529273155.807548546]: waitForService: Service [/mobile_base_nodelet_manager/load_nodelet] has not been advertised, waiting...
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py", line 7, in <module>
    from arlobot_msgs.msg import arloSafety
ImportError: No module named msg
[ INFO] [1529273156.049969370]: Initializing nodelet with 8 worker threads.
[ INFO] [1529273156.057883337]: waitForService: Service [/mobile_base_nodelet_manager/load_nodelet] is now available.
[ INFO] [1529273156.076984243]: CmdVelMux : (re)configured
[arlobot_safety-5] process has died [pid 7776, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py __name:=arlobot_safety __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5*.log
[arlobot_safety-5] restarting process
process[arlobot_safety-5]: started with pid [7890]
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py", line 7, in <module>
    from arlobot_msgs.msg import arloSafety
ImportError: No module named msg
[arlobot_safety-5] process has died [pid 7890, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py __name:=arlobot_safety __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5*.log
[arlobot_safety-5] restarting process
process[arlobot_safety-5]: started with pid [7910]
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py", line 7, in <module>
    from arlobot_msgs.msg import arloSafety
ImportError: No module named msg
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/propellerbot_node.py", line 38, in <module>
    from arlobot_msgs.msg import usbRelayStatus, arloStatus, arloSafety, arloButtons
ImportError: No module named msg
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_usbrelay/scripts/arlobot_usbrelay.py", line 14, in <module>
    from arlobot_msgs.msg import usbRelayStatus
ImportError: No module named msg
[arlobot_safety-5] process has died [pid 7910, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py __name:=arlobot_safety __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5*.log
[arlobot_safety-5] restarting process
process[arlobot_safety-5]: started with pid [7913]
[arlobot-2] process has died [pid 7760, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/propellerbot_node.py --respawnable arlobot/sensor_state:=mobile_base/sensors/core imu/data:=mobile_base/sensors/imu_data imu/raw:=mobile_base/sensors/imu_data_raw __name:=arlobot __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot-2.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot-2*.log
[arlobot_usbrelay-6] process has died [pid 7789, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_usbrelay/scripts/arlobot_usbrelay.py __name:=arlobot_usbrelay __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_usbrelay-6.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_usbrelay-6*.log
[arlobot-2] restarting process
process[arlobot-2]: started with pid [7916]
[arlobot_usbrelay-6] restarting process
process[arlobot_usbrelay-6]: started with pid [7917]
Traceback (most recent call last):
  File "/home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py", line 7, in <module>
    from arlobot_msgs.msg import arloSafety
ImportError: No module named msg
[arlobot_safety-5] process has died [pid 7913, exit code 1, cmd /home/geist/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py __name:=arlobot_safety __log:=/home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5.log].
log file: /home/geist/.ros/log/938ced34-727a-11e8-9920-4851b7a28da2/arlobot_safety-5*.log
[arlobot_safety-5] restarting process

```

... and so forth. 

I'm really appreciate your time and your work that you have put into this. I'm using a Parallax ArloBot as well and if I could just get the serial communication to work between the bot and my ROS computer, I would be able to continue with my research.

Thanks so much,
Dat

vatbrain, 2018-06-18:
Hi. You might check out closed issue #32 regarding "No module named msg". You can not count on things working if you are not getting a clean catkin_make.
I don't understand your parenthetic comment "however it does when I earlier loaded the Simple IDE code into EEPROM". It sounds like it was working? Was it working with direct2PropSerialTest.sh or using the simple IDE terminal?

geistmate, 2018-06-18:
Hi, I checked out issue 32 and it indeed solved the issue after rebuilding a clean one. Thank's very much. As for what I meant in the parenthesis, when I run in the terminal the direct2PropSerialTest.sh, I get the return listed above with
`miniterm.py: error: argument baudrate: invalid int value: 'Not'`

instead of getting a stream of i  0. When I open simpleide instead and load EEPROM and run, I'll see the stream of i   0 in the simple IDE terminal. 

Although, now I can run my keyboard teleop however I'm running into a new error now.  I can only turn left and right using the teleop. (J key and L key are only keys that work. I'm not sure if this is related to my earlier issue and if I should open a new post for this (Very new to github).

vatbrain, 2018-06-19:
A quick suggestion as I don't have much time now: possibly check the "blocked" and "safe to proceed" status bits that get transmitted to see if the robot thinks it is trapped. This could be preventing it from moving forward or backward and could be due to some sensor configuration error. I think I may have had a similar problem.

vatbrain, 2018-06-20:
FYI, I also get the original "miniterm.py: error: argument baudrate: invalid int value: 'Not'
" error when running direct2PropSerialTest.sh
My keyboard teleop works fine though.

vatbrain, 2018-06-20:
In my case, the problem seems to be with find_ActivityBoard.sh when it does not initially find a "Propeller_Activity_Brd". Some of the text information echoed is being interpreted by miniterm.py as a baud rate which causes the problem.

geistmate, 2018-06-21:
Great, thanks much for your help to fix the initial issue. I actually have a new different serial issue though, but since this one is resolved, I will close it. With limited ROS knowledge, trying to get this set up has been frustrating. I'm lucky to receive help, though. Thank you.

closed 2018-06-21 by geistmate