chrisl8/ArloBot #36: Watchdog Timeout Reset

geistmate, 2018-06-21:
I am having an issue getting my arlobot to start-up. I previously had my teleop half working (only angular rotation, no vertical movement), but after loading in a code where I changed my sensor configurations to try to fix the issue I started to get this error and teleop will no longer work. Loading in the previous code that half-worked, I still get this issue now when I run roslaunch arlobot_bringup minimal.launch --screen and run the teleop command in another terminal. 

```
NODES
  /
    arlobot (arlobot_bringup/propellerbot_node.py)
    arlobot_safety (arlobot_safety/arlobot_safety.py)
    arlobot_usbrelay (arlobot_usbrelay/arlobot_usbrelay.py)
    cmd_vel_mux (nodelet/nodelet)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [17805]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 97f45e30-7445-11e8-b4d5-4851b7a28da2
process[rosout-1]: started with pid [17828]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [17846]
process[arlobot-3]: started with pid [17847]
process[mobile_base_nodelet_manager-4]: started with pid [17848]
process[cmd_vel_mux-5]: started with pid [17852]
process[arlobot_safety-6]: started with pid [17866]
process[arlobot_usbrelay-7]: started with pid [17873]
[ INFO] [1529470291.174246437]: Loading nodelet /cmd_vel_mux of type yocs_cmd_vel_mux/CmdVelMuxNodelet to manager mobile_base_nodelet_manager with the following remappings:
[ INFO] [1529470291.174334919]: /cmd_vel_mux/output -> /mobile_base/commands/velocity
[ INFO] [1529470291.181687790]: waitForService: Service [/mobile_base_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1529470291.596911012]: Initializing nodelet with 8 worker threads.
[ INFO] [1529470291.605935223]: waitForService: Service [/mobile_base_nodelet_manager/load_nodelet] is now available.
[ INFO] [1529470291.636239200]: CmdVelMux : (re)configured
[INFO] [1529470292.581021]: Starting with serial port: /dev/ttyUSB0, baud rate: 115200
[INFO] [1529470292.581247]: Starting OdomStationaryBroadcaster
[INFO] [1529470292.581683]: Serial Data Gateway starting . . .
[INFO] [1529470292.582722]: Serial Data Gateway started.
[INFO] [1529470292.582893]: Propellerbot_node has started.
[INFO] [1529470292.611700]: No USB Relay board installed.
[INFO] [1529470314.560246]: Watchdog Timeout Reset initiated
[INFO] [1529470314.560624]: Serial Data Gateway stopping . . .
[INFO] [1529470314.560901]: Stopping serial gateway
[INFO] [1529470314.661388]: Serial Data Gateway stopped.
[INFO] [1529470314.661767]: 5 second pause to let Activity Board settle after serial port reset . . .
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/dat/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/SerialDataGateway.py", line 60, in _Listen
    self.ReceivedLineHandler(data)
  File "/home/dat/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/propellerbot_node.py", line 159, in _handle_received_line
    self._initialize_drive_geometry(line_parts)
  File "/home/dat/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/propellerbot_node.py", line 847, in _initialize_drive_geometry
    self._write_serial(message)
  File "/home/dat/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/propellerbot_node.py", line 738, in _write_serial
    self._SerialDataGateway.Write(message)
  File "/home/dat/catkin_ws/src/ArloBot/src/arlobot/arlobot_bringup/scripts/SerialDataGateway.py", line 65, in Write
    self._Serial.write(data)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 517, in write
    raise portNotOpenError
SerialException: Attempting to use a port that is not open
```

vatbrain, 2018-06-29:
I have been modifying and uploading the latest propeller code over the past couple of days keeping an eye out for serial data problems. The only time I have seen problems is when I accidentally have a terminal connected to the serial port when I run roslaunch arlobot_bringup minimal.launch --screen.
Have you made any progress with this issue?

geistmate, 2018-08-06:
Hi, I took a long break from this project due to being busy with summer classes. I'm returning, but I'm still having no luck with this. What do you mean by having a terminal connected to a serial port? My process is: start a roscore, run arlobot_bringup minimal.launch --screen, and then run arlobot_teleop keyboard_teleop.launch.

vatbrain, 2018-08-06:
I meant having a program with a terminal like simple IDE or an arduino IDE open on the same computer that is running ROS at the same time you are attempting to run ROS. The only time I have seen serial problems was following direct terminal interaction with the propeller board when I forgot to close the terminal program before launching ROS.

geistmate, 2018-08-06:
Oh, I see. Yeah I occasionally had that issue too where I forgot to close SimpleIDE before moving on; however it doesn't seem that to be the case to the error this time. I'm going to be doing various things to try to troubleshoot today so I'll get back here with my efforts soon.

geistmate, 2018-08-09:
I ran all of the connection tests provided by Parallax and all of them passed. I don't think there are any errors with the board or the motors.

vatbrain, 2018-08-11:
Hi,
A couple of ideas, probably long shots.
1) Does your user id have dialout group permissions? When you type "groups", does dialout appear? I was searching for instructions that I followed to eliminate serial data/USB problems with a camera that I have been experimenting with that involved making some changes to the linux environment that may have been helpful, but I have been unable to find them.
2) Did you happen to run the instructions from the link on Chris's blog to prevent the propeller activity board from being powered over USB? I have NOT done this.  Are you doing anything with modified cables to accomplish the same thing? In reading about this I got the impression that this could potentially be a source of problems in some situations.
Sorry I can not be more help.

geistmate, 2018-08-16:
Hi Vatbrain,

Yeah I've added my id to dialout already (it was already added during the quick install too), so I don't think that is the problem too much.

I haven't actually done the disable USB power; I'm just the robot to my computer via the mini-usb. I tried multiple mini-usb to see if that was the issue yet no. 

Yeah, the board is set at 2 as well. I tried it at both switch positions.

Looking at a previous opened issue, Chris stated that the Watchdog Timeout Reset initiated signifies that the serial communication doesn't work, and times out and resets while waiting for data from the board and tries again. It seems fine; I can still load in code properly. When it's connected to ros and I have teleop running, hold down some keys, I have both the RX and TX lights flashing, and the serial light is on too; until it times out for a bit. 

I was thinking something was on in the per_robot_settings_for_propeller_c_code.h, so that when I turned on the motor switch making the DHB-10 turn on, something became initialized and the propeller board was constantly waiting for information back from it, since the timeout happens only when when I turn on the motor controller; however, I ran all the encoder and motor tests and they seem to all be communicating well. When it's off, the four ping sensor configuration still flashes its green lights signifying its working. 

I thought my board might be the problem so I just got a new Activity Board, hooked it up, loaded stuff in, but still no luck. The only thing I need to be changing is the per_robot_settings_for_propeller_c_code.h right? 

Thanks for staying in communication and lending me some suggestions I really appreciate it. I just need for ROS to be able to connect with the board so I could move on to use it with my own research at my university. It's becoming extremely frustrating because this is just the only step I need to work before I can move on, yet I've no clue how to fix this issue.

vatbrain, 2018-08-17:
Hi,
If you post/attach your  per_robot_settings_for_propeller_c_code.h, I'd be happy to compare it to mine and see if anything jumps out at me.  The other files that might be interesting to compare would be propellerbot_node.py and SerialDataGateway.py just to see if we are using the same versions.

geistmate, 2018-08-17:
Great; I will do that first thing tomorrow morning when I get back to my work station. Also, I'm quite new to github. What/where is the best way for me to upload this code for you to view it?

vatbrain, 2018-08-17:
I'm pretty new to github myself. I see the note at the bottom of the box where I am entering this that says "Attach files by dragging & dropping, selecting them, or pasting from the clipboard". That sounds like the simplest way to me. I just tried to drag and drop my per_robot_settings... .h file and it gave me a message that it does not support that file type. It did list ZIP as a listed type, so you could try zipping the files together and then dragging and dropping. I suspect the correct way would be for you to fork Chris's repository and upload your changes to your own fork. I have done this, but it is more involved.
I can't promise that I will get to comparing the files quickly, as I am going to be quite busy the next couple of days, but I should have more time after that.

geistmate, 2018-08-18:
That's completely fine with me; I already appreciate all the effort you're putting in to help me. Thanks very much. I've attached a .zip file with the per robot settings file as well as the propellerbot_node.py and SerialDataGateway.py. 

For the robot settings, simply all I did was edit the ping sensor configuration. I am using the Arlo Complete Robot Kit provided by parallax, with the 4 ping sensors connected directly to the P14, P15, P16, and P17 headers. I also changed the ROS timeout and increased it to see if that would have made any difference.
[geistmateArloBot_perRobotSettings.zip](https://github.com/chrisl8/ArloBot/files/2299844/geistmateArloBot_perRobotSettings.zip)

vatbrain, 2018-08-22:
Hi.
1) What strikes me about your configuration is that your front ping sensors are not contiguous.
#define FIRST_FRONT_PING_SENSOR_NUMBER 14
#define HOW_MANY_FRONT_PING_SENSORS 3
#define FIRST_REAR_PING_SENSOR_NUMBER 15
While I don't understand how this could cause serial communication problems, I believe it will cause other problems. I would recommend changing the rear ping to P17.
Also, I assume you have the jumpers next to the servo headers on the center pin and the 5V pin and not on the center pin and the VIN pin?

2) #define ROS_TIMEOUT 1000   Probably not an issue, but maybe try setting this to something < 127 in case it is somehow being passed through a byte and being truncated? Don't have time to study the code to see if this could be a problem.

3) One other difference I just noticed. Your file has the following:
// Quickstart Board Settings
// I have a SECOND Propeller chip on a Quickstart board that polls the PING and IR sensors, so this is settings for that.
// Which pin on the QuickStart Board is the RX pin connected out to the Activity board?
#define ACTIVITYBOARD_RX_PIN 5
// Which pin on the QuickStart Board is the TX pin connected in from the Activity board?
#define ACTIVITYBOARD_TX_PIN 6

My file does not have this. This is related to serial communications and merits further investigation but I do not have time right now. 

I have to get ready to go to work  now, but will look again this evening.

vatbrain, 2018-08-22:
#3 above should not be an issue.
Forgot to mention that the propellerbot_node.py and SerialDataGateway.py you uploaded are identical to what I am using.

vatbrain, 2018-08-22:
I'm at a loss. Spent some more time looking at your configuration versus mine and did not find anything other than what I have already noted above. 
A few more miscellaneous questions:
1) What types/brands of laptops have you tried?
2) Does find_ActivityBoard.sh (in the scripts directory) run properly?
3) Does check_hardware.sh (in the scripts directory) run properly?

geistmate, 2018-09-05:
Hey Vatbrain,

I've tried many different laptops. 
I tried running off a Dell XPS 13, Aorus X3, Razer Blade 14, and a few other low-powered laptops. And yes, find_ActivityBoard.sh and check_hardware.sh both run properly. 

Sadly, I've given up on this package for it since I cannot afford to waste more time. I've been using an Arduino Mega and the rosserial library in order to publish my data. 

I appreciate all the effort you put into trying to help me; hopefully this can get resolved in the future for others that want to use this ArloBot package. 

Best regards,
Matthew

chrisl8, 2019-11-14:
This was a great discussion. I apologize that I was not able to get involved at the time.

I'm going to close this as it is old now. If you feel I can still contribute to a solution for you please reopen this issue or open another one.

Thank you!

closed 2019-11-14 by chrisl8