chrisl8/ArloBot #38: Issue using start-robot.sh and starting the web interface

qtdavidson55, 2018-07-24:
When I run start-robot.sh, I get several errors printed to the screen. First the arlobot_goto.py script gets the error 'ImportError: No module named hector_nav_msgs.srv'. Then after initializing the nodelets, I get messages saying that 'No matching device found.... waiting for devices' and 'Invalid device number 1, there are 0 devices connected.' In addition to this the script causes the wheels on the arlobot to rotate, and they don't stop rotating upon killing the script, I have to manually restart the power to the motors. 

When starting the web interface with the desktop shortcut, the webpage appears with the repository url as well as stating that no robot is found. When running the start-robot.sh, this screen changes to the actual GUI but after several seconds reverts back to the no robot is found message. 

I have already compiled the firmware with SimpleIDE, and have performed the direct2PropSerial.sh script, with both being successful. I'm using the ArloBot kit, and all of the basic ArloBot-provided tests have worked successfully as well. I'm running all of this on the UbiquityRobotics image whch is Ubuntu 16.04 with ROS Kinetic on Rasberri Pi. This is my first time working with the ArloBot, and I'm relatively new with ROS, so I'm not sure if there is something obvious I'm doing wrong. I've attached the messages printing while running robot-startup.sh if that helps at all. 
[robot-startup.log](https://github.com/chrisl8/ArloBot/files/2224638/robot-startup.log)

qtdavidson55, 2018-07-25:
So I was able to resolve the hector_nav_msgs.srv error. I was missing package that I had to find and install. However, the rest of my issues are the same. In addition to this, I'm receiving two warning messages: "Unable to start server: [Errno 98] Address already in use." and "Joint state with name: "left wheel_joint" was received but not found in URDF." I'm doing everything the same as before, and I've attached the messages I was getting. 

```Clearing ROS Logs . . .
Checking all configured devices to make sure they are available . . .
Checking Activity Board . . .
Hardware Check SUCCESS! All devices found.
Usage: rosparam set [options] parameter value

rosparam: error: too many arguments
Use kill_ros.sh to close.
... logging to /home/ubuntu/.ros/log/175f4368-8f6e-11e8-ab64-2d38325e088b/roslaunch-ubiquityrobot-9151.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubiquityrobot:38411/
ros_comm version 1.12.13

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES

roscore cannot run as another roscore/master is already running. 
Please kill other roscore/master processes before relaunching.
The ROS_MASTER_URI is http://ubiquityrobot:11311/
The traceback for the exception was written to the log file
... logging to /home/ubuntu/.ros/log/175f4368-8f6e-11e8-ab64-2d38325e088b/roslaunch-ubiquityrobot-9221.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubiquityrobot:39011/

SUMMARY
========

PARAMETERS
 * /arlobot/baudRate: 115200
 * /arlobot/bonus: False
 * /arlobot/driveGeometry/distancePerCount: 0.00338
 * /arlobot/driveGeometry/trackWidth: 0.403
 * /arlobot/ignoreCliffSensors: False
 * /arlobot/ignoreFloorSensors: False
 * /arlobot/ignoreIRSensors: False
 * /arlobot/ignoreProximity: False
 * /arlobot/monitorACconnection: False
 * /arlobot/update_rate: 30.0
 * /arlobot/usbLeftMotorRelay: 6
 * /arlobot/usbLeftMotorRelayLabel: LeftMotor
 * /arlobot/usbRelayInstalled: False
 * /arlobot/usbRightMotorRelay: 2
 * /arlobot/usbRightMotorRelayLabel: RightMotor
 * /arlobot_usbrelay/relay1enabled: False
 * /arlobot_usbrelay/relay1label: Empty
 * /arlobot_usbrelay/relay2enabled: True
 * /arlobot_usbrelay/relay2label: RightMotor
 * /arlobot_usbrelay/relay3enabled: False
 * /arlobot_usbrelay/relay3label: Empty
 * /arlobot_usbrelay/relay4enabled: False
 * /arlobot_usbrelay/relay4label: Empty
 * /arlobot_usbrelay/relay5enabled: False
 * /arlobot_usbrelay/relay5label: Empty
 * /arlobot_usbrelay/relay6enabled: True
 * /arlobot_usbrelay/relay6label: LeftMotor
 * /arlobot_usbrelay/relay7enabled: True
[... 124 lines cut ...]
[INFO] [1532519337.280767]: Serial Data Gateway started.
[INFO] [1532519337.281948]: Propellerbot_node has started.
[INFO] [1532519337.396316]: Active Controller: idle
[ INFO] [1532519339.425299978]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[WARN] [1532519342.156535]: Unable to start server: [Errno 98] Address already in use Retrying in 5s.
[ INFO] [1532519342.426119438]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[ INFO] [1532519345.427039210]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[ WARN] [1532519345.700374970]: Joint state with name: "left_wheel_joint" was received but not found in URDF
[WARN] [1532519347.161551]: Unable to start server: [Errno 98] Address already in use Retrying in 5s.
[ INFO] [1532519348.428141221]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[ INFO] [1532519351.429053389]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[WARN] [1532519352.170121]: Unable to start server: [Errno 98] Address already in use Retrying in 5s.
[ INFO] [1532519354.430273004]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[ WARN] [1532519355.900331854]: Joint state with name: "left_wheel_joint" was received but not found in URDF
[WARN] [1532519357.181692]: Unable to start server: [Errno 98] Address already in use Retrying in 5s.
[... cut ...] /openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.
[WARN] [1532519372.209268]: Unable to start server: [Errno 98] Address already in use Retrying in 5s.
[ INFO] [1532519372.435829098]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.3.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.```

chrisl8, 2019-11-14:
I apologize that I was not able to respond when you posted this.

Are you still using this code, and if so, is there still anything I can do to assist?

If not, I will close this issue.

I'm sorry that I didn't have an answer for you at the time.

chrisl8, 2019-11-30:
Closing due to inactivity and the amount of changes that have happened since this issue was opened.

closed 2019-11-30 by chrisl8