chrisl8/ArloBot #39: Issue bringing up arlobot

MStojkovic, 2018-08-24:
Hello,
I have a problem bringing up arlobot. When I use the command: "roslaunch arlobot_bringup minimal.launch --screen", I get an error saying there is no module named "msg" : "File "/home/pixy/catkin_ws/src/ArloBot/src/arlobot/arlobot_safety/scripts/arlobot_safety.py", line 7, in <module>
    from arlobot_msgs.msg import arloSafety
ImportError: No module named msg".
I am using ubuntu 16.04 and have installed everything using your quick install script. My python version is 2.7.12. Arlo was bought as a complete package and was assembled using Parallax guide.

vatbrain, 2018-08-24:
Hi. See issue #32 . Probably better to delete the offending package rather than white listing it.

MStojkovic, 2018-08-24:
Hello, that did work around the issue and I apologize for not seeing issue #32. However I am still unable to get teleoperation working. When I launch teleop I can't get the robot to move. I can change speeds with "q/z/w/x/e/c/" but can't get the robot to move. I also tried using the "quick start" script, but had no luck. I even tried the option to ignore ping sensors, but it didn't work either. While running the script I kept getting this error: "Error connecting to RosLibJS server: [object Event]", is that normal?
Lastly, thank you for your quick response.

vatbrain, 2018-08-24:
Hi, No need to apologize. Glad that helped.
Have you run the basic parallax arlo tutorials to make sure the motors and encoders are wired and work properly? If not, that is the first place to start.
Next, I would try getting the robot to move running the ROSInterfaceForArloBotWithDHB10.c propeller code and sending it commands using a terminal program as outlined in direct2PropSerialTestLines.txt  I had better luck using the terminal from the Arduino IDE than other methods. This is a key step. Let us know if you can move the robot this way. Good luck!

MStojkovic, 2018-08-24:
Hello, yes I ran everything you mentioned and it all works fine. The robot even moves back if I put my hand to close to one of the ping sensors. When I run the "quick start" script I don't get true/false for atributes for keys in "telemetry", but I guess thats because I disabled those modules like it is suggested in isssue #32. Also I don't get anything in "relays". Thank you for your help.

vatbrain, 2018-08-24:
I'm not sure whether I have ever run the quick start script. I have never been able to get the robot to move using the arlobot desktop icon and web interface.   What has worked for me is:
```
roslaunch arlobot_bringup minimal.launch --screen
<New Terminal>
roslaunch arlobot_teleop keyboard_teleop.launch
```
The last thing I did to get that to work was to add some DEFAULT_MOTOR_ADC_VOLTAGE code to per_robot_settings_for_propeller_c_code.h and ROSInterfaceForArloBotWithDHB10.c to cause a non-zero voltage to be sent to propellerbot_node.py, but these changes have been merged for some time.

MStojkovic, 2018-08-26:
There is also 1 other thing I wanted to mention: When I run direct2PropSerialTest I says I should be getting a steady stream of " i 0", but I seem to be getting just one "i 0" combination. At first it looked like the next line was just overwriting the first one, but I am not so sure now. Also when I copy the suggested line for DHB-10 I am getting the expected odometry output, but the safety line seems to exist for only half a second or so and then it gets overwritten. Despite it, when I run the commands for the robot to move forwards/backwards/turn it still does so normally.

MStojkovic, 2018-09-04:
I have looked at some other issues and the google group and I found out I wasn't the only one having an issue with serial communication. Can you please test teleop on your own setup and if it works can you please tell me what configuration are you using (pc, laptop, Raspberry, etc.. and what os)? I have been looking through the scripts for serial communication and it looks like "SerialDataGateway.py" is trying to read from the activity board, before it is reset, but that doesn't stop the script and it continues forward. I assume it is supposed to constantly print out messages: 
[INFO] [1536078851.143213]: Watchdog Timeout Reset initiated
[INFO] [1536078851.143887]: Serial Data Gateway stopping . . .
[INFO] [1536078851.144469]: Stopping serial gateway
[INFO] [1536078851.245311]: Serial Data Gateway stopped.
[INFO] [1536078851.245865]: 5 second pause to let Activity Board settle after serial port reset . . .

In the same way "propellerbot_node.py" is trying to write data via serial when the activity board is not ready, but the script still moves on. 

Just now I finally found out what is wrong:
The error is in "propellerbot_node.py" in "handle_velocity_command" function. I believe that "self._clear_to_go("to_stop"):" always happens. It seems like the robot always "thinks" that it's not safe to move. It might be due to the same issues I had at the beggining with "no module named msg", especially "ArlobotSafety".

vatbrain, 2018-09-05:
Hi. My teleop works. I was driving the robot all weekend and just now. I am using a lenovo thinkpad laptop running lubuntu 16.04.1 I am connected to a USB 3 port.  Originally, I was getting tripped up at the top of the clear_to_go routine. I put in a bunch of print statements and eventually figured out that it was the _leftMotorPower and _rightMotorPower that was preventing my robot from moving. You should be able to do the same and figure out exactly what is causing a problem in this routine. I would start with the conditions at the top.

MStojkovic, 2018-09-06:
Hello, thank you for your help. I will take a look at that. Do you know how to find the exact log it is writing in since there are a lot of them? Also, the serial connection seems to work for 20 seconds, then resets for 5s and keeps doing it all the time. It looks like it is supposed to be like that, but just wanted to confirm. If it is intentional, do you know why is it like that. Does ROS send too much commands to the activity board so it overflows its buffer, or is there another reason? You can contact me at stojkovicmatija@gmail.com if you don't want to prolong this thread.

vatbrain, 2018-09-07:
Hi. The robot will not drive if left/rightMotorPower are false. Chris accepted a pull request of mine that was supposed to fix this under certain configurations. 
I just found a link related to some potential serial port issues under linux that I had used before:
http://www.jevois.org/doc/USBserialLinux.html
Check out what it says about Permission Issues and ModemManager. I took the actions outlined there and have not been having the serial port issues others have been having recently. That may or may not the the reason. Worth a try though.

geistmate, 2018-10-01:
Hi MStojkovic. I just wanted to know if you were able to solve the serial timeout issue?

MStojkovic, 2018-10-02:
Hello geistmate,
sadly no. I have tried everything I know butt the connection still breaks after a couple of secconds and just repeats the final command.

chrisl8, 2019-11-14:
I apologize that I was not able to respond when you posted this.

Are you still using this code, and if so, is there still anything I can do to assist?

If not, I will close this issue.

I'm sorry that I didn't have an answer for you at the time.

chrisl8, 2019-11-30:
Closing due to inactivity and the amount of changes that have happened since this issue was opened.

closed 2019-11-30 by chrisl8