chrisl8/ArloBot #40: heading error when running PropellerSerialTest.sh

robotsir, 2019-04-05:
Hi,

I was testing with ~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh, and I noticed that when I moved the robot forward with keyboard, the heading number increased from 0 to 2.0, then it overflew and jump back to 0. I would expect the heading to maintain around 0, since the robot moved forward, without any turning. Also, is the unit here in radians? if so, shouldn't it overflow when it reach to 3.14? I confirmed that the encoders were working properly with: https://learn.parallax.com/tutorials/robot/arlo/arlo-activity-board-brain/control-arlo-simpleide-terminal. Any ideas?

Thanks!

robotsir, 2019-04-10:
OK, I found the bug. In broadcastOdometryAndRunMotors.h, change uint8_t dhb10_ticksLeft, dhb10_ticksRight... to int64_t ..., since the encoder value can be +/- and way beyond 255

chrisl8, 2019-04-19:
Yay, bug fixes!! 

I'll try to update the code this weekend with your fix, or you can send me a pull request if you want credit on github. :)

commit ed8d3876 (2019-04-23): Closes: #40
closed 2019-04-23 by chrisl8