chrisl8/ArloBot #121: PropellerSerialTest.sh - ActivityBoard Error

Paumila, 2020-04-02:
Hello,

I am trying to follow your ArloBot projecte,because few months ago I bought it. I'll try to manage it with ROS but with Arduino board. And it was successfull but I was stuck on the odometry.

Now, I am trying to undertsand a little bit your project to execut this with propeller board to after try to execute the same with Arduino.

I follow all your README.md

1. Requirements
2. Quick Install
3. All tutorials from parallax
4. Propeller Serial Interface Test

At this point I can not continue because it apperas the following error: (I have the computer connected to the propeller board and the main and motor interruptors ON and propeller switch at position 2)

pau@pau-Lenovo-Legion-Y730-15ICH:~/catkin_ws/src/ArloBot/scripts$ ./PropellerSerialTest.sh
Checking all configured devices to make sure they are available, attempt 1
Checking Activity Board . . .
/home/pau/catkin_ws/src/ArloBot/scripts/find_ActivityBoard.sh: línea 20: node: orden no encontrada
/home/pau/catkin_ws/src/ArloBot/scripts/find_ActivityBoard.sh: línea 26: node: orden no encontrada
Activity Board missing! If this is a test install with no Activity Board, edit /home/pau/.arlobot/personalDataForBehavior.json and set 'hasActivityBoard' to false
ERROR: Hardware failed to come on line:

I hope somebody could help me with this error.

I amb very interested on understand this project.

Best Regards,
Pau

chrisl8, 2020-04-05:
The error is indicating that it cannot find the Activity Board when it runs the script

`home/pau/catkin_ws/src/ArloBot/scripts/find_ActivityBoard.sh`

You said the serial test scripts worked through?

chrisl8, 2020-04-05:
I just realized your output is in Spanish, and it is saying that the node command is not found.

Did you run the install script?

Can you run `node` from the command line?

I can give you more tips on getting node installed, if the installation script isn't working,
Or you can just use NVM https://github.com/nvm-sh/nvm
Which the installation script should have installed,

Or go to nodejs.org and follow their installation instructions if you prefer, although I think NVM is better.

chrisl8, 2020-07-18:
@Paumila Do you still need assistance with this, or can I close this issue?

chrisl8, 2020-08-18:
Closing due to inactivity. Please reopen if you still need help.

closed 2020-08-18 by chrisl8