chrisl8/ArloBot #154: Install Script not working /dev/fd/63: line 601: rosdep: command not found

Tristan9497, 2020-10-16:
Hi,
i am interested in exploring your project. But the install script doesn't seem to work.
Of course there is the option to just manually clone, but i wanted to let you know about this.

I am running Ubuntu Mate 20.04 and am using ROS noetic on my machine allready
Let me know if you need further informations. The following is the Terminal output of your script:

------------------------------------------------------------------------------------------------------------------------------

SETTING UP ROS noetic FOR YOUR ARLOBOT!
---------------------------------------------------
You will be asked for your password for running commands as root!

[Checking the Ubuntu version]
Ubuntu focal found

[Updating Root CA Certificates from Ubuntu]
[sudo] password for tristan: 
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

done.
done.

[Updating & upgrading all existing Ubuntu packages]
Hit:1 http://de.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease              
Hit:3 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease            
Get:4 https://deb.opera.com/opera-stable stable InRelease [2.591 B]            
Get:5 http://security.ubuntu.com/ubuntu focal-security InRelease [107 kB]  
Get:6 https://deb.opera.com/opera-stable stable/non-free amd64 Packages [1.876 B]
Get:7 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [1.256 B]
Hit:8 http://packages.ros.org/ros/ubuntu focal InRelease             
Fetched 113 kB in 1s (81,4 kB/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
3 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Target Packages (non-free/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Packages (non-free/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Packages (non-free/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Translations (non-free/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Translations (non-free/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target CNF (non-free/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target CNF (non-free/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Packages (non-free/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Packages (non-free/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Packages (non-free/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Translations (non-free/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target Translations (non-free/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target CNF (non-free/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
W: Target CNF (non-free/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:58 and /etc/apt/sources.list.d/opera-stable.list:4
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  python3-distupgrade ubuntu-release-upgrader-core ubuntu-release-upgrader-gtk
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 137 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 ubuntu-release-upgrader-gtk all 1:20.04.28 [9.364 B]
Get:2 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 ubuntu-release-upgrader-core all 1:20.04.28 [23,7 kB]
Get:3 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 python3-distupgrade all 1:20.04.28 [104 kB]
Fetched 137 kB in 0s (746 kB/s)               
(Reading database ... 340452 files and directories currently installed.)
Preparing to unpack .../ubuntu-release-upgrader-gtk_1%3a20.04.28_all.deb ...
Unpacking ubuntu-release-upgrader-gtk (1:20.04.28) over (1:20.04.25) ...
Preparing to unpack .../ubuntu-release-upgrader-core_1%3a20.04.28_all.deb ...
Unpacking ubuntu-release-upgrader-core (1:20.04.28) over (1:20.04.25) ...
Preparing to unpack .../python3-distupgrade_1%3a20.04.28_all.deb ...
Unpacking python3-distupgrade (1:20.04.28) over (1:20.04.25) ...
Setting up python3-distupgrade (1:20.04.28) ...
Setting up ubuntu-release-upgrader-core (1:20.04.28) ...
Setting up ubuntu-release-upgrader-gtk (1:20.04.28) ...
Processing triggers for man-db (2.9.1-1) ...

[Installing additional Ubuntu and ROS Packages for Arlo]
This runs every time, in case new packages were added.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.25.1-1ubuntu3).
ros-noetic-joy is already the newest version (1.14.0-1focal.20200929.230623).
ros-noetic-move-base is already the newest version (1.17.1-1focal.20200930.014742).
ros-noetic-navigation is already the newest version (1.17.1-1focal.20200930.015946).
ros-noetic-rosbridge-server is already the newest version (0.11.10-1focal.20200930.001520).
ros-noetic-tf2-tools is already the newest version (0.7.5-1focal.20200930.011506).
ros-noetic-twist-mux is already the newest version (3.1.1-1focal.20200930.012057).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[Cloning or Updating git repositories]
ArloBot repository
Already on 'noetic'
Your branch is up to date with 'origin/noetic'.
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Total 5 (delta 4), reused 5 (delta 4), pack-reused 0
Unpacking objects: 100% (5/5), 1.09 KiB | 33.00 KiB/s, done.
From https://github.com/chrisl8/ArloBot
 * [new branch]      renovate/react-monorepo -> origin/renovate/react-monorepo
Already up to date.

Slam Toolbox
There *is* a package for this, but the Github Repo is usually much more up to date.
Already on 'noetic-devel'
Your branch is up to date with 'origin/noetic-devel'.
Already up to date.

TEB Local Planner Tutorials
The TEB Local Planner is installed via apt, but there is no noetic package for the tutorials.
The tutorials are not required, but handy for reference.
Already on 'noetic-devel'
Your branch is up to date with 'origin/noetic-devel'.
Already up to date.

[Installing dependencies for ROS build-from-source packages.]
/dev/fd/63: line 601: rosdep: command not found

INSTALL FAILURE!!!
The Install Script has failed. Please investigate the cause, correct, and run again before proceeding.

If this was a transient error, such as a network failure connecting to something, you may just need to run it again.

If this persists, please file an issue against the repository at https://github.com/chrisl8/ArloBot/issues

chrisl8, 2020-10-16:
It cannot find `rosdep`, which is strange.
If you run:
`which rosdep` what is the result?

Tristan9497, 2020-10-16:
Ok, that is strange, i don't get any response in the terminal. Probably indicating it isn't even there...
I'll try install it manually and check again if your skript works

Thanks for the fast response

Tristan9497, 2020-10-16:
Ok, everything is working now.
I should have figured this out myself but thanks for your help.

I'll close the issue.

closed 2020-10-16 by Tristan9497

chrisl8, 2020-10-16:
No problem. My script should have installed it for you. My guess is that you had ROS partly installed, but my logic assumed it was "all or nothing", which is incorrect on my part.

Let me know if you run into any other issues. I appreciate the feedback!