chrisl8/ArloBot #155: front and Rear Caster of arlo.urdf.xacro and arlo_2stack.urdf.xacro clip through gazebo groundplane (no defined collision)

Tristan9497, 2020-10-21:
Hi,

i wanted to play around with a simulated bot before running my "real" one. It's awesome that you have xacro descriptions for your robot, thanks for that!

I have launched your bot in a gazebo world and the front and rear caster wheels clip through the Groundplane, my guess is that here is no collision defined for the link. 
The definition for these wheels in arlo_2stack.urdf.xacro is the same and as expected this clips the ground plane aswell.

closed 2020-11-09 by Tristan9497

chrisl8, 2020-11-09:
@Tristan9497  If you think this would help everybody, please make a Pull Request for your updates.

I was going to try it myself, but I am very short on time recently.

Tristan9497, 2020-11-09:
First of all i want to mention, that the fix i proposed earlier isn't really a clean solution. Imo the collision should be a fairly primitive shape to save resources (probably a sphere to have only one collision point). 
Furthermore i had problems with your version since the casters where taller then the diff_drive wheels and are lifting the robot from the ground.

I have changed a lot in the urdf and configured it to the structure and sensors of my bot especially regarding plugins for camera lidar and diff drive. 
If i have the time later this week i'll try to modify your urdf-file and make a request so you can check it out and decide if you want to use it or not. Otherwise a urdf like "gazebo compatible" might be nice in addition to your stuff so we don't have to change anything and have full functionality everywhere.

Let me know if you want the gazebo plugins or not. If yes i can porvide an additional .gazebo file with the plugin implementation.