chrisl8/ArloBot #170: Failed install

truckwiz, 2021-07-10:
At the end of the install process it ask me to log into github, I log in and it say's "Repository not found" and then "INSTALL FAILURE".

chrisl8, 2021-07-12:
It should not require you to log in to github. I will run the setup script again on my test box and see if perhaps I introduced a bug to the install script recently...

truckwiz, 2021-07-13:
Hi,
 I tried installing the Melodic version on 2 machines with the same Git
password at the end. I did get Noetic installed on one machine with no
problem.The reason I need to use the Melodic version is that I'm trying to
use it on a Jetson Nano and that's the newest Linux available for it.

Thanks'

chrisl8, 2021-07-13:
Oh sorry, I didn't realize that you were trying Melodic. I don't tend to keep up with the old versions, so they can definitively have issues.

I'm pretty sure that the issue is here:
https://github.com/chrisl8/ArloBot/blob/772bf88dc1e0b317cf478a1e2036815592c1ac46/setup-melodic.sh#L720

Because it is attempting to pull from a repo that doesn't exist anymore.

I can try to fix up that script to make it run again, but it may be a few days before I get to it.

You can also just copy it to your PC and edit it, or fork the repo, clone it, fix it, and make a PR. 😄

chrisl8, 2021-07-13:
@truckwiz I made an update to the melodic setup script. Give it a try and see if it gets further.

chrisl8, 2021-07-13:
Update:
1. I did find some time.
2. It appears there are a few issue.
3. Fortunately Travis.ci is helping me find them, so I'm updating them one at a time.
4. I will post another update when he Travis.ci build passes, and then hopefully it will work for you also.

chrisl8, 2021-07-13:
@truckwiz  My last push passed in Travis, so please give it a try on your system now and let me know if it is working.

truckwiz, 2021-07-14:
Hi,
 Yes that worked on the desktop, Now I will try it on the Jetson Nano.
Thank You for getting it fixed so fast!

truckwiz, 2021-07-15:
I tried it on the Jetson Nano today , if fails at the simple IDE
installation. I looked at the installation script and it looked like the
only thing left was to copy the yaml file to . arlobot so I did that from
the terminal, still testing to see if works.
 I do have another question, if I use the activity board to read the
encoders the code fails in the encoders file, something about the variable
not being declared or something like that.

chrisl8, 2021-07-16:
If you got it as far as cloning the repository, you might try going into the cloned folder and edit the setup-melodic.sh file to comment out the parts that don't work, then run it again from their directly instead of via wget. 

At least that way you know it entirely finished before trying to debug issues.

I'm in my phone now, but I can look closer at the code tomorrow on my PC and suggest more ideas.

truckwiz, 2021-07-16:
It did fully install by taking the Simple IDE out. I have the serial port
test running. although the motor's don't respond. I tested them with the
Ros Noetic version that I have on my laptop.

Thanks' for all the help.

chrisl8, 2021-07-19:
>  if I use the activity board to read the
encoders the code fails in the encoders file, something about the variable
not being declared or something like that.

Did you edit the `~/.arlobot/per_robot_settings_for_propeller_c_code.h` file and then compile the code for the Propeller board and install it?

truckwiz, 2021-07-21:
I'm using a Windows machine to modify and load code into the activity
board, when I modify the "per_robot_settings_for propeller_c_code.h" and
then try to load the "ROSInterfaceForArloBot" I get these error's (see
screenshots)

There is another thing I see in the DHB-10 firmware, is it supposed to be
using the same pin for RX and TX ?

Thanks'

TCIII, 2021-12-06:
I am having a similar problem when trying to install the ArloBot software on the Propeller Activity Board when running on a Nano 4GB.
Please see [issue #174](https://github.com/chrisl8/ArloBot/issues/174)

TCIII, 2021-12-12:
@truckwiz,

If you are trying to use a small SBC, like the NVIDIA Nano 4GB, in place of a laptop but need a x86 processor, I can recommend the LattePanda 4GB/64GB eMMC SBC. Since the Nano 4GB price has risen from $99 to around $200 or more the [LattePanda](https://www.amazon.com/LattePanda-4G-64GB-Development-without/dp/B01ID4HYE4/ref=sr_1_3?crid=15EY42XIK5JOX&keywords=lattepanda+4g%2F64gb&qid=1639331096&sprefix=LattePanda+4G%2F64GB+%2Caps%2C214&sr=8-3) at $169 has become a very attractive alternative.

I have successfully installed Ubuntu 20.04.3 and Chris' ROS Noetic Arlobot software on the LattePanda and since it uses a x86 processor, PropWare and PropGCC successfully built and compiled on the LattePanda.

I am now in the process of setting up my Arlobot configuration (four PING sensors, wheel encoders, RPLidar A1M8, and a Xbox One gamepad) so I can download the Propeller software to the Propeller Activity Board.

Regards,
TCIII

TCIII, 2021-12-19:
> I'm using a Windows machine to modify and load code into the activity board, when I modify the "per_robot_settings_for propeller_c_code.h" and then try to load the "ROSInterfaceForArloBot" I get [...]

@truckwiz,

Are you still working with your Arlobot?
I have had good success following Chris' guide for configuring and installing the PropWare on the PAB. I believe that you should be loading the PropWare with "install_Propeller_code.sh". It worked for me.

Please respond as I will be glad to provide any help that I can.
Regards,
TCIII

truckwiz, 2021-12-23:
I did walk away from this for awhile, I did manage to get it to compile by
declaring last_left_A and last_right_A as int's in the encoderCount.h file,
I'm really don't know if that was missed or if I messed something up by
doing that.

Thanks' for the feedback. I'll see if I can make some progress on this now.

TCIII, 2021-12-23:
@truckwiz,

Thanks for the response, much appreciated.

I have the encoders attached to the DHB-10 and used SimpleIDE with a laptop to verify that the encoders and the PINGs functioned correctly before I downloaded the ArloBot PropWare to the PAB.

I have validated the functionality of the ArloBot PropWare using the PropellerSerialTest.sh script and am presently adding components to the configuration mix. I am using a Slamtech A1M8 RPLidar and a Xbox One gamepad with the ArloBot, but have yet to enable them in personalDataForBehavior.json.

I have verified the operation of the Slamtech A1M8 RPLidar using rplidar_ros and have paired and connected the Xbox One gamepad to the Rpi 4B/4GB that I am using in place of the laptop. I adjusted the gamepad name in "find_xbox_controller.sh" since I am not using a Xbox 360 gamepad.

I also installed the "Workstation via x11docker" on a spare NVIDIA Nano 4GB running JP4.6 (Ubuntu 18.04.3) and got that working with the ArloBot.

I will be glad to help you wherever I can to get your Arlobot functional.

Regards,
TCIII

closed 2023-09-15 by chrisl8