chrisl8/ArloBot #171: flash Melodic version of Arlobot C code with Simple IDE fails with '_fdserial_rxCount'

sunglyoungKim, 2021-09-08:


chrisl8, 2021-09-09:
I haven't used SimpleIDE in a while now, because Parallax does not provide a working version for Linux. Instead this version of the system uses PropWare.

If you ran the setup script to install this, then the only thing you need to do after editing the .h files is run the script `install_Propeller_code.sh` from the scripts folder to automatically install the required code to your Propeller Activity Board.

This page should explain the use of PropWare if you want more details:  
https://ekpyroticfrood.net/?p=551

Please let me know if this helps or not.

sunglyoungKim, 2021-09-09:
Thank you for replying my questions.

May I know how I can check the actual array length and number of the array count variable?

Also, it could be a problem if I didn't update dhb_10 firmware? I am not sure current dhb_10 firmware version and whether it's the newest one or not.

Thank you

sunglyoungKim, 2021-09-10:
> I haven't used SimpleIDE in a while now, because Parallax does not provide a working version for Linux. Instead this version of the system uses PropWare.
> 
> If you ran the setup script to install this, then the only thing you need to do after editing the .h files is run the script `install_Propeller_code.sh` from the scripts folder to automatically install the required code to your Propeller Activity Board.
> 
> This page should explain the use of PropWare if you want more details:
> https://ekpyroticfrood.net/?p=551
> 
> Please let me know if this helps or not.
> 
> P.S. Be sure that you did edit `per_robot_settings_for_propeller_c_code.h` to set the specifics for your setup, and that you didn't create any errors in that code file. Based on the error, it could be that you have a mismatch between the array count variable and the actual array length?

And also, I think you are confused with noetic and melodic version. There is no install_propeller_code.sh in melodic branch. Can I use the code in melodic, too?

Thank you

sunglyoungKim, 2021-09-10:
Using PropeWare and noetic version of code works fine in Melodic. however, my Arlobot motor still doesn't move by sending signal using keyboard_teleop.launch

chrisl8, 2021-09-10:
Oh, I'm so sorry, you said Melodic, but I forgot that Noetic is the current version. These run together in my mind.

Have you tried running `~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh` from your  system that is connected to the Propeller board? It should help you debug the connection and see in real time what is happening.

sunglyoungKim, 2021-09-10:
The output of PropellerSerialTest.sh is 
(0.36xx,  0.0033249xxx, 1.0, 0, 0, 0, 0) 
The first two numbers are modified and plugged in part has been modified through the setting, too
I just realized there is 1.0 which means motor is disconnected. However, I do know the motor is connected and it worked with old version (indigo). May I know where I debug this error?

sunglyoungKim, 2021-09-14:
I didn't have permission to open the Serial port on my laptop. Now it works. Thank you!

closed 2021-09-14 by sunglyoungKim