chrisl8/ArloBot #174: test

TCIII, 2021-12-06:
test

TCIII, 2021-12-06:
Hi All,

SBC: NVIDIA Nano 4GB
OS: Jetpack 4.6/Ubuntu 20.04 (upgraded from Ubuntu 18.04)
ROS: Noetic

When I run this sequence of commands to load the Arlobot Propeller software:

cd ~/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/
mkdir bin
cd bin
rm -rf *
cmake -G "Unix Makefiles" ..
make run # Remember, make run installs it to EEPROM

And I run cmake -G "Unix Makefiles" .. I get the following: 

thomas@thomas-desktop:~/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin$ cmake -G "Unix Makefiles" ..
-- Found PropWare: /usr/share/PropWare/CMakeModules/PropellerToolchain.cmake (found version "3.0.0.224")
-- The C compiler identification is unknown
-- Check for working C compiler: /opt/parallax/bin/propeller-elf-gcc
-- Check for working C compiler: /opt/parallax/bin/propeller-elf-gcc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/opt/parallax/bin/propeller-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make cmTC_8fa6a/fast && /usr/bin/make -f CMakeFiles/cmTC_8fa6a.dir/build.make CMakeFiles/cmTC_8fa6a.dir/build
    make[1]: Entering directory '/home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_8fa6a.dir/testCCompiler.c.obj
    /opt/parallax/bin/propeller-elf-gcc    -o CMakeFiles/cmTC_8fa6a.dir/testCCompiler.c.obj   -c /home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeTmp/testCCompiler.c
    /opt/parallax/bin/propeller-elf-gcc: 1: Syntax error: Unterminated quoted string
    make[1]: *** [CMakeFiles/cmTC_8fa6a.dir/build.make:66: CMakeFiles/cmTC_8fa6a.dir/testCCompiler.c.obj] Error 2
    make[1]: Leaving directory '/home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_8fa6a/fast] Error 2

 CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!
See also "/home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeOutput.log".
See also "/home/thomas/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin/CMakeFiles/CMakeError.log".
thomas@thomas-desktop:~/catkin_ws/src/ArloBot/PropellerCodeForArloBot/ROSInterfaceForArloBot/bin$

I get the following error message:

CMakeError.log output:

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /opt/parallax/bin/propeller-elf-gcc
Build flags:
Id flags:

The output was:
2
/opt/parallax/bin/propeller-elf-gcc: 1: Syntax error: Unterminated quoted string

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /opt/parallax/bin/propeller-elf-gcc
Build flags:
Id flags: -c

The output was:
2
/opt/parallax/bin/propeller-elf-gcc: 1: Syntax error: Unterminated quoted string

I appear to have two versions of "propeller-elf-gcc" in /opt/parallax/bin:
propeller-elf-gcc
propeller-elf-gcc-4.6.1
Is it possible to use only "propeller-elf-gcc-4.6.1" which is one I think I should be using?

Any ideas as it looks like there is an punctuation issue with either "testCCompiler.c" or "CMakeCCompilerId.c"?

Regards,
TCIII

TCIII, 2021-12-07:
The Raspbian as well as the Linux version of PropGCC will not run on a Rpi 4B due to the processor not being X86.
Also the Linux version of PropGCC will not run on a Nano 4GB due the processor not being X86.
Therefore I am closing this issue.

chrisl8, 2021-12-07:
I'm working on porting this project to Raspberry Pi and ROS 2, but **the Propeller code will probably never compile on a non-x86 system**.  It will have to be built on an x86 system and uploaded to the Pi.

Commit ae670c79ccb314d9e70cf357963591e7f8f7e10e should have fixed the setup file to not even try doing the build on a Raspberry Pi if it was run there, but it hasn't been tested much yet.

Ultimately I will probably also migrate away from the Propeller board to possibly a Roboclaw.

However, this is all really low priority stuff for me right now. My Arlobot is currently in a pile on the floor as I rebuild it around a Raspberry Pi instead of a laptop.

TCIII, 2021-12-08:
@chrisl8,

I have an Ubuntu 20.04 and ROS Noetic image on a Rpi 4B 4GB and will be glad to help you with porting the ArloBot project to the Rpi. And I do have a spare Roboclaw motor driver laying around.
In the meantime I have purchased a LattePanda 4GB/64GB eMCC X86 SBC to try the ArloBot project on a SBC instead of a laptop like I was trying to do with the Rpi/Nano.

TCIII, 2021-12-08:
@chrisl8,

I have a tested Ubuntu 20.04 image for the Rpi 4B/4GB that I can send you to help speed your porting project.

However there are over 400 updates against this version of Ubuntu 20.04 and the built-in "unattended upgrade" app during the install process takes about three hours. So it takes patience and a fan cooled/heatsinked Rpi for this install.

TCIII, 2021-12-09:
@chrisl8,

I have installed Ubuntu 20.04.3 and ROS Noetic on my Windows HP Laptop (X86) and the install looks good. Setup-noetic.sh installed the PropWare and PropGCC without an issue. So I will need a little guidance here.

1) I used "find_ActivityBoard.sh" to ensure that the laptop could see the Propeller Activity Board when running your ArloBot software.
2) I assume that I should edit "per_robot_"persettings_for_propeller_c_code.h" and "personalDataForBehavior.json" as required to reflect my ArloBot hardware.
3) Since setup-noetic.sh did not install Propware on my Rpi 4B/4GB when I ran your ROS Noetic "install_Propeller_code.sh" script, should I install it manually since I believe that David Zemon's PropWare "propware_3.0.0.224-1_all.deb" works on "Linux - Debian/Ubuntu/Mint/Raspbian/etc" according to David?
4) If I can use "find_ActivityBoard.sh" to see if the Rpi 4B can see the Propeller Activity Board and it is successful, then I should be able to eventually connect with the Propeller Activity Board to use the Web Interface to control the robot?
5) I assume that the "per_robot_"persettings_for_propeller_c_code.h" and "personalDataForBehavior.json" should be the same on the Rpi 4B as was downloaded to the Propeller Activity Board by the laptop or does Web Interface only use the "personalDataForBehavior.json" configuration at startup? This is where I need your help!

Regards,
TCIII

chrisl8, 2021-12-09:
The `.h` files are **only read during the build of the Propeller code** so you only have to edit them on the system on which you are going to build (compile) the Propeller C code. After the code is compiled, they are never used again. They don't have to be copied anywhere else. Once you get that code built and the binary code uploaded to the Propeller Activity Board, you technically never need the `.h` files again.

The running code on the laptop will never see or use the `.h` files. Just the `.json` configuration files along with any `.yaml` files you may edit as well.

I hope that helps.

chrisl8, 2021-12-09:
Another option is to use the "SimpleIDE" tool from Parallax to build and install the code to the Propeller Activity Board:

https://www.parallax.com/download/propeller-1-software/

The reason I don't use it is because, the last time I checked, only the Windows version worked. The Linux version has been broken for years, although that could have changed.

It provides a GUI though and might be easier to work with if you have  Windows system to plug the Activity Board into.

chrisl8, 2021-12-09:
And to answer your questions directly:

1. Good, yes, that script is checking for the USB port that the Propeller Activity Board is on. If it works, then you know that the board is connected and that Linux can talk to it.
2. Yes, you must edit the `.h` files to populate the specifics about your robot that the Propeller Activity Board has to know about.
3. No, there is no need to put the PropWare software on the Pi if you are going to build and install the Propeller Activity Board code from another system. And once you have that code on the Activity Board, you never need PropWare again, unless of course you need to update the code. **NOTE that if you update the `.h` files, you must build and install the code onto the Propeller Activity Board again!**
4. Yes, that is correct. Before trying to get the web interface working, be sure to check out the `PropellerSerialTest.sh` script. This will run on the Pi or it should run on any Linux system, and you can use it to ensure that the Arlobot code is working on the Propeller Activity Board, and that your settings are correct without having to debug everything else first.
5. Again, no, the `.h` files are not used by the web interface or by ROS at all. Once the build is done with PropWare they are never used again.

chrisl8, 2021-12-09:
> @chrisl8,
> 
> I have a tested Ubuntu 20.04 image for the Rpi 4B/4GB that I can send you to help speed your porting project.
> 
> However there are over 400 updates against this version of Ubuntu 20.04 and the built-in "unattended upgrade" app that runs during the install process takes about three hours. So it takes patience and a fan cooled/heatsinked Rpi for this install.

Any version of Ubuntu 20.04 is fine, I presume that the latest point release, currently 20.04.3 I think) will have more of the updates "built in". That said, yes, there are often a lot of updates to install after a clean install.

Right after you install, if you run `sudo apt update;sudo apt upgrade` that should install the upgrades so that the setup will take less time.

However, the setup also installs the entirety of ROS, which probably does take a long time depending on your Internet connection.

I definitely use heat sinks and a fan on my Rpi 4's. I pretty much always buy the CanaKits now with a case, fast SD card, heat sinks and fan. Gone are the days of buying a bare Pi and using a phone charger. Without a "Class A" (App Performance) SD card and heat sinks or a fan, and a good power supply, the Rpi 4 will be throttled.

TCIII, 2021-12-10:
@chrisl8,

Thanks for detailed and informative responses, much appreciated.

The LattePanda 4GB/64GB eMMC arrived yesterday and I have it all prepped for installing Ubuntu 20.04.3 for the Atom x86 processor today. If that is successful, then I will install your ROS Noetic software and proceed from there.

I also plan to continue using my laptop to load the software to the Activity Board and then use the Rpi 4B 4GB in place of the laptop as I described above.

Regards,
TCIII

TCIII, 2021-12-11:
@chrisl8,

I managed to install a minimal Ubuntu on the LattePanda 4GB/64GB (x86) today and will load on your ArloBot ROS Noetic software tomorrow. Hopefully the LattePanda can substitute for a the laptop.

In parallel I plan to see if I can use the Rpi 4B and Nano 4GB with the Propeller Activity Board by downloading the ArloBot Activity Board software with my Laptop that is running Ubuntu MATE and ROS Noetic as I described above.

Regards,
TCIII

TCIII, 2021-12-12:
@chrisl8,

Your ROS Noetic ArloBot package installed without issue on the LattePanda yesterday.

I will not run some basic checks before attempting to use the LattePanda to install the Propeller code on the Propeller activity board.

Regards,
TCIII

TCIII, 2021-12-15:
@chrisl8,

I edited my per_robot_settings_for_propeller_c_code.h and personalDataForBehavior.json configurations and successfully used the install_Propeller_code.sh script to program the Propeller and the LattePanda.

Now to use PropellerSerialTest.sh to test the Propeller serial interface.

Regards,
TCIII

TCIII, 2021-12-17:
@chrisl8,

I had a spare Nano 4GB laying around so I installed JP4.6 (Ubuntu 18.04) on it and installed your "Workstation via x11docker" on it.
Works great.

TCIII, 2021-12-18:
@chrisl8,

I am closing this issue since it has been determined that a non x86 processor cannot be used to build PropWare and PropGCC.

However, a laptop can be used to build and install PropWare and PropGCC on the PAB and then either a Rpi 4B/4GB or a Nano 4GB running Ubuntu MATE/ROS ArloBot can be used to control the Arlobot.

closed 2021-12-18 by TCIII