chrisl8/ArloBot #177: rplidar.launch is coded specifically for the A3 RPLidar

TCIII, 2021-12-24:
I am using a Slamtech A1M8 with my Arlorobot and not the more expensive A3.

Because I am running the A1M8 in place of the A3, I encountered a number of errors, that were easy to correct, when I ran "roslaunch arlobot_ros rplidar.launch". So I compared the rplidar.launch in arlobot_ros with the rplidar.launch in rplidar_ros and made the necessary corrections to the arlobot_ros rplidar.launch:

1) My A1M8 is always on ttyUSB0 so I changed ttyUSB1 to ttyUSB0.
2) The A1M8 communicates at 115200 baud so I changed 256000 to 115200.
3) Since I am running a A1M8 instead of a A3, value="rplidar" must be changed to value="laser".

Once these corrections are incorporated into arlobot_ros rplidar.launch, the A1M8 will function correctly without throwing any pesky errors.

Regards,
TCIII

chrisl8, 2021-12-29:
Thanks, unfortunately I only own the A3, so I didn't have a way to test multi-hardware compatibility.

I'll leave this issue open until I have time to implement something that works well with both. I hope to get an A1 next year some time, but we'll see.

chrisl8, 2022-01-04:
@TCIII Actually, I want to go ahead and close this. I think this issue is already covered:

1. In the `~/.arlobot/personalDataForBehavior.json` file there is a place to set the `rplidarBaudrate`
2. When you run ROS using the shell scripts, it uses that configuration to get the baud rate as well as using a script to find and set the USB port. Both of these are set into environment variables and then used by ROS.

So I think this is already covered.

If this needs to be documented better, please suggest how it could be documented and where, or make PR for an update to the Readme.MD.

It may just be that I need to remove the suggesting to call `roslaunch` directly from the Readme.MD, because that is kind of an advanced operation and the shell scripts already do that for you.

Thoughts?

TCIII, 2022-01-05:
@chrisl8,

Points well made. I think that you can close this issue now.

closed 2022-01-05 by chrisl8