chrisl8/ArloBot #178: 'slam_toolbox::SlamToolboxPlugin' failed to load

TCIII, 2021-12-25:
@chrisl8, 

SBC: Rpi 4B/4GB
OS: Ubuntu 20.04.3/ROS Noetic

When running the roslaunch commands in "Basic TeleOp with 3D sensor use" I received this error when running "roslaunch arlobot_ros view_navigation.launch" from the Ubuntu Desktop:

rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [5102]
[ERROR] [1640451452.432491635]: PluginlibFactory: The plugin for class 'slam_toolbox::SlamToolboxPlugin' failed to load.  Error: According to the loaded plugin descriptions the class slam_toolbox::SlamToolboxPlugin with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
QXcbClipboard: Unable to receive an event from the clipboard manager in a reasonable time
[rviz-1] process has finished cleanly
log file: /home/thomas/.ros/log/0dcde9bc-65a3-11ec-bd19-0b651bddef40/rviz-1*.log
all processes on machine have died, roslaunch will exit

Rviz did open, but there was no RPLidar scan displayed on the grid.

Suggestions on how to troubleshoot this issue with the Slam Toolbox?

TCIII, 2021-12-25:
@chrisl8,

A search of "setup-noetic.sh" for the installation of the Slam Toolbox node came up empty so I assume that "setup-noetic.sh" did not install Slam Toolbox.

If so could you please provide instructions for installing Slam Toolbox on ROS Noetic as the ROS installation instructions are not too clear for a beginner like me?

Regards,
TCIII

TCIII, 2021-12-27:
@chrisl8,

I have opened an [issue](https://github.com/SteveMacenski/slam_toolbox/issues/463) on Steve Macenski's
slam_toolbox github, but he has not been of much help. 

The issue with the Slam Toolbox Plugin was the subject of [issue #234](https://github.com/SteveMacenski/slam_toolbox/pull/234) on Steve's github, but it looks like Steve's corrective action may not have cured the issue.

I would really appreciate a response from you as I have validated both my hardware and software configurations and would really like to have "view_navigation.launch" work correctly so that I can move ahead.

Regards,
TCIII

TCIII, 2021-12-28:
@chrisl8,

I am really desperate here as I have four junior college students who can't move ahead with their robot project until we get this issue put to bed. I have documented my/our efforts above, however there has been no resolution of this issue.

TCIII, 2021-12-29:
@chrisl8,

Observation:
I believe that the directory "slam_toolbox_rviz" is not being created in /opt/ros/noetic/share and populated with "rviz_plugins.xml" and "package.xml" at the very least during the build and installation of the Slam Toolbox for a start.

TCIII, 2021-12-29:
@chrisl8,
I have done the following to try and correct this issue with the Slam Toolbox Plugin:

Installed ROS Noetic, modules, dependencies, and Slam Toolbox using the setup-noetic.sh script at Chris Lofland's Arlobot Github.
Chris had some issues installing the Slam Toolbox as seen in closed issue #291 and it looks like there are still issues now.

Purged "ros-noetic-slam-toolbox" and "ros-noetic-slam-tool-box-msgs" and performed an "autoclean".

Used the following to attempt to install the noetic-devel slam toolbox:
cd ~/catkin_ws/src/
git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git
cd ..
rosdep install -q -y -r --from-paths src --ignore-src
catkin_make

However catkin_make hung at the 84% point:
[ 84%] Building CXX object slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o

^Cmake[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:63: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/slam_toolbox_common.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/build.make:63: slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/solvers/ceres_solver.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:89: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/loop_closure_assistant.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:102: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:2358: slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/all] Interrupt
make[1]: *** [CMakeFiles/Makefile2:2331: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/all] Interrupt
make: *** [Makefile:141: all] Interrupt
Traceback (most recent call last):
File "/opt/ros/noetic/bin/catkin_make", line 306, in
sys.exit(main())
File "/opt/ros/noetic/bin/catkin_make", line 249, in main
run_command(cmd, make_path)
File "/opt/ros/noetic/lib/python3/dist-packages/catkin/builder.py", line 241, in run_command
proc.wait()
File "/usr/lib/python3.8/subprocess.py", line 1083, in wait
return self._wait(timeout=timeout)
File "/usr/lib/python3.8/subprocess.py", line 1806, in _wait
(pid, sts) = self._try_wait(0)
File "/usr/lib/python3.8/subprocess.py", line 1764, in _try_wait
(pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt

Also "apt install ros-noetic-slam-toolbox" installs without any issues, but still no Slam Toolbox Plugin in the rviz1 Panel dropdown.

chrisl8, 2021-12-29:
Isn't the plugin only used by Rviz and not by the actual mapper on board? Are you trying to run Rviz on the Pi? I wouldn't do that. Run Rviz on another computer.

That might be completely off the mark though if I'm failing to understand your problem.

chrisl8, 2021-12-29:
Sorry, my robot is in a bit of a disarray at the moment

and I've been basically offline during the holiday season.

You mentioned that you got errors when running the "Basic TeleOp with 3D sensor use" commands, but which one exactly?

If this is the one that is failing:
```
# From a Terminal in the desktop (NOT over SSH):
roslaunch arlobot_ros view_navigation.launch
```

then I'm thinking that you are tryin to run that on the Pi itself, but you should run that on your x86 laptop or desktop running Linux with ROS installed. The Pi isn't going to be very happy trying to run Rviz is my guess. I cannot even try it as I am running the Ubuntu 64 bit server install on my Pi, which has no desktop environment to load Rviz in.

Finally, if you did get Rviz going and can see the output from the robot, then you are done. You don't actually need any of the Slam Toolbox plugins in Rviz to use the mapping. They are bonus features so to speak.

TCIII, 2021-12-29:
@chrisl8,

As a sanity check I did the following:

1) Did a fresh install of Ubuntu 20.04.3 on a Rpi 4B/4GB
2) In stalled ROS Noetic desktop so the install included rqt and rviz
3) Installed ros_rplidar to ensure that rviz was working and displayed the output of the RPLidar A1M8.
4)Attempted to install Slam Toolbox as follows:
   cd ~/catkin_ws/src/
   git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git
   cd ..
   rosdep install -q -y -r --from-paths src --ignore-src
   catkin_make

The build hangs at the 74% point in the same place as above: "Building CXX object slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o".

So again it looks like there is an issue with the "laser_utils.ccc.o" build?

Comments?

TCIII, 2021-12-29:
@chrisl8,

After the catkin_make build failed, I figured what the heck and ran "sudo apt install ros-noetic-slam-toolbox" which as usual installed without issue.

Then I ran "roslaunch rplidar_ros view_rplidar.launch".

The rplidar launched, followed by startup data and then rviz opened.

When rviz opened, I looked at the "Panels" dropdown menu and the Slam Toolbox Plugin was not there.

So I selected "Add New Panel" and when the "Panel Type" display opened, there was the "Slam Toolbox Plugin" under the "slam_toolbox_rviz folder.

So I selected the "Slam Toolbox Plugin" and it was added to the "Panels" dropdown menu and appeared on the left side of the rviz display.

I can turn the ""Slam Toolbox Plugin" on and off using the "Panels" dropdown menu.

I would say at this point that there is an issue with the ROS Noetic installation/packages/dependencies in the Arlobot "setup-noetic.sh script possibly?

Comments?

TCIII, 2021-12-29:
@chrisl8,

Rviz runs just fine on my Rpi 4B/4GB and displays RPLidar data just fine as described in the post above.

The problem is that your setup-noetic.sh script installs Slam Toolbox in such a way that the Slam Toolbox Rviz Plugin is not displayed in the rviz display when "roslaunch arlobot_ros view_navigation.launch" is run.

The "Slam Toolbox Plugin" is in the rviz "Panels" display dropdown menu with a check by it, but it does not display on the left side of the rviz display.

Also, in the "Add New Panel" "Panel Type" display there is no "slam_toolbox_rviz" file with the "Slam Toolbox Plugin" under it as described in my post above.

Comments?

chrisl8, 2021-12-30:
I don't use the Slam Toolbox Plugin in Rviz, and I think that you can move forward without it. It is not required for any of the mapping features. It is just a convenience GUI for digging into the guts of Slam Toolbox's operation, which I never do.

Is Rviz crashing and refusing to come up due to this error when called?

If so, then I think that if you edit [ArloBot/arlobot_ros/rviz/navigation.rviz](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/rviz/navigation.rviz) and remove lines [34 and 35](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/rviz/navigation.rviz#L34):
```
  - Class: slam_toolbox::SlamToolboxPlugin
    Name: SlamToolboxPlugin
```
and possibly also lines [641 and 642](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/rviz/navigation.rviz#L641):
```
  SlamToolboxPlugin:
    collapsed: false
```

then it shouldn't try to load the Slam Toolbox Plugin and it should load without complaining.

---

As to why it doesn't work, it works for me, so I will have to spin up a fresh install of Linux and do some experimenting to reproduce the problem and see if and how that can be fixed. Again, though, I don't actually ever use Slam Toolbox Plugin in Rviz. Perhaps the correct solution for future users (if there ever are any on this discontinued robot platform) is to disable the loading of it to avoid possible issues.

chrisl8, 2021-12-30:
P.S. The `.rviz` files are just a way of saving the current state of Rviz. Once you get Rviz set up the way you want it, you can just overwrite the settings with your own by saving under the Files menu I think it is, or just run Rviz and I believe it will load the previous settings automatically. The `navigation.rviz` file was just me trying to provide a default setup that will show all of the required bits to operate and monitor the robot.

You could also build your own from scratch. The video at https://youtu.be/k15B0RdpVYQ demonstrates what it should look like, although I do see it is showing the Slam Toolbox Plugin loaded, again, I never use it.

TCIII, 2021-12-30:
@chrisl8,

I get the same response, shown below, whether I am running your ArloBot software on my Rpi 4B/4GB or my LATTEPANDA 4GB/64GB eMMC atom quad core X86 SBC:

When running the roslaunch commands in "Basic TeleOp with 3D sensor use" I received this error when running "roslaunch arlobot_ros view_navigation.launch" from the Ubuntu Desktop:

rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [5102]
[ERROR] [1640451452.432491635]: PluginlibFactory: The plugin for class 'slam_toolbox::SlamToolboxPlugin' failed to load. Error: According to the loaded plugin descriptions the class slam_toolbox::SlamToolboxPlugin with base class type rviz::Panel does not exist. Declared types are rviz_plugin_tutorials/Teleop
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
QXcbClipboard: Unable to receive an event from the clipboard manager in a reasonable time
[rviz-1] process has finished cleanly
log file: /home/thomas/.ros/log/0dcde9bc-65a3-11ec-bd19-0b651bddef40/rviz-1*.log
all processes on machine have died, roslaunch will exit

Rviz did open, but there was no RPLidar scan displayed on the grid.

For some reason your "setu-noetic.sh" script is incorrectly installing the noetic Slam Toolbox software such that a "slam_toolbox_rviz" file with the "Slam Toolbox Plugin" under it is not created in the "Panels" dropdown menu "Add New Panel" Panel Type" display.

Your "view_navigation.launch" script which launches "/rviz/navigation.rviz" is causing the error above because the "Slam Toolbox Plugin does not exist in the rviz "plugin descriptions.xml" file due to a faulty install of the noetic Slam Toolbox software.

TCIII, 2021-12-30:
> P.S. The `.rviz` files are just a way of saving the current state of Rviz. Once you get Rviz set up the way you want it, you can just overwrite the settings with your own by saving under the Files [...]

TCIII, 2021-12-30:
@chrisl8 

> Is Rviz crashing and refusing to come up due to this error when called?

After "view_launch.rviz" threw the error, rviz launched, but would not display RPLidar data on the grid.

chrisl8, 2021-12-30:
I don't think that the Slam Toolbox Plugin errors are related to the lack of RPLidar data.

We need to troubleshoot the RPLidar data.

1. Are you running `roslaunch arlobot_ros rplidar.launch` and `roslaunch arlobot_ros view_navigation.launch` on the same system?
2. What is the terminal output when you run `roslaunch arlobot_ros rplidar.launch`?
3. What is the output of `rostopic list`?

TCIII, 2021-12-30:
@chrisl8,

Thanks for the insights and advice, much appreciated.

I will take a look at your suggestions, implement as necessary, and report back.

Regards,
TCIII

TCIII, 2021-12-30:
@chrisl8,

I did as you suggested to keep the Slam Toolbox Plugin from loading when running view_navigation.launch:
1) Removed "- Class: slam_toolbox::SlamToolboxPlugin" (line 34) from ArloBot/ros/rviz/view_navigation.rviz
2) Removed "Name: SlamToolboxPlugin" (line 35) from ArloBot/ros/rviz/view_navigation.rviz
3) Removed "SlamToolboxPlugin:" (line 641) from ArloBot/ros/rviz/view_navigation.rviz
4) Removed "collapsed: false" (line 642) from ArloBot/ros/rviz/view_navigation.rviz

Ran just "roslaunch arlobot_ros view_navigation.launch" from the Ubuntu Desktop, without the RPLidar running as a sanity check, and rviz1 opened without throwing the Slam Toolbox Plugin error. So far so good.

Then on my PC I ran the following:
1) roslaunch arlobot_ros minimal.launch
# In a new Terminal:
2) roslaunch arlobot_ros keyboard_teleop.launch
# In a new Terminal:
RPLIDAR_USB_PORT=$(find_RPLIDAR.sh) 
3) roslaunch arlobot_ros rplidar.launch
# From a Terminal in the Rpi 4B desktop (NOT over SSH):
4) roslaunch arlobot_ros view_navigation.launch

5) Set the Global Options->Fixed Frame to "odom

Expected result:
Control the Arlobot motors with the keyboard on the PC and see the RPLidar output on rviz1 on the Rpi 4B Ubuntu desktop.

Actual result:
No motor control with the PC keyboard and no RPLidar output on the rviz1 on the Rpi 4B Ubuntu desktop.
Additionally:
No "RPLidarLaserScan" appeared with a checkmark
             No "checkmark Status: Ok"

Sanity check:
Ran "roslaunch rplidar_ros view_rplidar.launch" from the Rpi 4B Ubuntu desktop

Expected result: RPLidar output displayed in rviz1on the Rpi 4B Ubuntu desktop.
     "RPLidarLaserScan" appeared with a checkmark
            checkmark Status: Ok

Actual result: RPLidar output displayed in rviz1on the Rpi 4B Ubuntu desktop
      "RPLidarLaserScan" appeared with a checkmark
            checkmark Status: Ok

My Arlobot is up on blocks and there is plenty of room in front of the PING sensors (>70cm). 
Am I expecting teleop motor control from my PC when I shouldn't?
 
I have verified motor control and PING sensor output functionality with "~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh"

How should I troubleshoot this issue with "roslaunch arlobot_ros view_navigation.launch?

TCIII, 2021-12-30:
@chrisl8,

1) "roslaunch arlobot_ros rplidar.launch" I ran from a terminal on my PC, "roslaunch arlobot_ros view_navigation.launch" was run from the Rpi 4B Ubuntu desktop.

2) Output when I ran "roslaunch arlobot_ros rplidar.launch from a terminal on my PC:

thomas@thomas-desktop:~$ roslaunch arlobot_ros rplidar.launch
... logging to /home/thomas/.ros/log/83bb6d72-698e-11ec-b8f5-e5eb48c77c0d/roslaunch-thomas-desktop-11339.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thomas-desktop.local:40625/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/scan_mode: Sensitivity
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)

auto-starting new master
process[master]: started with pid [11357]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 83bb6d72-698e-11ec-b8f5-e5eb48c77c0d
process[rosout-1]: started with pid [11377]
started core service [/rosout]
process[rplidarNode-2]: started with pid [11384]
[ INFO] [1640882150.842095434]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0
[ INFO] [1640882150.896414840]: RPLIDAR S/N: D8B2EDF9C7E29BCEA7E39EF248614304
[ INFO] [1640882150.896542393]: Firmware Ver: 1.29
[ INFO] [1640882150.896602578]: Hardware Rev: 7
[ INFO] [1640882150.947925777]: RPLidar health status : OK.
[ INFO] [1640882151.257228040]: current scan mode: Sensitivity, sample rate: 8 Khz, max_distance: 12.0 m, scan frequency:10.0 Hz,
^C[rplidarNode-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
thomas@thomas-desktop:~$

3) Output of "rostopic list":

thomas@thomas-desktop:~$ rostopic list
ERROR: Unable to communicate with master!
thomas@thomas-desktop:~$

Looks alike a communication error between rviz and the RPLidar?

TCIII, 2021-12-30:
@chrisl8,

If I run "roslaunch arlobot_ros minimal.launch" from the PC terminal and then "roslaunch arlobot_ros rplidar.launch" followed by "roslaunch arlobot_ros view_navigation.launch" in two separate terminals on the Rpi 4B Ubuntu desktop I get the following from "rostopic list: 

thomas@thomas-desktop:~$ rostopic list
/arlo_status
/arlobot_safety/safetyStatus
/arlobot_usbrelay/usbRelayStatus
/buttons
/cmd_vel
/cmd_vel_mux/input/joy
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/teleop
/cmd_vel_mux/input/web
/diagnostics
/infrared_scan
/initialpose
/joint_states
/joy_priority
/map
/map_updates
/move_base/GlobalPlanner/plan
/move_base/TebLocalPlannerROS/local_plan
/move_base/TrajectoryPlannerROS/global_plan
/move_base/TrajectoryPlannerROS/local_plan
/move_base/current_goal
/move_base/global_costmap/costmap
/move_base/global_costmap/costmap_updates
/move_base/local_costmap/costmap
/move_base/local_costmap/costmap_updates
/move_base_simple/goal
/odom
/particlecloud
/pause_navigation
/rosout
/rosout_agg
/scan
/stop_closing_loop
/tf
/tf_static
/ultrasonic_scan
thomas@thomas-desktop:~$

"RobotModel" appears in place of "RPLidarLaserScan" on the rviz1 Display so it looks like rviz is not receiver RPLIdar Laser Scan data.

Comments?

TCIII, 2021-12-30:
@chrisl8,

I decided to see what would happen if I started ROS from the Arlobot Control Panel. My Arlo is up on blocks so it will not go anywhere. The front PING sensors are over 70cm from an object and the rear PING sensor is over 70 cm from an object. I have previously used "~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh" to validate motor control and PING sensor range values.

Observations:

1) When using the Remote Control Panel with the mouse, if I push the small circle within the larger circle to the top edge of the large circle, Arlo will rotate the wheels in the forward direction for a while and then stop even though the small circle is at the top edge of the large circle. It is the same for reverse. Am I doing something wrong here with the Remote Control Panel?

2) If I put my hand within two inches in front of the forward center PING, Arlo will turn the wheels in reverse to backup and will stop the wheels when I take my hand away.

3) If I put my hand within two inches in front of the back center PING, Arlo will turn the wheels forward to go forward and will stop the wheels when I take my hand away.

4) If I put my hand within two inches of either the right or left 60 deg mounted front sensors, Arlo will not attempt to turn away either to the right or the left.

5) I did not try to do any mapping since rviz will not show an RPLidar scan data  when I run "view_navigation.launch".

Comments?

chrisl8, 2021-12-30:
I was able to start RViz on my Pi. I didn't think it would run it, but it does. Thanks for pointing out that it can work!

I did get the complaint about Slam Toolbox Plugin failing, but Rviz still launched, and I just closed the panel.

Once I set the "Fixed Frame" to "odom", as per the instructions, the RPLidar  view on the screen shows up.

So at least it can work, if only on my individual setup.

---

I noticed in the output you said it ended like this:

```
^C[rplidarNode-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

That text indicates that it is shutting down, as if as soon as you start it, it stops, which is going to just leave everything dead.

---

I wonder, have you tried opening up the IP Address of your robot's Pi at port 8080 to see if the web interface is up and working?

http://<IP ADDRESS>:8080/

For instance, on my network my robot is at http://192.168.1.116:8080/

From the command line on the Pi you can run `ip a` and somewhere in the output you should see the inet address of your Pi.

If the web interface is working, it might be simpler to start ROS form there, and either see the errors there, or if it works, then go run `view-navigation.sh` from the Pi inside of xWindows to see Rviz.

chrisl8, 2021-12-30:
My poor discombobulated robot at the moment:
[image]

Running Rviz with RPLidar data visible as red lines on the ridiculously tiny display that I have attached to the Pi on the robot:
[image]

TCIII, 2021-12-31:
@chrisl8,

Quote:
<I was able to start RViz on my Pi. I didn't think it would run it, but it does. Thanks for pointing out that it can work!

I did get the complaint about Slam Toolbox Plugin failing, but Rviz still launched, and I just closed the panel.

Once I set the "Fixed Frame" to "odom", as per the instructions, the RPLidar view on the screen shows up.

So at least it can work, if only on my individual setup.>

Repsonse:
So you can see that the Slam Toolbox Plugin doesn't install correctly when using setup-noetic.sh on the Rpi.

Are you using catkin_make to install the Slam Toolbox or "sudo apt install ros-noetic-slam-toolbox"?

I tried to use catkin_make today and got all the way to 90% before it hung on a build statement. It appears to get into a wait state that it never comes out of. I think that part of the problem is that "rosdep install -q -y -r --from-paths src --ignore-src" is not installing all of the dependencies necessary to build the Slam Toolbox with catkin_make.

Comments?

TCIII, 2021-12-31:
@chrisl8,

I have been able to determine why I can't see the RPLidar A1M8 laser display in rviz (ref [open issue #177](https://github.com/chrisl8/ArloBot/issues/177).

Your A3 rplidar.launch:

<arg name="usb_port" value="$(optenv RPLIDAR_USB_PORT /dev/ttyUSB1)"/>
<arg name="baudrate" value="$(optenv RPLIDAR_BAUDRATE 256000)"/>
<node name="rplidarNode" pkg="rplidar_ros" type="rplidarNode" output="screen">
            <param name="frame_id" type="string" value="rplidar"/>

My A1M8 rplidar.launch: 
<arg name="usb_port" value="$(optenv RPLIDAR_USB_PORT /dev/ttyUSB0)"/>
<arg name="baudrate" value="$(optenv RPLIDAR_BAUDRATE 115200)"/>
<node name="rplidarNode" pkg="rplidar_ros" type="rplidarNode" output="screen">
            <param name="frame_id" type="string" value="laser"/>

SlamTech's A1M8 rplidar.launch which runs fine with rviz:
<node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>
  <param name="serial_baudrate"     type="int"    value="115200"/>
  
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  
  You will notice that the ""frame_id" type="string" value="laser" in both my rplidar.launch and the SlameTech rplidar.launch compared to your ""frame_id"            type="string" value="rplidar"

chrisl8, 2021-12-31:
I use the `sudo apt install` method for installing Slam Toolbox, although I have also used catkin_make on x86 I have not tried it on Pi. My guess is that it would take ages to build on the Pi, but again, I haven't tried.

I don't have any further comments on the Rviz Slam Toolbox Plugin at the moment. I don't use it myself.

---

So if you **were** able to see RPLidar data in Rviz at any point, then it appears everything is working.

I'm not sure why the web interface isn't available to you. That is the goal, to just be able to start/stop things from there.

What you should do next is:

1. Open up and edit `~/.arlobot/personalDataForBehavior.json` and go all of the way to the bottom. Make sure these lines look like this:
```
"hasASUSXtion": false,
"hasKinect": false,
"active3dCamera": "None",
"hasXV11": false,
 "hasScanseSweep": false,
"hasRPLIDAR": true,
"rplidarBaudrate": 115200,
"scanTopicSource": "rplidar",
"hasBeenEdited": true
```

That should make sure that everything is set up to use the RPLidar.

2. Run the entire ROS Robot by running this on the Pi and leaving it running:
` ~/catkin_ws/src/ArloBot/scripts/start-robot.sh`

(This is the equivalent to using the "Start ROS" button in the web interface.)

3. Assuming that seems to be working, then start a map making session by running this in another terminal and leaving it running:
`~/catkin_ws/src/ArloBot/scripts/make-map.sh`

(This is the equivalent of starting a map in the web interface.)

4. Start up Rviz to view everything. This should work:
`~/catkin_ws/src/ArloBot/scripts/view-navigation.sh`

If it complains about Slam Toolbox Plugin, ignore that.

Now you should have a fully functioning SLAM operation started. You can hopefully see the room starting to take shape.

You can set destination points for the robot to go to within the visible area, and it should go to them and start building the map as it goes.

If you get a map done, then in another terminal you can run this to save the map:

`~/catkin_ws/src/ArloBot/scripts/save-map.sh my-new-map`

TCIII, 2021-12-31:
@chrisl8,

When I run the following on the Rpi Ubuntu desktop terminals I am not getting Lidar scan data in the rviz display window even after changing "map" to "odom":

1) roslaunch arlobot_ros minimal.launch
2) roslaunch arlobot_ros rplidar.launch
3) roslaunch arlobot_ros view_navigation.launch

I do get the red/green axes displayed in the center of the grid, but no laser scan data.

Fixed Frame odom
Scan
   Status: Error
    Topics
    Points
    Transform [sender=unknown_publisher] For frame [rplidar]:Frame[rplidar] does not exist/scan

When I run the following on the Rpi Ubuntu desktop terminals I am getting Lidar scan data in the rviz display window when I change "map" to "rplidar":

1) roslaunch arlobot_ros minimal.launch
2) roslaunch arlobot_ros rplidar.launch
3) roslaunch arlobot_ros view_navigation.launch

I get a white dot displayed in the center of the grid along with the laser data displayed around it.

Fixed Frame rplidar
Scan
   Status: Ok
   Points
   Transform [sender=unknown_publisher]
          Transform Ok
           Topic xxxx messages received   (xxxx =increasing messages received)

Comments?

TCIII, 2021-12-31:
@chrisl8,

> I wonder, have you tried opening up the IP Address of your robot's Pi at port 8080 to see if the web interface is up and working?
> 
> http://:8080/

Yes I did exercise the web interface.

Please take a look at the post seven posts above this post.

TCIII, 2022-01-01:
@chrisl8,

I have moved the issue where the view_navigation. launch will not display the Slamtech A1M8 RPLIdar output on rviz to issue [#179](https://github.com/chrisl8/ArloBot/issues/179) because it is not germane to the Slam Toolbox Plugin not loading on rviz.

Since you now have your Arlobot software installed on a Rpi SBC, do you plan to identify the reason as to why the Slam Toolbox Plugin will not load on rviz or do you consider this a dead issue?

chrisl8, 2022-01-01:
@TCIII Sorry, I think I'm getting mixed up and out of sync with the messages.

This has turned into more of a "working session" than a bug report, which is fine, but we might need to take it to another platform. Even maybe do a Zoom call or something?

Because everyone's hardware is unique, and because the dependencies change daily,  this project will always require a fair amount of "tweaking" by anyone using it.

I want to ensure that you can use it to accomplish the basic tasks demonstrated in the Youtube video:
1. Starting ROS with all hardware enabled.
2. Starting a SLAM mapping session.
3. Visualizing that mapping session in Rviz either locally or remotely.
4. Building a map in Rviz using waypoins.
5. Saving that map and reloading it and using it again.

If that works, then I would consider everything 100% functional and no need to chase down random errors or missing pieces that aren't required.

TCIII, 2022-01-02:
@chrisI8,

I have already run most of what you have suggested above after watching the
video, but have had mixed results:

 A couple of days ago, before I corrected the Arlobot urdf configuration, I
decided to see what would happen if I started ROS from the Arlobot web
interface. My Arlo is up on blocks so it will not go anywhere. The front
PING sensors are over 70cm from an object and the rear PING sensor is over
70 cm from an object. I have previously used
"~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh" to validate motor
control and PING sensor range values.

I made "Unplugged" True, but maybe I should have turned off monitoring the
PING sensors for this test?

Observations:

   1.

   When using the Remote Control Panel with the mouse, if I push the small
   circle within the larger circle to the top edge of the large circle, Arlo
   will rotate the wheels in the forward direction for a while and then stop
   even though the small circle is at the top edge of the large circle. It is
   the same for reverse. Am I doing something wrong here with the Remote
   Control Panel?
   2.

   If I put my hand within two inches in front of the forward center PING,
   Arlo will turn the wheels in reverse to back up and will stop the wheels
   when I take my hand away.
   3.

   If I put my hand within two inches in front of the back center PING,
   Arlo will turn the wheels forward to go forward and will stop the wheels
   when I take my hand away.
   4.

   If I put my hand within two inches of either the right or left 60 deg
   mounted front sensors, Arlo will not attempt to turn away either to the
   right or the left.
   5.

   I did not try to do any mapping since rviz will not show an RPLidar scan
   data when I run "view_navigation.launch".

Comments?

TCIII, 2022-01-02:
@chrisI8,

I usually use Google Meet to do face to face.

Since I am retired I have very flexible hours.

Regards,
Tom C

TCIII, 2022-01-03:
@chrisl8,

I just received this from Steve Macenski over at the Slam Toolbox github:

"Did you install the rviz package? ros-noetic-slam-toolbox-rviz

I just did a clean install of noetic on my Ubuntu 20.04 machine, installed slam toolbox and all of the related packages and opened rviz and works fine."

My response was:

"Thanks for the response, much appreciated.

Interesting to say the least.

Chris Lofland over at Arlobot has seen the same issue as me when installing Slam Toolbox on Ubuntu 20.04 running ROS Noetic.

However I think that the issue has been that we both have been using "apt install ros-noetic-slam-toolbox" and not "apt install ros-noetic-slam-toolbox-rviz" as you did.

Comments?"

This may be the solution to the Slam Toolbox Plugin issue.

I will give it a try tomorrow to see if it works.

chrisl8, 2022-01-03:
That makes sense. The Slam Toolbox package is installed automatically because the ROS Arlobot package depends on it, but there is no dependency on the `-rviz` package.

I should probably update the setup script to just install both ros-noetic-slam-toolbox and ros-noetic-slam-toolbox-rviz directly.
I need to experiment a bit to ensure that doesn't cause any unintended consequences first though.

Leave this issue open and I'll work on the setup at some point.

Of course you may be the last person to ever run it too, so it might not matter.

TCIII, 2022-01-04:
@chrisl8,

I don't recommend installing "ros-noetic-slam-toolbox-rviz" as it causes a "terminate called without an active exception" error when rviz is **quit** whether or not the Slam Toolbox Plugin is terminated or not before quitting rviz.

If you really don't need the Slam Toolbox Plugin I suggest only installing "ros-noetic-slam-toolbox" and take the references to the Slam Toolbox out of the "navigation.rviz" to prevent the Slam Toolbox Plugin error from occurring.

chrisl8, 2022-01-04:
@TCIII 

Ok, fair enough. Here I was thinking that you were greatly in need of it working.

To be fair, getting errors when you open and/or close Rviz is pretty normal. It is very chatty. As long as it works, I'm happy.

Leave this open though and I will revert bde7aa14a4edbf84439e15580f354c579c75b678 and remove the reference to the Plugin from the .rviz file instead later on.

Thank you.

chrisl8, 2022-01-04:
@TCIII 

This should be fixed now. If you want to test it and let me know that would be appreciated.

However, if you have already edited the `.rviz` file then you won't be able to pull the new one.

If you did edit it and still want to wipe your changes to test these changes do this:

First, if you did install `ros-noetic-slam-toolbox-rviz` or even think it might be installed, remove it with this command:

```
sudo apt remove ros-noetic-slam-toolbox-rviz
```

Next pull down my updated copy of everything:

```
cd ~/catkin_ws/src/ArloBot
git pull
```

You will get an error like this:
```
error: Your local changes to the following files would be overwritten by merge:
        arlobot_ros/rviz/navigation.rviz
```

If so, then run this to clear your changes and try the pull again:
```
git checkout -- arlobot_ros/rviz/navigation.rviz
git pull
```

Now you should have my copy of the navigation.rviz file and you can see if `~/catkin_ws/src/ArloBot/scripts/view-navigation.sh` works as expected without errors.

closed 2022-01-05 by chrisl8