chrisl8/ArloBot #179: view_navigation will not display Slamtech A1M8 RPLidar output on rviz

TCIII, 2022-01-01:
@chrisl8,

When I run the following on the Rpi Ubuntu desktop terminals I am not getting Lidar scan data in the rviz display window even after changing "map" to "odom":

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch

I do get the red/green axes displayed in the center of the grid, but no Laser scan data is displayed.

Fixed Frame odom
Scan
Status: Error
Topics
Points
Transform [sender=unknown_publisher] For frame [rplidar] : Frame[rplidar] does not exist/scan

When I run the following on the Rpi Ubuntu desktop terminals I am getting Lidar scan data in the rviz display window when I change "map" to "rplidar":

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch
I get a white dot displayed in the center of the grid along with the laser data displayed around it.

Fixed Frame rplidar
Scan
Status: Ok
Points
Transform [sender=unknown_publisher]
Transform Ok
Topic xxxx messages received (xxxx =increasing messages received)

Comments?

TCIII, 2022-01-01:
@chrisl8,

I have run "roslaunch rplidar_ros view_rplidar.launch" from the Rpi 4B Ubuntu desktop

Expected result: RPLidar output displayed in rviz1on the Rpi 4B Ubuntu desktop.
"RPLidarLaserScan" appeared with a checkmark
checkmark Status: Ok

Actual result: RPLidar output displayed in rviz1on the Rpi 4B Ubuntu desktop
"RPLidarLaserScan" appeared with a checkmark
checkmark Status: Ok 

Therefore there must be an issue with "view_navigation. launch" linking to rplidar and displaying the RPLidar scan output?

TCIII, 2022-01-01:
@chrisl8,

I reran "bash <(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)" last night before shutting down for the evening.

This morning I ran the following on the Rpi 4B desktop using separate terminals:

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch

Changed "map" to "odom".

Observed result:

I still get the red/green axes displayed in the center of the rviz grid, but no Laser scan data is displayed around it.

Both the RobotModel and the TF are blue now instead of red. 
However, there is no rplidar transform displayed in either the RobotModel or the TF Status dropdowns, but both Status show "Ok".

The Scan is blue also, the Status is "Ok", Points has a green check, and Topic is /scan.

The rplidar transform appears to be missing for some reason?

TCIII, 2022-01-01:
@chrisl8,

I believe that I have discovered why I am having the issue with the RPLidar data not appearing in the rviz display and why there is no "rplidar" link or tf.

Like you, I have my Slamtech RPLidar on the top (second) deck, but I believe that the common_default.urdf.xarco presently does only "includes" "arlo.urdf.xarco" and not "arlo_2stack.urdf.xarco."

I wish that changing to "arlo_2stack.urdf.xarco." cured the problem, but unfortunately it didn't and here is why.

On my Rpi 4B when I bootup, Ubuntu assigns the A1M8 RPLidar to USB0 and the PAB to USB1.
This USB assignment works fine with "arlo.urdf.xarco", but when I change to

chrisl8, 2022-01-01:
If you run things using the `~/catkin_ws/src/ArloBot/scripts/start-robot.sh` script, it does exactly that, it auto-detects the locations of the devices before starting the ROS pieces.

If you call ros directly with `roslaunch ...` then you are bypassing the auto-detection and just uses the default settings in the files.

chrisl8, 2022-01-01:
It **does** make sense to me that you might have to switch which xacro file is used. In theory every robot has to have its own version of that file, but the "generic" one is a starting place.

You may have to edit the .launch files to specify a particular xacro file to load, or edit and copy a working one over the existing one that it always loads.

I'm confused about how the USB port and the xacro file could be related. I wasn't aware that there is any connection between the xacro files and the USB ports.

TCIII, 2022-01-01:
@chrisl8,

Thanks for the quick response, much appreciated.

I changed "arlo.urdf.xarco" to "arlo_2stack.urdf.xarco" and then changed "port = rospy.get_param("~port", "/dev/ttyUSB0")" to USB1 in "propellerbot_node.py".

I then ran the following on terminals on the Ubuntu desktop:

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch

And finally got rviz to display the RPLidar scan data when I switched from "map" to "odom".

So "roslaunch arlobot_ros minimal.launch" will not autodetect the USB ports for the RPLidar and the PAB?

So I need to run "~/catkin_ws/src/ArloBot/scripts/start-robot.sh" before running the following to avoid a USB conflict:

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch

Comment?

TCIII, 2022-01-01:
@chrisl8,

I am closing this issue as it turned out I needed to use the correct "urdf.xacro" (arlo_2stack.urdf.xacro) in common_default.urdf.xacro for my Arlobot two stack configuration.

closed 2022-01-01 by TCIII

chrisl8, 2022-01-02:
The shell script detects the device locations and then sets variables and then calls the ROS launch files for you. If you run it, then you don't have to run the launch files individually.

What you are changing is the defaults that are used when the variable are not set.

The Readme probably needs some improvement to make that clear.

TCIII, 2022-01-02:
@chrisI8,

I had to change the following in the Arlobot software:

1) In "ArloBot/blob/noetic/arlobot_ros/scripts/propellerbot_node.py" I had
to change "port = rospy.get_param("~port", "/dev/ttyUSB0")" USB0 to USB1
because the Rpi consistently puts the PAB on USB1.

2) In "ArloBot/blob/noetic/arlobot_ros/launch/rplidar.launch" I had to
change USB1 to USB0 because the Rpi consistently puts the RPLidar on USB0.

To run the following:

roslaunch arlobot_ros minimal.launch
roslaunch arlobot_ros rplidar.launch
roslaunch arlobot_ros view_navigation.launch

However, other SBCs might put the PAB on USB0 and the RPLidar on USB1. I
will have a look at what the LattePanda x86 does.