chrisl8/ArloBot #180: Moving "Unplug" to Yes does not result in "Plugged In" changing from Yes to No

TCIII, 2022-01-02:
@chrisl8,

**Required Info:**

SBC: Rpi 4B/4GB
OS: Ubuntu 20.04.1
ROS: Noetic
Sensors: Four PING sensors: three forward, one back
RPLidar: Slamtech A1M8

**Steps to reproduce issue:**

After successfully starting ROS using the Arlobot web interface the following occurs when the "Unplug" is moved from No to Yes:

**Expected behavior:**

"Plugged In" changing from Yes to No

**Actual behavior:**

"Plugged In" never changes from Yes to No

 **Additional information:**

The Arlobot PAB to HDB-10 interface was validated using the Parallax SimpleIDE and Arlo C library.

The Rpi to PAB interface was validated using "PropellerSerialTest.sh" at which time using "Settings" I was able to toggle Settings: "plugged In: from "No" to "Yes" and back to "No" using "p - toggle Plugged In".

How should I go about troubleshooting the Arlobot web interface "Unplug" issue?

Is there something I have overlooked?

chrisl8, 2022-01-03:
The switch here:
[image]

is meant to cause the robot to automatically unplug itself. It relies on the output of the laptop power settings to tell it if/when the laptop is unplugged.

If you are using a device other than a standard laptop, I would not expect this to work. It certainly won't work on a Pi, as a Pi cannot be used when it isn't plugged in. It must be connected to power.

If you edit `~/.arlobot/arlobot.yaml` and set the line about monitoring AC connection tot false like this:  
`monitorACconnection: False`

then it should ignore the plugged in status and allow the robot to move.

TCIII, 2022-01-03:
@chrisl8,

Thanks for the quick response, much appreciated.

In the "~/.arlobot/arlobot.yaml" the "monitorACconnection:" default value is "**False**" and appears that way in the Arlobot web interface Sensors dropdown panel: "AC" is set to **ignore.**

If I attempt to use the Remote Control dropdown panel to move Arlobot forward, with the mouse pointer, the following occurs:

Moving the small circle to the top of the large circle causes the robot wheels to turn forward for about 5 seconds and then stop. 
I have to repeatedly keep moving the small circle to the top of the large circle to get the robot wheels moving again every time the wheels stop after about 5 seconds.

The "Plugged In" message at the top of the gray panel area says "**Yes.**"

It seems like the robot wheels are turning just far enough to "**Unplug**" as shown in your video?

Comments?

chrisl8, 2022-01-03:
That is strange, and should not happen.  I don't think that it is related to the unplugging function or the plugged in status.

If it moves at all, then it is ignoring the plugged in state.

The self-unplug function doesn't restrict movement, it operates on its own, so I do not think that it is related to this issue.

As a debugging operation, if you leave everything going as it is, the web site up and ROS running,
and then go to a terminal on the robot's computer and run:
`~/catkin_ws/src/ArloBot/scripts/keyboard-teleop.sh`
then can you drive the robot via keyboard? Or does it do the same thing?

TCIII, 2022-01-03:
@chrisl8' ,

The keyboard-teleop.sh works as it should with the robot wheels going forward as long as I held the "**i**" key down continuously and backwards as long as I held the "**,**" key down continuously.
The other keys in the teleop layout functioned as expected.

The robot wheels will turn slowly backwards if I put my hand within two inches of the front facing center PING sensor and will go slowly forward if I put my hand in front of the rear facing center PING sensor. 
However, if I put my hand in front of either of the two right and left 60 deg front PING sensors, the wheels do not turn the robot away.

So it looks like there is an issue with the Arlobot web controller in relation to motion control?

Next step?

chrisl8, 2022-01-03:
Yes, as you said, your debugging has proven that everything on the robot itself is working perfectly. There is an issue with the web interface code apparently.

I'm still working on getting my robot working well enough to be able to test that feature. Hopefully this evening I can get it there.

The only quick ideas I have is to experiment with how you are using the "joystick". You have to hold your mouse in the location. If you move too far, it will snap away. So when you are moving it, it should always show the small dot under your mouse cursor to indicate that it is still operating.
That probably isn't helpful, but just in case.

TCIII, 2022-01-03:
@chrisl8, 

> The only quick ideas I have is to experiment with how you are using the "joystick". You have to hold your mouse in the location. If you move too far, it will snap away. So when you are moving it, it should always show the small dot under your mouse cursor to indicate that it is still operating.

Okay, I gave your suggestion a shot:

**Observations:**

1) I left clicked on the gray panel a small white circle appeared under the mouse point inside of a larger light gray circle. The Linear Speed displayed **0**
2) I moved the small circle slowly towards the top of the larger gray circle with the mouse pointer. The robot wheels started to move forward and ran for ~5 seconds and then stopped.  The Linear Speed displayed **0.26** even when the robot wheels stopped moving.
3) The robot wheels act the same way when I move the  the small circle slowly towards the bottom of the larger gray circle with the mouse pointer. The robot wheels started to move backwards and ran for ~5 seconds and then stopped.  The Linear Speed displayed **0.26** even when the robot wheels stopped moving.

TCIII, 2022-01-03:
@chrisl8,

I am going to open a [new issue](https://github.com/chrisl8/ArloBot/issues/181) concerning the web interface motor control issue as I think that we can close out the "Plug In" issue.

closed 2022-01-03 by TCIII