chrisl8/ArloBot #187: Slam Toolbox fails to launch either manually or when using the Arlobot Web Controller Navigation Panel

TCIII, 2022-01-11:
@chrisl8,

SBC: Rpi 4B/4GB
OS: Ubuntu 20.04
ROS: Noetic

When I attempted to use the Arlobot Web Controller Navigation Panel to make a map it hung on the statement "launching Slam Toolbox".

I ran  "bash <(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)" to make sure that I had the latest updates.

I then tried to launch the Slam Toolbox manually by doing the following:

start-robot.sh
# In a new Terminal:
roslaunch arlobot_ros slam_toolbox.launch

**Expected Behavior:**

The Slam Toolbox to launch without error.

**Observed Behavior:**

NODES
  /
    move_base (move_base/move_base)
    slam_toolbox (slam_toolbox/async_slam_toolbox_node)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [slam_toolbox/async_slam_toolbox_node]: Cannot locate node of type [async_slam_toolbox_node] in package [slam_toolbox]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[move_base-2]: started with pid [33833]
[ WARN] [1641936916.995136487]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100661 timeout was 0.1.
[ WARN] [1641936922.030140981]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100514 timeout was 0.1.

How should I troubleshoot this issue?

Am I missing something in my urdf configuration?

Regards,
TCIII

chrisl8, 2022-01-12:
In a terminal run:
`start-robot.sh`
(NOTE: This is the same as pressing the "Start ROS" button on the web site, but from the terminal we can see all of the output and possibly errors.)

Assuming it seems to be working, then in another terminal run:
`make-map.sh`
(NOTE: This is the same as pressing the "Make Map - Slam Toolbox"  button on the web site, but from the terminal we can see all of the output and possibly errors.)

My guess is that once `make-map.sh` starts, you will see some errors. Let me know what you see.

(FYI: I'm also updating the readme to reference these shell scripts instead of **some of** the direct calls to `roslaunch` , since the shell scripts do things like find your RPLIDAR USB port, and such.)

TCIII, 2022-01-12:
@chrisl8,

Since you are using rviz with the Slam Toolbox, it might be that, even though we are not using the rviz Slam Toolbox Plugin, that you have to install ros-noetic-slam-toolbox-rviz to setup a path to the Slam Toolbox in the noetic/lib/slam_toolbox directory?

TCIII

TCIII, 2022-01-12:
> Please believe me that you *DO NOT NEED OR WANT* the Slam Toolbox Rviz
> plugin. I don't know how to make this clear. It is NOT required to make
> maps with Slam Toolbox. I don't use it and have never used it.
>
> Please follow the instructions above and let me know the results.

Hi Chris,

Will do.

Tom C

chrisl8, 2022-01-12:
I just ran `bash <(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)` on a Pi with a freshly installed copy of Ubuntu last night, and it is working now, or at least getting further than that.

If you want to you can try running `sudo apt install ros-noetic-slam-toolbox` it should just say that it is already installed. If it doesn't, then something is broken with your install and I might need to look at the output of the setup-noetic.sh script to tell you what.

chrisl8, 2022-01-12:
Oh I just thought of something, it could be that we have some conflicts in your clone of the repository causing it not to pull updates anymore.

```
cd ~/catkin_ws/src/ArloBot
git pull
```

Does it say "Already up to date." or does it complain about conflicting files?

TCIII, 2022-01-12:
@chrisl8,

thomas@thomas-desktop:~$ cd ~/catkin_ws/src/ArloBot
thomas@thomas-desktop:~/catkin_ws/src/ArloBot$ git pull
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 6 (delta 4), reused 4 (delta 2), pack-reused 0
Unpacking objects: 100% (6/6), 799 bytes | 14.00 KiB/s, done.
From https://github.com/chrisl8/ArloBot
   f16dfa9..ab461bc  noetic     -> origin/noetic
Updating f16dfa9..ab461bc
Fast-forward
 README.md | 11 +++++------
 1 file changed, 5 insertions(+), 6 deletions(-)
thomas@thomas-desktop:~/catkin_ws/src/ArloBot$

Tom X

chrisl8, 2022-01-12:
@TCIII 

When you ran `sudo apt install ros-noetic-slam-toolbox` did it install it or say it was already installed?

chrisl8, 2022-01-12:
Now I'm really lost, so when you run `start-robot.sh` it works fine right?

And then when you run `make-map.sh` in another terminal on the same system it complains about slam_toolbox missing?

TCIII, 2022-01-12:
@chrisl8,

Start-robot.sh works fine every time I run it.

For some reason "async_slam_toolbox_node", which is in the noetic/lib/slam_toolbox directory, is not on a path that the launch function can find. There is a slam_toolbox directory in both noetic/lib/ and in noetic/share.

Tom C

TCIII, 2022-01-12:
@chrisl8,

Should I do a fresh install?

Tom C

chrisl8, 2022-01-12:
Two ideas:
1. Wipe your setup and install Ubuntu from scratch and start over, which I realize might be a pain.

2. We can try to clean up your system to see if we can work back to what went wrong. If you want to do that, start by running these:
```
sudo apt remove ros-noetic-slam-toolbox
sudo apt remove ros-noetic-slam-toolbox-msgs
sudo apt remove ros-noetic-slam-toolbox-rviz
cd ~/catkin_ws
rm -rf build
cd ~/catkin_ws/src
ls
```
You should only see these folders, remove anything else:
`ArloBot  CMakeLists.txt  rplidar_ros  teb_local_planner_tutorials`

Then `cd` and run the bash setup script again.

TCIII, 2022-01-12:
@chrisl8,

Okay, I will give it a shot before doing a fresh install.

Tom C

TCIII, 2022-01-12:
@chrisl8,

Looks like I am going to have to do a clean install as I am getting the same message about not being able to launch "async_slam_toolbox_node" after doing your suggested remove and install.

Tom C

chrisl8, 2022-01-12:
@TCIII Sorry about that. Hopefully the clean install doesn't take too long for you. As I've been working on this for over 5 years, I've got wiping and setting up from scratch down to a pretty quick process. It is the only way to make sure it works for new systems.

I'll keep my fingers crossed that it fixes it. If not, then **I** will have to wipe mine again and install from scratch and see if I can reproduce your scenario. I set mine up literally yesterday though, so I hope my environment is pristine.

chrisl8, 2022-01-12:
@TCIII  If it isn't too late, don't forget to keep a copy of  `~/.arlobot/personalDataForBehavior.json` to reuse.

TCIII, 2022-01-12:
@chrisl8, 

> If it isn't too late, don't forget to keep a copy of `~/.arlobot/personalDataForBehavior.json` to reuse.

A good suggestion and I always keep an edited copy of personalDataForBehavior.json and per_robot_settings_for_propeller_c_code.h on my PC so I can use Windows PowerShell to scp into the Rpi to quickly replace the default files as necessary.

Tom C

TCIII, 2022-01-12:
@chrisl8,

Before I started setting up the Ubuntu 20.04 install on the Rpi, I hooked up the LattePanda to the PAB and the RPLidar to see if it could get though the "async_slam_toolbox_node" issue.

I first tried making a map with the SBC Web Controller Navigation Panel and Slam Toolbox launched just fine except that rviz did not launch. So I tried the map-make.sh route.

Map-make.sh got through the "async_slam_toolbox_node" launch, but hung at INFO receiving odom.

The Rpi Ubuntu install is going faster than I expected and I should be at the point of the ArloBot/ROS software install pretty soon.

Tom C

chrisl8, 2022-01-12:
@TCIII 

I'm 99% sure that "but hung at INFO receiving odom." is exactly here you want it to be. At this point, mapping is running. The easiest way to tell for sure is to use Rviz by running `view-navigation.sh`.

TCIII, 2022-01-12:
@chrisl8

Why didn't rviz launch when using the SBC Web Controller Navigation Panel to make a map?

chrisl8, 2022-01-12:
There is no way to launch Rviz from a web browser, sorry. You have to launch Rviz by hand on whatever system you want to run it on.

I guess I could add that information to the web site somewhere.  In the past most people who use this have spent months already learning about ROS and know the ins and outs, so my "beta testers" never asked that before.

TCIII, 2022-01-12:
@chrisl8,

Where did you get your Ubuntu 20.04 (focal) for the Rpi?

I am using what is called Ubuntu MATE 20.04 for the Rpi and that could be a problem?

chrisl8, 2022-01-12:
@TCIII 

I think Ubuntu MATE should work fine. You proved that things worked well, just that weird issue with slam_toolbox somehow getting installed in ROS without adding itself to the ROS "path" even though dozens of other ROS components worked fine.

That said, I'm following these instructions to install Ubuntu:
https://ubuntu.com/tutorials/how-to-install-ubuntu-on-your-raspberry-pi#1-overview

The problem with those instructions is that the image that the Raspberry Pi imager has access to does NOT contain an **20.04** image **with Desktop** it only has the "Server" version. So if you want to have a GUI on that install, after you get it running you have to run `sudo apt-get install ubuntu-desktop` to get it to install the XWindows system.

Again, I don't see why your "MATE" system wouldn't work fine, especially since you got so far with it.

It is up to you what route you want to take though.

TCIII, 2022-01-12:
@chrisl8,

Okay, the Arlobot/Noetic install completed successfully.

I will now modify personalDataForBehavior.json and see what will happen.

TCIII, 2022-01-12:
@chrisl8,

Okay, a fresh install corrected the "async_slam_toolbox_node" issue as I got to INFO odom received when running make-map.sh
I opened a third terminal on the SBC desktop and ran view-navigation.sh and rviz opened and I set Fixe Frame to odom, but RPLidar scan is being displayed and when I put my hand near a PING sensor a blue dot appears. I can see the room scan in red outside of a rectangular box where the red/green center resides, but there a light blue blobs within the rectangle.

Time for dinner. I will experiment more. Suggestions?

chrisl8, 2022-01-12:
@TCIII 

Great, now switch your fixed from to **map** and you should see the map that is starting to be created. I suggest going back to the Youtube video https://www.youtube.com/watch?v=k15B0RdpVYQ and watching again and following along once it gets to the part in Rviz, which should be the same for you now.

If the blue and green dots don't show in the correct places, it is probably an issue with the URDF file or possibly my code isn't resilient to all of the variations available. That said, the PING and IR sensors, while displayed in Rviz, are purely there for information, ROS doesn't "see" them and they don't affect the map. If they are cluttering the display, click the check boxes next to them on the left side to turn them off.
You can try flipping various elements on/off on the left side to see what is what and also to customize your display down to what you want to see. Just don't turn off important things like map, or the local and global cost map.

The only data source that creates the map is the "lidar" scanner, the red line.

TCIII, 2022-01-12:
@chrisl8,

Thanks for the guidance, much appreciated.

I will give it a shot tomorrow.

One nice thing is that the latest software on the RPLidar A1M8 scans at 8kHz instead of the older 4kHz. I have a older, around three years old, A1M8 with the older software and it scans at 4kHz while the two newer A1M8s that I have with the newer software scan at 8kHz. I have the latest version of the A1M8 on my ArloBot.

An issue with the MATE version of Ubuntu 20.40:

Even though I have installed vino on the Ubuntu OS, there is no "Screen Share" in the System Menu so there is no way to enable Ubuntu MATE to work with VNC from my PC. The LattePanda has the "Screen Share" in the System Menu and works fine with my PC. Ubuntu MATE articles on the IoT are packed with how to use vino for VNC on the Ubuntu MATE, but so far I have struck out in getting vino to work with Ubuntu MATE.

chrisl8, 2022-01-12:
I am keen to see how this all works out with the A1M8. It **looks** a lot like an old lidar I have that worked terribly, so I was reticent to buy one. At close to $100 instead of like $300, I should probably give one a try. I'll wait to see some images of your maps though.

If we intend to branch this code out and make many robot models that use it though, I'll need a cheap lidar like the A1M8 to be able to mount on on everything I build!

TCIII, 2022-01-13:
@chrisl8,

It turns out that "screen share" is not available on Ubuntu MATE desktop which not gnome.

I might have to wipe the SD card and install Ubuntu server for the Rpi and then install the desktop as you have suggested.

Do I really need VNC to make a map from my PC?

chrisl8, 2022-01-13:
@TCIII 

Wow, I do not remember paying $600 for the A3, but I probably got it on sale and/or with some gift card or something. It has been some years now.
Donkey Car sounds pretty amazing, I should look into it more.

>"screen share" is not available on Ubuntu MATE desktop which is not gnome.

I'm surprised there isn't any other option, like RDP or something? I'm not a MATE user though, so I don't know much about it

> Do I really need VNC to make a map from my PC?

No, but it might be the easiest way.

Rviz does exist for Windows, but I've never tried it. If you are willing to wait a few days I might be able to write up some instructions, or you can start reading at http://wiki.ros.org/Installation/Windows and do some web searches. The problem is you are messing with your desktop system, so if you mess it up, it might not be so easy to recover by wiping it and starting over. Also there are several methods of getting ROS/Rviz onto Windows, so it can be difficult to know which is best and what trouble shooting steps apply to which install type.

Personally I run Rviz on another Linux host, but I have a lot of computers around. I don't like to run it on the robot itself, because I feel like it competes with the mapping processes for CPU time, but I might be overthinking it.

chrisl8, 2022-01-13:
@TCIII 

Try following these instructions if you want to run Rviz on Windows:

https://github.com/chrisl8/ArloBot/wiki/Running-Rviz-on-Windows

Let me know if you run into any issues or if it works well. It worked for me.

TCIII, 2022-01-13:
@chrisl8,

I decided to tryout installing the Ubuntu 20.04 focal server and then adding the ubuntu desktop (gnome).

I was able to SSH into the Ubuntu server after the server install, but after installing the Ubuntu desktop (gnome) I can still SSH into the Ubuntu server, however there is no WiFi available on the gnome desktop.

Running nmcli:

wlan0: unavailable
        "Broadcom BCM43438 combo and Bluetooth Low Energy"
        wifi (brcmfmac), DC:A6:32:A5:88:D3, hw, mtu 1500

Running sudo lshw -C network:

*-network:0
       description: Wireless interface
       physical id: 1
       logical name: wlan0
       serial: dc:a6:32:a5:88:d3
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=brcmfmac driverversion=7.45.229 firmware=01-2dbd9d2e ip=192.168.0.65 multicast=yes wireless=IEEE 802.11

Running lsmod indicates that the brcmfmac driver is loading at bootup and restart.

I am at a loss to understand why the Rpi wlan0 is unavailable even though the driver is loading.

Have you seen this when installing the Ubuntu focal server and then the Ubuntu desktop on your Rpi?

TCIII, 2022-01-14:
@chrisl8,

I suspect that setting the Ubuntu server up for SSH via WiFi is what is keeping the Ubuntu gnome desktop from accessing the Rpi WiFi as is evidenced by "wlan0: unavailable".

What I should have done is not setup the WiFi on the Ubuntu server and installed the Ubuntu desktop from the CLI after the Ubuntu server finished installing from the SD card,

Comments?

chrisl8, 2022-01-14:
@TCIII 
I **did** notice on one of my Pi's that the Wi-Fi settings in XWindows were entirely useless, as if they just had no connection to the system.
I'm not 100% sure if a system set up with `sudo apt-get install ubuntu-desktop` normally can configure Wi-Fi from XWindows or not? I typically initially set up my Pi's at my desk plugged into an Ethernet cable and only move them to Wi-Fi after I have them set up fully.
At some point I'll look into that more.

On that system where I needed Wi-Fi to work but it wasn't, what I had to do was:

Edit `/etc/netplan/50-cloud-init.yaml` (you must do it as root)

Set up the "wifis" section like this:

```
    wifis:
      wlan0:
      dhcp4: true
      optional: true
      access-points:
        "home network":
          password: "123456789"
```

A **key point** here is that the Wi-Fi network name and the password MUST BE IN QUOTES, even if they have no spaces. Otherwise it won't work.

Also, this is a `.yaml` file, so don't mess up the indentations or it won't work. Hopefully the existing content of the file makes it clear how to indent stuff.

Then reboot to activate the changes.

TCIII, 2022-01-14:
@chrisl8,

Thanks for the guidance, much appreciated.

I took your suggestion and did not setup the server with WIFI access and instead used an ethernet cable connected to my router.

After the server installed I had to wait for the "unattended upgrade" app to quit and then I was able to install the Ubuntu gnome desktop.

After I unplugged the ethernet cable, setup remote access and desktop sharing and selected my WiFi router, and rebooted I was able to SSH into the Rpi 4B and used VNC Viewer to view the Ubuntu desktop once I had disabled the vino encryption.

I use "sudo dpkg-reconfigure unattended-upgrades" to turnoff the unattended upgrades app.

Now to install the Arlobot/ROS Noetic software.

chrisl8, 2022-01-14:
@TCIII That is good information to know. At some point I should write up another "how to spin up from scratch" document for Ubuntu, on the Pi, with XWindows.

It is too bad that the standard Pi install doesn't have an option for "Desktop" Ubuntu 20.04, only Server.

They do have a "Desktop" install for Ubuntu 21, but ROS Noetic is specifically targeted at 20.04, and my experience in the past is that using any other version than the one they target is just asking for more problems to diagnose.

TCIII, 2022-01-14:
@chrisl8

I joined the MATE Ubuntu community forum and did a search on "vnc" and came up with a laundry list of user issues. Most of the posts were not resolved as users tried different vnc servers on MATE.

I found this [guideline](https://bytexd.com/how-to-install-configure-vnc-server-on-ubuntu-20-04/) to be the most comprehensive for installing 
TigerVNC on Ubuntu and use it with eight different desktops including MATE. There is a lot of flag waving involved that is transparent when using the built in screen sharing (vino) on the gnome desktop.

closed 2023-09-15 by chrisl8