chrisl8/ArloBot #188: The Arlobot software does not presently support Bluetooth Gamepads

TCIII, 2022-01-13:
@chrisl8,

**Issue:**
The Arlobot software presently only supports the MS Xbox 360 gamepad which requires a proprietary USB wireless receiver (WiFi) and is no longer in production.

**Solution:**
The Xbox One and One/S and the PS4/5 gamepads all use Bluetooth and are compatible with all Laptops and SBCs like the Rpi and Nano.
Provide an alternate Arlobot Bluetooth interface for newer Xbox and PS gamepads.

TCIII

chrisl8, 2022-01-13:
Some breadcrumbs:

It uses [the ROS joy package](http://wiki.ros.org/joy) so any joystick that it will support should work.

The configuration is in [xbox360_teleop.launch](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/launch/xbox360_teleop.launch)

If you can edit that file such that another joystick works, and you feel this is useful to future users, please post the new file here and then I can sort out how to "automate" the easy use for new users of that or other joysticks.

TCIII, 2022-01-14:
@chrisl8,

Like webcams, the gamepads have different device names.

The Xbox 360 device name is "Xbox 360 Wireless Receiver" while the Xbox One is "Xbox Wireless Controller" so you need to provide a way in "personalDataForBehavior.json" to change the name of the device.

Tom C

chrisl8, 2022-01-14:
@TCIII 

Oh cool. I have a bunch of controllers around. I should try them all out and see if there is a good way to auto detect any joystick connected to the system rather than requiring a specific call out in the configuration.

Leave this open and I'll look into that eventually.

I hadn't thought about it until now, but connecting a joystick to my other projects would be nice, and if I get this code to do it, then I could use it.

TCIII, 2022-01-17:
@chrisl8,

I presently have my Xbox One paired and connected with the Ubuntu MATE OS.

When I change "Xbox 360 Wireless Receiver" to "Xbox Wireless Controller" in 'find_xbox_controller.sh' otherwise my Xbox One is not detected.

Running the following results in:

thomas@thomas-desktop:~$ find_xbox_controller.sh
/dev/input/js0

Obviously the Xbox One is not a a USB wireless receiver which might cause an issue.

I used jstest to verify the gamepad hex keycodes and found the following:

Xbox One:

Left joystick vertical axis: 1

Left joystick horizontal axis: 0

Left button: 6

Right trigger; 4

From the arlobot_ros/shared/teleop_twist_joy/config/xbox.config.yaml file:

Left thumbstick vertical: 1

Left thumbstick horizontal; 0

So both the Xbox One and Xbox 360 joystick (thumbstick) hex keycodes are in agreement.

Left trigger button: 2

Right trigger button: 5

So the Xbox One and Xbox 360 trigger button hex keycodes are **not** in agreement.

TCIII, 2022-01-18:
@chrisl8,

For those Arlobot users who want to use the Xbox One gamepad to remotely drive your robot around, here are a few observations concerning using the Xbox One gamepad:

I presently have my Xbox One paired and connected with the Ubuntu MATE OS.

I had to change "Xbox 360 Wireless Receiver" to "Xbox Wireless Controller" in 'find_xbox_controller.sh' otherwise my Xbox One is not detected.

Running the following results in:

thomas@thomas-desktop:~$ find_xbox_controller.sh
/dev/input/js0

thomas@thomas-desktop:~$ ls -l /dev/input/js*
crw-rw-r--+ 1 root input 13, 0 Jan 17 16:11 /dev/input/js0

chrisl8, 2022-02-27:
@TCIII Which model Xbox One controller are you using?

I have tried two different models and as far as I can tell it is not possible to connect an Xbox One controller via Bluetooth to a Linux system.

It juts enters into a loop of connecting and disconnecting and never works, which is what many others also report:
https://github.com/atar-axis/xpadneo/issues/295

TCIII, 2022-02-27:
@chrisl8,

I am using both an XBox One and an XBox One/S. Both are BT and work fine with all versions of Linux.

You have to turnoff ertm to get the connection to stop doing that.

Have you had a chance to move ahead with the docking station project?

TCIII

TCIII, 2022-06-01:
@chrisl8,

I have found the ROS default value for the joystick dead zone of 0.1 to cause ArloBot to abruptly jump forward if the gamepad joystick is not judiciously pushed slowly forward to start ArloBot moving in the forward direction. 
In other words the default 0.1 dead zone value is like a step function when applying forward joystick motion and can cause the DHB-10 motor controller to go into overcurrent when the batteries are low.
The ROS dead zone value along with the maximum forward velocity can be adjusted in the /opt/ros/noetic/ directory, however changes to those two value will cause any ArloBot software updates to fail.

Regards,
TCIII

closed 2023-09-15 by chrisl8