chrisl8/ArloBot #189: pm2 save no longer functional in latest Arlobot/ROS install

TCIII, 2022-01-15:
@chrisl8,

**Steps to Reproduce Issue:**

I just successfully installing the Arlobot/ROS Noetic software on my Rpi 4B/4GB and tried to execute 'pm2 save' to synchronize the current process with the saved list and received the following message:

ubuntu@ubuntu:~$ pm2 save

Command 'pm2' not found, did you mean:

  command 'pmc' from deb linuxptp (1.9.2-1)
  command 'pmi' from deb powermanagement-interface (0.3.21)
  command 'pms' from deb pms (0.42-1build4)
  command 'pmw' from deb pmw (1:4.30-1)
  command 'gm2' from deb gm2 (4:10.0-1ubuntu2)
  command 'pm' from deb powerman (2.3.5-1build2)
  command 'wm2' from deb wm2 (4+svn20090216-4build1)
  command 'pom2' from deb libpod-pom-perl (2.01-3)

Try: sudo apt install <deb name>

TCIII, 2022-01-15:
@chrisl8,

It turns out that 'pm2 save' functions once the OS has been rebooted after the Arlobot/ROS Noetic install has been completed.

Maybe a note should be included at the end of the Arlobot/ROS Noetic install that 'pm2 save' be executed after a reboot?

Therefore I am closing this issue.

closed 2022-01-15 by TCIII

chrisl8, 2022-01-25:
The setup script definitely tells the user to reboot after it is run

https://github.com/chrisl8/ArloBot/blob/7ae706cc79a23d1d32432d21d84358065767a5f5/setup-noetic.sh#L948

Maybe the notice isn't clear enough amongst all of the noise of the setup process?