chrisl8/ArloBot #193: The Xbox One does not control the Arlobot speed or direction

TCIII, 2022-01-18:
@chrisl8,

I presently have my Xbox One paired and connected with the Ubuntu MATE OS.

I have to change "Xbox 360 Wireless Receiver" to "Xbox Wireless Controller" in 'find_xbox_controller.sh' otherwise my Xbox One is not detected.

Running the following results in:

thomas@thomas-desktop:~$ find_xbox_controller.sh
/dev/input/js0

thomas@thomas-desktop:~$ ls -l /dev/input/js*
crw-rw-r--+ 1 root input 13, 0 Jan 17 16:11 /dev/input/js0

Obviously the Xbox One is not a a USB wireless receiver which might cause an issue?

I used **jstest** to verify the gamepad hex keycodes and found the following:

**Xbox One:**

Left joystick vertical axis: 1

Left joystick horizontal axis: 0

Left button: 6

Right trigger; 4

**Xbox 360 Wireless Game Controller:**

From the arlobot_ros/shared/teleop_twist_joy/config/xbox.config.yaml file:

Left thumbstick vertical: 1

Left thumbstick horizontal; 0

**So both the Xbox One and Xbox 360 joystick (thumbstick) hex keycodes are in agreement.**

Left trigger button: 2

Right trigger button: 5

So the Xbox One and Xbox 360 trigger button hex keycodes are **not** in agreement.

**Steps to Duplicate Issue:**

1. Set "hasXboxController": true, in 'personalDataForBehavior.json'
2. Ran 'start-robot.sh'
    At startup the Xbox Controller was found (My Xbox One is paired/connected with the Ubuntu MATE OS at /dev/input/js0)
    Under PARAMETERS the 'teleop_twist_joy' parameters were listed as found in the xbox.config.yaml file.
    Under NODES the joy_node (joy/joy_node) is listed.
    process [teleop_twist_joy-10]:  started with pid  [15539].
3. Start completed after establishing the serial connection with the PAB.
4. I could then put my hand in front of the front/rear PING sensors and obtain wheel movement (Escaping)

**Expected Behavior;**

According to the Arlobot writeup remote control with the Xbox 360 (Xbox One) gamepad should work as soon as you Start ROS, either with 'start-robot.sh' or with the Web Interface.

**Actual Behavior:**

No control of the Arlobot wheels with the Xbox One left joystick.

**Additional Information**:

I observe the same behavior when using the Web Interface with ROS running.
There is no wheel response when using the Xbox One gamepad left joystick.
However I can control Arlobot with the Remote Control Panel using the Mouse Cursor so Motor Control is functional.

Suggestions to troubleshoot this issue.

Tom C

chrisl8, 2022-01-18:
Did you get a chance to look at the blog link from the Readme here: https://ekpyroticfrood.net/?p=115

It has some guidance on how to use the joystick (you have to hold down a "trigger" button to make it move, to avoid it randomly moving due to joystick bumps or drift).

It also has some debugging ideas.

I will post some more suggestions here later when I have more time.

TCIII, 2022-01-18:
@chrisl8,

I have found a possible [solution](https://answers.ros.org/question/384456/ros2-joy_node-debugging/) on ROS Answers even though it involves ROS2.

Tom C

TCIII, 2022-01-18:
@chrisl8,

Issue closed out due to "Cockpit Error".

Tom C

closed 2022-01-18 by TCIII