chrisl8/ArloBot #195: Clicking on BLINKY in the Behavior Panel causes ROS to shutdown

TCIII, 2022-01-24:
@chrisl8,

SBC: Rpi 4B/4G
OS: Ubuntu MATE (focal 20.04)
ROS: Noetic

**Steps to Duplicate Issue:**

1. Programmed an Arduino Uno R3 with 'neoPixelControl.ino' and set #define NUM_LEDS 16 because I am using an Adafruit Flora NeoPixel Ring with 16 LEDs
2. Connected the Arduino Uno R3 to the USB 2.0 HUB attached to the Rpi 4B USB 2.0 port
3. Ran 'find_Arduino.sh' to determine that the Arduino was on 'ttyACM0'
4. Ran 'udevadm info -n /dev/ttyACM0' to find the Arduino ID_SERIAL: Arduino__www.arduino.cc__0043_7503330323535170E162
5. Made "arduinoUniqueString": "Arduino__www.arduino.cc__0043_7503330323535170E162", in 'personalDataForBehavior.json'
6. "arduinoStringLocation": "ID_SERIAL", in 'personalDataForBehavior.json'
7. "hasBeenEdited": true in 'personalDataForBehavior.json'
8. Connected the Flora NeoPixel Ring Data In pin to the Arduino #6 data pin
9. Rebooted the Rpi 4B
10. Browsed to the Rpi 4B ip;8080 to view the Web Interface
11. Started ROS
12. When Start ROS completed opened the Behavior Panel
13. Clicked on BLINKY
14. After approximately 5 seconds ROS stopped and the Web Interface displayed Start ROS

 **Expected Behavior;**

A light display on the Flora NeoPixel Ring

**Actual Behavior:**

No display visible on the Flora NeoPixel Ring and ROS stopped and went back to Start ROS

**Additional Observations:**

1. Once I had completed programming the Arduino Uno R3 with the 'neoPixelControl.ino' code, I opened the Serial Terminal and verified that a "Ready' prompt had been transmitted from the Uno.

Suggestions on how to troubleshoot this issue?

Regards,
Tom C

chrisl8, 2022-01-25:
That code is not very "portable". I threw it together in one evening in order to add some lights to my robot for an event, and it has worked for many years, but never really been set up for easy use by various hardware.

You say that "start-robot.sh" isn't detecting the Arduino USB port. What makes you mention this? Is it claiming that it tried and failed?
I ask because I don't see any reason why start-robot.sh would care about the port for the neopixles, as they are not run by ROS.

If you run:
`~/catkin_ws/src/ArloBot/scripts/find_Arduino.sh`
does it find the port where the Arduino is?

If not, then your arduinoStringLocation and/or arduinoUniqueString aren't correct.

If that works, then
You should be able to start the light show without ROS running by going to a terminal on the Pi and running

`node ~/catkin_ws/src/ArloBot/node/Arduino.js`

Let me know what the output of that command is if it doesn't work.

TCIII, 2022-01-26:
@chrisl8,

Thanks for the response, much appreciated.

The results from running **'~/catkin_ws/src/ArloBot/scripts/find_Arduino.sh'**:
**/dev/ttyACM0**
Which is the result I described in my first post above when I ran **'find_Arduino.sh'**.

The result from running **'node ~/catkin_ws/src/ArloBot/node/Arduino.js'**:
 
 thomas@thomas-desktop:~$ node ~/catkin_ws/src/ArloBot/node/Arduino.js
/home/thomas/catkin_ws/src/ArloBot/node/Arduino.js:137
            webModel.relays.find((x) => x.name === 'arduino').relayOn &&
                                                             ^

TypeError: Cannot read properties of undefined (reading 'relayOn')
    at Timeout._onTimeout (/home/thomas/catkin_ws/src/ArloBot/node/Arduino.js:137:62)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)

I think this verifies what I mentioned in my first post above in **"Additional Observations"** item #3: "Is the fact that I have no USB relays causing a problem with the 'Arduino.js' code?"

So how do I get around the "no relay" issue?

Regards,
Tom C

chrisl8, 2022-01-26:
Yes, this looks like a bug in the code. It tries to deal with alternate relay options, but clearly fails. I never tested it without the relays. The fun part is that **I do not have the relays anymore either** since moving to the Raspberry Pi.

I will try to look at the code and fix it later this week.

chrisl8, 2022-01-26:
@TCIII  I pushed an update that attempts to fix this.

Try this:

```
cd ~/catkin_ws/src/ArloBot
git pull
node ~/catkin_ws/src/ArloBot/node/Arduino.js
```

and let me know if it works any better now.

TCIII, 2022-01-26:
@chrisl8,

Thanks for the quick response, however no joy.

I did as you directed above and here is the results:

 thomas@thomas-desktop:~$ node ~/catkin_ws/src/ArloBot/node/Arduino.js
<-READY
0,0,0,0,33,119,25->
<-BUSY
<-ADY
<-ADY
<-ADY
<-ADY
Then no additional output on the CLI.
No display on the Adafruit Flora NeoPixel Ring (16 LEDs) even though I changed the '#define LOOP_START 33' to '#define LOOP_START 1' and '#define NUM_LEDS 119' to '#define NUM_LEDS 16' in the 'neoPixelControl.ino' code.

Your Arduino.js defines the LED array as follows:

// Named pixels
    this.pixel = {
      LOOP_START: 33, // First LED on the top of the ring.
      LOOP_END: 119, // My last pixel has no green :(

Where as I have changed the following in the 'neoPixelControl.ino' code to conform to the Flora NeoPixel Ring LED configuration:

'#define LOOP_START 1'
'#define NUM_LEDS 16'

Could the above conflict cause an issue between 'Arduino.js' and 'neoPixelControl.ino' as far as the display goes?
Also, even though ['FastLED.h'](https://github.com/FastLED/FastLED) does support the WS2812 Flora NeoPixel Ring?

Tom C

TCIII, 2022-01-26:
@chrisl8,

I fixed the problem by changing the following in Arduino.js:

LOOP_START: 33 to 1
LOOP_END: 119 to 16

Works great now. Very impressive light show.

Looks like you need to make the node Arduino.js LOOP_START and LOOP_END variables in the 'personalDataForBehavior.json' file to account for displays other than the one you were using? Also, '#define LOOP_START XX' and '#define NUM_LEDS XXX' in the 'neoPixelControl.ino' code should be adjusted to be the same as in the node 'Arduino.js'

Thanks for the help, much appreciated.

Tom C

chrisl8, 2022-01-26:
@TCIII 

> Looks like you need to make the node Arduino.js LOOP_START and LOOP_END variables in the 'personalDataForBehavior.json' file to account for displays other than the one you were using? Also, '#define [...]

Good idea. Leave this issue open and I'll add it that.

I'm glad someone else found this code useful!

TCIII, 2022-01-26:
@chrisl8,

I try to adhere to the old engineering saying "If it ain't broke, don't fix it", however Adafruit has come out with a library specifically for the NeoPixel LED family and you might want to consider replacing 'FastLED.h' with '[Adafruit_NeoPixel.h](https://github.com/adafruit/Adafruit_NeoPixel)' when you update your software.

Tom C

chrisl8, 2022-01-26:
@TCIII 

Yeah, Adafruit has has this for many years. My guess is that the reason I used FastLED.h was that I was following a tutorial at the time. If I was going to do this again, instead of an Arduino and all of this manual code, I would use an ESP32 and the https://kno.wled.ge/ project, which is what I've done on other things.

TCIII, 2022-01-27:
@chrisl8,

I believe that I have uncovered a minor bug in both node 'Arduino.js' and 'neoPixelControl.ino'.

Both 'FastLED.h' and 'Adafruit_NeoPixel.h' assume that the first pixel address is "0", which it is, according to the Adafruit NeoPixel documentation.
 
The only NeoPixel configuration value required, according to the Adafruit documentation, when using either 'FastLED.h' or 'Adafruit_NeoPixel.h' is the number of pixels in the string.

When I set the 'Arduino.js' LOOP_START and LOOP_END to 1 and 16 respectively and 'neoPixelControl.ino' '#define LOOP_START' and '#define NUM_LEDS' to 1 and 16 respectively, the first LED in the string will not light because its address is "0" and not "1".

If I set 'Arduino.js' LOOP_START and LOOP_END to 0 and 16 respectively and 'neoPixelControl.ino' '#define LOOP_START' and '#define NUM_LEDS' to 0 and 16 respectively, the first LED does light up when the light show starts, however after a while the light show freezes up with just four LEDs, in a box pattern, glowing white in color.

I believe therefore there is an issue in the way the LOOP_START value and the actual address ("0") of the first pixel in the string are being used by 'Arduino.js' generating the light show pattern and 'neoPixelControl.ino' controlling the pixel string.

Comments?

Regards,
Tom C

chrisl8, 2023-09-15:
My 'blinky lights' still work, so I am going to close this out, as it has been a whiel.

closed 2023-09-15 by chrisl8