chrisl8/ArloBot #198: Add Docking ability to charge robot when power is low

chrisl8, 2022-02-16:
Similar to how most robot vacuums can dock themselves and recharge.

Consider using this code: https://github.com/wennycooper/mybot_autodocking/

chrisl8, 2022-02-16:
As an initial test do this on the robot:

```
source ~/catkin_ws/devel/setup.bash
cd ~/catkin_ws/src
git clone https://github.com/wennycooper/mybot_autodocking.git
cd ~/catkin_ws
rosdep update
rosdep install -q -y -r --from-paths src --ignore-src
catkin_make
```
That should build and install the code.
Now, in a terminal window, on the robot run `start-robot.sh` OR start ROS from the web Interface.

Once it is going, in another terminal run:

```
source ~/catkin_ws/devel/setup.bash
rosrun mybot_autodocking mybot_autodocking
```
It should start with no output and just stay that way, I think.

In another window, run:
```
source ~/catkin_ws/devel/setup.bash
rostopic echo /irValue
```
Is there any output? If so, can you tell me what it is?
If you "bring up" or "turn on" your IR device, does it that change things?

This is where I'm needing help, because I'm not sure what the output should look like.

Finally try this one:
```
rostopic echo /cmd_vel
```
and again. does it have any output when then IR system is operating?

TCIII, 2022-02-16:
@chrisl8,

I will give it a shot this weekend.

Regards,
TCIII

chrisl8, 2022-02-17:
@TCIII 

When you run the `rosrun mybot_autodocking mybot_autodocking` I suggest having the robot on blocks, because it will probably start trying to move. I could be wrong, but that is the idea.

Also be aware that it will probably lock out all other input to the robot. I will have to add the setup to "mix" the input into the other sources, OR you will need to create some automation to only start the autodocking when you are ready to dock. We can work that out later.

You will want to watch the output of `rostopic echo /cmd-vel` and `rostopic echo /irValue` and if there IS output, then move the robot or IR sender to see if it appears to change the output. Honestly, if you get anything on `rostopic echo /irValue` that will be a successful first test.

I don't actually know how the Arduino talks to the ROS code or how it knows which input device to use.
In fact, when I look into the code here I don't even see anything that is talking to other code:
https://github.com/wennycooper/mybot_autodocking/tree/master/src

Both .cpp files seem to have code to send twist commands to ROS, and they subscribe to an /irValue topic, but what is publishing messages to the /irValue topic?

Maybe the Arduino code itself is doing that? If the Arduino is somehow able to publish right into the ROS topics running on the host, that would make sense. I didn't know that this was possible, but I've never tried either.

Also, I could be wrong about running
`rosrun mybot_autodocking mybot_autodocking`

You might need to run this instead:
`rosrun mybot_autodocking_irCode mybot_autodocking_irCode`

I guess try each or both to see what happens?

Make sure your Arduino is plugged in when you try it.

If nothing works, we'll have to do more debugging. I do have an Arduino, so I could probably test that code too, even without the IR hardware.

Let me know what you find.

TCIII, 2022-02-17:
@chrisl8,

Thanks for the quick response, much appreciated.

I will integrate your detailed update into the testing.

Regards,
TCIII

chrisl8, 2022-02-18:
@TCIII

Have you tried it yet?
Did you upload code to the Arduino? If so, what code did you use? I haven't looked at the Arduino code myself, so I don't know how it works. If you send me a link to the `.ino` file you used, I could give some insight on how it appears to work.

It seems premature to me to call it a failure if you haven't tried it yet.

chrisl8, 2022-02-18:
@TCIII My **guess** is that you probably need to do everything I mentioned above, **and also** run:

rosrun rosserial_python serial_node.py /dev/tty<USB# or ACM#>

which will provide the bridge between the Arduino and ROS, and then you would see data on the /irValue topic.

I don't see "irValue" in the cod you pasted, so it may also be necesary to pass the "topic" to serial_node.py somehow?
Unless the `.ino` file provided by the project already encodes that somehow?

chrisl8, 2022-02-18:
@TCIII 

No harm in just trying things either. Run the code, see what happens.

It sounds like I will have to put some code on an Arduino to fully know how to make it work. I might get to that this weekend or I might not.

This kind of complex setup process, long string of dependencies, and lots of half-documented solutions is why I made this repository to begin with. ROS is great, but it is not full of people who like to document things or maintain fully functioning systems. 😄

TCIII, 2022-02-19:
@chrisl8,

I have completed the steps in "Setup" and received the following:

`[100%] Built target mybot_autodocking_irCode
[100%] Built target mybot_autodocking
`

TCIII, 2022-02-19:
@chrisl8,

I have run `rosrun mybot_autodocking mybot_autodocking_irCode` and received the following on the CLI until I hit ^C:

`thomas@thomas-desktop:~$ rosrun mybot_autodocking mybot_autodocking_irCode`
`x=0.000000, y=0.000000`
`x=0.000000, y=0.000000`

I had to change `rosrun mybot_autodocking_irCode mybot_autodocking_irCode` to `rosrun mybot_autodocking mybot_autodocking_irCode` because `mybot_autodocking_irCode` is in the 'mybot_autodocking' directory. The 'mybot_autodocking_irCode' directory does not exist.

Running the following:

`source ~/catkin_ws/devel/setup.bash`
`rostopic echo /irValue`

Yielded the following:
`WARNING: topic [/irValue] does not appear to be published yet`

Running the following:

`source ~/catkin_ws/devel/setup.bash`
`rostopic echo /cmd_vel `

Yielded this output with and w/o the IR beam:

linear:
x: 0.0
y: 0.0
z: 0.0
angular:
x: 0.0
y: 0.0
z: 0.1

Looks like we are making some progress using `rosrun mybot_autodocking mybot_autodocking_irCode`, but the mybot_autodocking_irCode node is not getting input from the Metro 328 IR Receiver. Therefore I need advice on installing rosserial_python.

Comments?

TCIII

TCIII, 2022-02-19:
@chrisl8,

You are right about Cooper's ROS code being incomplete. Here is the ROS Answers response:

"On the host running rosserial, ROS is not responsible for "finding" the linux /dev/[somehing] device. The author of the ROS launch file must create entries on the ros-parameter-server specifying a linux device name and a baudrate.

For USB, creating the linux device in /dev is typically done with a udev rule. You must install this rule before you insert the usb device and before the ROS launch file is executed. This new udev rule will be triggered when your specific USB-device is inserted, and then udev will create a fixed linux /dev name that you specify in the rule e.g. /dev/arduino

For this to work, the name in the udev rule has to exactly match the device name you specify in your ROS launch file."

I know that the IR Receiver Metro 328 is on USB0 and its baud rate is set at the default 57600.

Should I use udevadm to determine the IR Receiver Metro 328 info? 

Regards,
TCIII

TCIII, 2022-02-21:
@chrisl8,

Will you be able to do the following this week so that I can continue to validate the robot docking software:

1) Create a `find_IR_Receiver.py` script
2) Create a ROS launch file for the ROS node `mybot_autodocking mybot_autodocking_irCode` that specifies the Arduino IR Receiver USB port and any other support modules that are required.

Regards,
TCIII

TCIII, 2022-02-22:
@chrisl8,

I would appreciate some input here as I would like to move this project along to completion.

Regards,
TCIII

chrisl8, 2022-02-22:
@TCIII 

Did you install `ros-noetic-rosserial-python` yet?

TCIII, 2022-02-22:
@chrisl8,

Thanks for the response, much appreciated.

Not yet as I was waiting for your feedback which you have given. 

Once I install `ros-noetic-rosserial-python` do you want me to run `rosrun mybot_autodocking mybot_autodocking_irCode` and try running `source ~/catkin_ws/devel/setup.bash` followed by `rostopic echo /irValue` and then try running `source ~/catkin_ws/devel/setup.bash` followed by `rostopic echo /cmd_vel`?

Regards,
TCIII

chrisl8, 2022-02-22:
@TCIII 

Yes, and then in addition run
`rosrun rosserial_python serial_node.py /dev/ttyUSB0`
(or whatever the device name is for the IR controller that is connected)

TCIII, 2022-02-22:
@chrisl8,

Unfortunately `sudo apt-get install ros-noetic-rosserial-python` failed to install:

```
Get:1 http://packages.ros.org/ros/ubuntu focal/main arm64 ros-noetic-rosserial-msgs arm64 0.9.2-1focal.20210424.032443 [23.8 kB]
Err:2 http://packages.ros.org/ros/ubuntu focal/main arm64 ros-noetic-rosserial-python arm64 0.9.2-1focal.20210922.181500
  404  Not Found [IP: 140.211.166.134 80]
Fetched 23.8 kB in 1s (33.2 kB/s)
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-rosserial-python/ros-noetic-rosserial-python_0.9.2-1focal.20210922.181500_arm64.deb  404  Not Found [IP: 140.211.166.134 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```
TCIII

chrisl8, 2022-02-22:
@TCIII  

You are either having network/Internet issues or there is some outage for the particular ROS mirror you hit.

You can try running these to confirm/update your package settings:
```
sudo apt update
sudo apt upgrade
```

and then try again. It could just be transient. packages.ros.org does sometimes go down or get overloaded.

TCIII, 2022-02-22:
@chrisl8,

It will take a while as there are 291 packages to upgrade.

TCIII

TCIII, 2022-02-22:
@chrisl8,

Running `sudo apt install -y ros-noetic-rosserial-python` worked after the update/upgrade finished.

TCIII

TCIII, 2022-02-22:
@chrisl8,

Since the update/upgrade, `find_camera.sh UVC` cannot find the camera so `start-robot.sh` will not start.

Running `dmesg | grep -i "Camera"` gives:

[    6.253150] input: UVC Camera (046d:0825) as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/input/input0

Suggestions?

TCIII

TCIII, 2022-02-22:
@chrisl8,

Made some progress:

Ran the following sequence in separate terminals:

`start-robot.sh`

`source ~/catkin_ws/devel/setup.bash`
`rosrun mybot_autodocking mybot_autodocking_irCode`

`rosrun rosserial_python serial_node.py /dev/ttyUSB0`

`source ~/catkin_ws/devel/setup.bash`
`rostopic echo /irValue`

`rosrun mybot_autodocking mybot_autodocking_irCode` output when both IR Receivers were illuminated:

x=0.000000, y=0.000000
x=1.000000, y=1.000000     #the wheels were turning in reverse as expected from the code.
x=0.000000, y=0.000000

`rosrun rosserial_python serial_node.py /dev/ttyUSB0` output:

[INFO] [1645563476.048943]: ROS Serial Python Node
[INFO] [1645563476.066510]: Connecting to /dev/ttyUSB0 at 57600 baud
[INFO] [1645563478.185020]: Requesting topics...
[INFO] [1645563478.280282]: Note: publish buffer size is 280 bytes
[INFO] [1645563478.284677]: Setup publisher on irCode [geometry_msgs/Vector3]

`rostopic echo /irValue` output:

WARNING: topic [/irValue] does not appear to be published yet

So it looks like there is a disconnect between /irValue in the    and /irCode in the Arduino ino. 

"irValue" is called "irCode" in the Arduino .ino code:

```geometry_msgs::Vector3 irCode;`
`ros::Publisher pub("irCode", &irCode);`
``irCode.x = flag2;`
`irCode.y = flag3;`
`pub.publish(&irCode);``

From Cooper's 'mybot_autodocking_irCode.cpp':

`irValue_sub = nh.subscribe("/irCode", 10, irValueCallback);
`
`void irValueCallback(const geometry_msgs::Vector3 &vector3)`
``vec3.x = vector3.x;`
` vec3.y = vector3.y;``

Comments?

TCIII

chrisl8, 2022-02-22:
@TCIII 

When all of this is running, what does the output of `rostopic echo /irCode` show?

TCIII, 2022-02-23:
@chrisl8,

I ran the setup code again and this time I ran `rostopic echo /irCode`.

When I had the IR beam illuminating both IR Receivers the output looked like this:

x: 1.0
y: 1.0
z: 0.0

When I had the IR beam illuminating the Right IR Receiver the output looked like this:

x: 0.0
y: 1.0
z: 0.0

When I had the IR beam illuminating the Left IR Receiver the output looked like this:

x: 1.0
y: 0.0
z: 0.0

When I had the IR beam illuminating both IR Receivers both wheels were turning backwards as the robot is now aligned to the dock and is trying to engage the dock. The wheels are turning backwards because Cooper's `mybot_autodocking_irCode` had the robot back into the dock:

`if (foundDockingStation) {`
    `geometry_msgs::TwistPtr cmd(new geometry_msgs::Twist());`
    `cmd->linear.x = -0.08;`  # Going in reverse
    `cmd->angular.z = 0.1 * (vector3.x - vector3.y);`
    `if (vector3.x || vector3.y)`
      `cmd_vel_pub.publish(cmd);` 

However, pointing the beam at just one sensor does not seem to make the `mybot_autodocking_irCode` attempt to rotate in the other direction in order to illuminate both detectors with the beam.

Comments?

TCIII, 2022-02-23:
@chrisl8,

Observation:
After testing last night I found that the Right wheel was turning much faster than the Left wheel when ArloBot was supposed to go in a straight line. I tried several software fixes to no avail. Finally I used an air can to blow any dust out of the Right wheel optical encoder and that cured the problem. It might be a good idea to blow the dust out of the wheel encoders periodically. This is another reason why hall effect motor encoders are superior to optical wheel encoders.  Also motor encoders tend to have much higher resolution than wheel encoders.
When I was having the Right wheel issue I ran bash `<(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)` to see if there was a software issue.

Now when I run `rosrun rosserial_python serial_node.py /dev/ttyUSB0` I get this error message:

[INFO] [1645631340.221298]: ROS Serial Python Node
[INFO] [1645631340.236635]: Connecting to /dev/ttyUSB0 at 57600 baud
[INFO] [1645631342.352447]: Requesting topics...
[ERROR] [1645631342.475269]: Mismatched protocol version in packet (b'\xbf'): lost sync or rosserial_python is from different ros release than the rosserial client
[INFO] [1645631342.483907]: Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)

Did running `<(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)` mess up rosserial client?

Suggestions to troubleshoot this issue.

TCIII

TCIII, 2022-02-23:
@chrisl8,

Here is a detailed [look at docking](https://www.theconstructsim.com/ros-qa-053-how-to-move-a-robot-to-a-certain-point-using-twist/) using Twist as in the irCode.cpp.

TCIII

TCIII, 2022-02-23:
@chrisl8,

Here is a snippet describing Cooper's docking algorithm as far as IR beam response goes:

If only R0 got RC5(0x43) code, the robot should rotate in CCW.
If only R1 got RC5(0x43) code, the robot should rotate in CW.
If both R0 and R1 got RC5(0x43) code, the robot should move forward.
If neither R0 nor R1 got RC5(0x43) code for a period of time, it should be treat as "lost docking station",
and it should go back to the previous stop.
Until the sensor limitSW is contact to the docking station. Then stop the base.

Comments?

TCIII

chrisl8, 2022-02-23:
Unfortunately Linux will move your USB devices around a lot. They are liable to change any time you reboot, or if you run any command that resets the devices, and of course if you unplug anything and plug it back in. That is why I made scripts to use stable values to find the ports when running things.
I can try to guide you through making such a script for your new device.

My testing causes my robot to just rotate constantly, which makes sense, as it is continuously in the state of "no IR signal received, rotate until it is found" since I have no input device.
Does your robot not do that?

What is the output of `rostopic echo /cmd_vel`?

TCIII, 2022-02-23:
@chrisl8,

I have not looked at `rostopic echo /cmd_vel` recently, but will give it a shot.

As I described above, with no IR beam detected (irCode x:0, y:0, and z:0) the wheels quit turning, while with both IR receivers illuminated (irCode x:1, y:1, and z:0) the wheels turn backward as the irCode.cpp tries to back into the docking station:
`cmd->linear.x = -0.08;` # Going in reverse
`cmd->angular.z = 0.1 * (vector3.x - vector3.y);`

With x and y equal 1, angular.z = 0 so only linear.x = -0..08 which is reverse.

Yes, I would be interested in writing such a script.

TCIII

TCIII, 2022-02-23:
@chrisl8,

Here is what I see with `rostopic echo /cmd_vel`:

1) No beam at the start:
Linear
x:0
y:0
z:0
Angular
x:0
y:0
z:0.1
The wheels try to rotate left

2) Both IR Receivers illuminated:
Linear
x:-0.08
y:0
z:0
Angular
x:0
y:0
z:0.0   # an occasional -0.1 will appear momentarily

3) Only the Left IR Receiver illuminated:
Linear
x:-0.08
y:0
z:0
Angular
x:0
y:0
z:0.1

3) Only the Right IR Receiver illuminated:
Linear
x:-0.08
y:0
z:0
Angular
x:0
y:0
z:-0.1

4) If both IR Receivers are being illuminated and then not illuminated:
Linear
x:0
y:0
z:0
Angular
x:0
y:0
z:0
The `irCode.cpp` will not try and hunt for the IR beam like it did when there was no IR beam at startup.

Observations:

1) I don't understand when only one IR Receiver is illuminated in 2 and 3 why Linear x = -0.08 while Angular z = either 0.1 or -0.1 when Linear x should be 0.0 so that Angular z will drive the robot either right or left to illuminate both IR Receivers.

2) I don't understand why the `irCode.cpp` will hunt for the beam at startup ( the first (1) above), but will not hunt for the beam if the beam has been illuminating both IR Receivers and then is taken away (beam lost).

Comments?

TCIII

chrisl8, 2022-02-23:
First, it looks like everything is working! 🥳  Maybe not perfectly, but your hardware is working, the code functions, and ROS is receiving input, and the robot is acting on it!

1. My assumption for why X is not set to 0.0 when one side sees the IR signal is that the developer's intention was for it to continue to move toward the dock while correcting the orientation, rather than stopping and rotating to correct every time it got off to one side. Since it sees the IR beam, it knows the goal is in sight, so it might as well move toward it, even while correcting the angle.
Imagine the difference between a smooth wavy line as it approaches vs. a series of sharp zig-zags as it stops, corrects starts again, stops, corrects, starts again.
My guess is that the specific values in the code may require tuning on your part to make it work best with your robot, with your dock, on your floor surface.

You can adjust the X value as you see fit at [line 23 of `mybot_autodocking_irCode.cpp`](https://github.com/wennycooper/mybot_autodocking/blob/d13e0e6bb25e1f3adc94eb94c10043c6b1dedea1/src/mybot_autodocking_irCode.cpp#L23) although that is the **only** place X is set, so if you make it 0.0, it will never move toward the goal. More logic will need to be added to the code if you want it to set X to 0.0 when rotating, but higher when moving straight.

Remember that changes to `cpp` code will not affect code until you [run `catkin_make` again like you did in the setup](https://github.com/chrisl8/ArloBot/issues/198#issuecomment-1042350414) to rebuild the code. (That is one drawback of C++ code vs. Python code in ROS. The benefit is that C++ code in theory runs faster with less load on the CPU, although that is debatable with modern Python and CPUs.)

2. If you look at the code in [`mybot_autodocking_irCode.cpp`](https://github.com/wennycooper/mybot_autodocking/blob/master/src/mybot_autodocking_irCode.cpp) you can see that it sets an initial value of 0 for `foundDockingStation ` in [line 9](https://github.com/wennycooper/mybot_autodocking/blob/d13e0e6bb25e1f3adc94eb94c10043c6b1dedea1/src/mybot_autodocking_irCode.cpp#L9), and then in [line 54](https://github.com/wennycooper/mybot_autodocking/blob/d13e0e6bb25e1f3adc94eb94c10043c6b1dedea1/src/mybot_autodocking_irCode.cpp#L54) sets it to 1, **but it never sets it back to 0 ever.**

So the code author assumed that once the IR beam was seen, it would remain seen until docked.
This was either a very poor assumption or just one that worked for the author and not you.

It will take some updates to the [`mybot_autodocking_irCode.cpp`](https://github.com/wennycooper/mybot_autodocking/blob/master/src/mybot_autodocking_irCode.cpp)  code to change that behavior.

TCIII, 2022-02-23:
@chrisl8,

Thanks for the analysis, much appreciated.

To use Cooper's code there has to be an input to ROS that tells the docking code when the robot is docked so that Linear x = 0.0. He apparently used a limit switch to tell the ROS code when the robot had docked, but never integrated it into the irCode.cpp.
What do you envision as the path forward now with the docking project? This could be useful for the "off the shelf" robot chassis that you have planned?

TCIII

TCIII, 2022-02-24:
@chrisl8,

Can we work up a plan to integrate the docking code into the ArloBot/ROS software package and make it an option in the `personalDataForBehavior.json` file?

TCIII, 2022-02-24:
@chrisl8,

I changed the `cmd->linear.x = -0.08;` to `cmd->linear.x = 0.08;` and reran `catkin_make`. Now ArloBot moves forward when both IR Receivers are illuminated.
I left the IR transmitter off at startup and ArloBot began a very slow CCW turn when looking from the front as it searched for the IR transmitter beam. I believe that you saw the same response when there was no IR transmitter beam at start up?

Thoughts about the docking station supplied voltage and current required to charge the battery and run the SBC at the same time:
1) Most three stage SLA battery chargers require around 16 vdc. My SBC and accessories stepdown voltage regulators can handle between 12-24 vdc so that won't be a problem.
2) The three SLA battery chargers I have charge at a 1 amp rate which means that the 1 amp current is split between the two SLA batteries.
3) I would estimate that the SBC and accessories (PAB/DHB-10/Arduinos/etc) are probably drawing around minimum current while docked so maybe figure around 3 amps.
4) Therefore the docking station should be capable of supplying 16 vdc and at least 5 amps through the docking contacts.

Comments?

TCIII

TCIII, 2022-02-25:
@chrisl8,

Bump?

TCIII

TCIII, 2022-02-25:
@chrisl8,

Instead of using the limit switch, that Cooper proposed, to indicate when the robot chassis had successfully docked with the docking station, might it be possible to use the front PING sensor to determine when the robot chassis is correctly docked with the docking chassis? The front PING sensor could be set to detect the distance between the front of the robot chassis and the docking station and the docking station code could issue a Linear x: = 0.0  Twist command to stop the robot chassis?

Comments?

TCIII

TCIII, 2022-02-26:
@chrisl8,

Bump?

I would really appreciate a response so I can move this project to completion.

TCIII

TCIII, 2022-02-27:
@chrisl8,

I experimented again today with the autodocking system and found that the autodocking software is quite capable of driving headlong into the IR Transmitter and being perfectly centered on the beam.

Observation:

The robot starts with a CCW search to detect the IR beam and as a result the Left detector is the first detector to see the beam. Since both the Right and the Left sides of the detector mount are open, the Left detector will be illuminated before the robot has almost centered itself towards the beam.
Therefore I think that is might be a good idea to shield both sides of the IR Receiver so that the FOV of each detector can be made narrower than it presently is in order to better center the robot on the IR beam.

Comments?

TCIII

TCIII, 2022-03-01:
@chrisl8,

I have found these [high current pogo pins](https://www.amazon.com/uxcell-Contact-Testing-Current-Terminal/dp/B0711FP7MT) that can be used on the robot side to make contact with brass strips on the docking station. 

I have built a fixture that attaches under the front PING mount and will support a plate to which the pogo pins can be mounted. I am also in the process of building the docking station that will house the IR Transmitter and the brass strip charging plates.

TCIII

TCIII, 2022-03-03:
@chrisl8

Here are two photos of the IR Receiver and the Docking Probe from head-on and from above.

You can see the high current pogo pins in the center of the docking probe. There will be three pins each for the positive and negative power connections with the docking station to be able to handle both the battery charging and SBC/Accessories current requirements.

During recent testing the ir_Code.cpp code was able to bring the pogo pins to within 1/2 inch or less of the simulated docking station four inch wide brass contact strips. The docking station brass contact strips will be mounted on a moveable carriage backed by a set of springs. The ability of the brass contact strips carriage to move backward when being contacted by the ArloBot docking probe will insure positive contact between the docking station brass contacts and the ArloBot probe pogo contact.

TCIII

[image]
[image]

TCIII, 2022-03-07:
@chrisl8

Here is a prototype docking station that I have completed.

The pogo pins on the ArloBot docking probe contact the middle of the brass strips on the docking plate.

The docking plate is spring loaded and can move horizontally to allow for horizontal play between the ArloBot docking probe and the docking station.

The pogo pins on the docking probe can move 1/4 inch horizontally which will allow for slight off angle contact between the ArloBot docking probe and the contact plate on the docking station.

TCIII 
[image]

TCIII, 2022-03-08:
@chrisl8,

Where have you gotten off to?

Regards,
TCIII

chrisl8, 2022-03-10:
@TCIII 

That is amazing! Lots of engineering there.

Life has gotten very busy for me. I'm still tinkering a bit, but not making enough progress on anything to report at the moment.

chrisl8, 2022-03-10:
Why so many volts? My 12 volt chargers only go to like 14 volts I think, usually more like upper 13's. I think 14 is the max charging voltage for a 12 volt SLA batter isn't it, or something around 14, not 16?

Why so many amps? I'm only using a 1.1 amp charger and it is able to keep the Pi running while also toping off two SLA batteries. Without the Propeller boards, just sing a Roboclaw, a Pi and a couple of other gadgets, my 750 milliamp chargers even do the job.

I feel like your setup is very different from mine if you need 16 volts at over 3 amps.

TCIII, 2022-03-10:
@chrisl8,

Will it be possible for you to integrate the ir_Code.cpp node into the ArloBot ROS package in the near future so that a user can use the webui to command the ArloBot to dock.

Presently the default 12 cm (~4.75 in) halt distance for the lower center PING sensor is keeping ArloBot from docking with the docking station as front of the docking probe is ~5.7 cm from the face of the PING sensor.

Would it be possible for the docking node code to temporarily modify the default halt distance to a user specified distance during the docking process?

Comments?

TCIII

chrisl8, 2022-03-10:
I think our goals and hopes for the electrical components are not the same. I don't intend to run an on-board voltage converter, but rather an external charger. I can see how getting one that provides more than 1 or 2 amps might be better, but I've had good luck just using the 1 amp chargers.
I also don't hope to make the Pi run Rviz on a normal basis. It is nice for initial "does it work" testing, but as a robot, it should be primarily controlled remotely.
The Roboclaw is a computer, so I intend to leverage it to replace both the DB-10 and the Propeller board and process the data to/from it on the Pi. I'm not sure what I will do about PING or other proximity sensors at this point. That is an open question.

Anyway, that shouldn't stop us from working together. The code isn't going to differ greatly, and the reality is that it may be man months before I actually have a second robot in operation anyway.

If you want me to help you customize the setup to allow you to shut down the SBC, etc. while docked, I am happy to. The biggest issue I see is how you will turn it on again when charged without having the SBC active, but if you can find a way that you are happy with, making the code work with it should be fun and doable.

---

Yes, I will try to get back to the ir code. That shouldn't be very difficult. The main difficulty is that I don't have the hardware, so I just have to let you test.

Yes, it is already possible to shut off the robot's response to PING/IR sensors, so it should be easy enough to automate that when in "docking mode". I will try to remember to do that.

TCIII, 2022-03-10:
@chrisl8,

Thanks for the quick response, much appreciated.

I will be glad to do any testing required as I am presently a Donkey Car Maintainer who does a lot of hardware/software validation.

Right now I would appreciate being able to have the ArloBot be commanded to dock from either the CLI or the webui so that I can experiment with a number of charging schemes.

Additionally I plan to build a smaller ArloBot on a 14 in diameter chassis using the Pololu D37 motors with encoders and 5.7 in diameter scooter wheels. 

The problem here is that a smaller robot chassis is not going to support two 12 vdc/7 amp-hr SLA batteries and I plan to move to Li-Ion batteries with charge protection circuits. I have used the Li-Ion batteries in ROVs with great success and a spotless safety record. But that is neither here nor there concerning which battery chemistry we employ.

Comments?

TCIII

TCIII, 2022-03-10:
@chrisl8, 

> I also don't hope to make the Pi run Rviz on a normal basis. It is nice for initial "does it work" testing, but as a robot, it should be primarily controlled remotely.

So you plan to use either a fast (I7) PC or Laptop  to [run rviz remotely on Windows](https://github.com/chrisl8/ArloBot/wiki/Running-Rviz-on-Windows) and offload that processing from the Rpi?

Regards,
TCIII

TCIII, 2022-03-11:
@chrisl8,

Do you have any ETA in mind as to the integration of the ir docking code into the ArloBot ROS package?

Regards,
TCIII

TCIII, 2022-03-13:
@chrisl8,

Bump?

I would really like to know about running the docker rviz as I posted above.

Regards,
TCIII

chrisl8, 2022-03-18:
I don't know what your question is about running rviz. You can run it anywhere. There are Windows instructions to run it on Windows. You can use the Docker container if you need to use a different version/distribution of Linux than ROS supports, and you can just run it normally on supported Ubuntu versions.

It runs fine on any modern computer and even pretty well on a Pi or other small ARM based computer.

You might double check your CPU usage if you are attempting to build a map and run RViz on the same Pi. It works, but map making can "degrade" if it is starved for CPU cycles. If it works though, then it works.

TCIII, 2022-03-18:
@chrisl8,

When I run docker rviz on the Nano 4GB is it running on the Nano or the Rpi that the Nano is communicating with which is running the ArloBot software?

By the way, how is the ir_Code.cpp code integration into the ArloBot software package coming along? I have completed all of the necessary ArloBot and Docking Station hardware and am just awaiting the software update to be able to perform docking maneuvers. 

TCIII

TCIII, 2022-03-22:
@chrisl8,

I have essentially completed the docking station and the ArloBot docking probe.

I have performed the following to complete the docking station/docking probe:

1) Installed a 10 amp 24 vdc switching power that can be adjusted to power the battery charging module and the ArloBot SBC/Accessories.
2) Installed a 1.5 amp 5 vdc power converter to power the IR transmitter and the adjustable timing circuit.
3) Installed a relay controlled by an adjustable timing circuit in order to prevent arcing when the docking probe pogo pins make contact with the docking station power plates.
4) Installed all six pogo pins on the ArloBot docking probe.

All that I require now is the integration of the ir_Code.cpp into the ArloBot sofware package so that the ArloBot can perform the auto docking function.

Regards,
TCIII

 
[image]
[image]
[image]

chrisl8, 2022-03-27:
@TCIII 

That is an impressive hardware setup. 

I am sitting down now to build an initial attempt at the integration code for you to test. I should have it by the end of the day.

chrisl8, 2022-03-27:
@TCIII 

I have pushed new code that includes the auto docking functions. I haven't really tested it, so I expect it to have problems. Please give it a try and report back.

First, you need to add this to your `~/.arlobot/personalDataForBehavior.json` file:
```
"dockingStation": {
    "_has_dockingStation": "Set this to true if you have a docking station",
    "has_dockingStation": true,
    "dockingIrReceiverUniqueString": "Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_019990C9",
    "dockingIrReceiverStringLocation": "ID_SERIAL",
    "angular_z": 0.1,
    "linear_x": -0.08
  },
```

Then run the setup again,
`bash <(wget -qO- --no-cache -o /dev/null https://raw.githubusercontent.com/chrisl8/ArloBot/noetic/setup-noetic.sh)`

Then start ROS and when it is going, run this script:
`auto-docking.sh`

TCIII, 2022-03-27:
@chrisl8,

Thanks for the update, much appreciated.

I will test the code first thing tomorrow.

I spent most of the day attempting to make a map using ArloBot. 

I found that ArloBot would veer off to the left when it should go straight when using 2D Nav Goal with Rviz.

So I ran `~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh` and used cut and try (instead of the "[Borenstein, J., & Feng, L. (1995). UMBmark: A benchmark test for measuring odometry errors in mobile robots"] method) to adjust the `wheelSymmetryError` and the `distancePerCount` until ArloBot would go straight for 1 meter. I converged on the correct values farily quickly though I found that adjustments in the `distancePerCount` affected the `wheelSymmetryError` and vice versa.
Here is what I found:
`wheelSymmetryError` IS: 1.0, S/B: 0.79

`distancePerCount` IS: 0.00338, S/B: 0.00350

In the `driveGeometry` tuning guide you state that the `wheelSymmetryError` "accounts for a difference in left/right wheel diameter." I have accurately measured both of the six inch diameter wheels and when inflated to the same pressure, they both are the same diameter. 
The only thing that I can think of that is causing ArloBot to veer off to the left of a straight line, when `wheelSymmetryError` is set to 1.0,  is that the right hand motor is turning faster than the left hand motor for the same drive voltage?

Comments?

Regards,
TCIII

chrisl8, 2022-03-28:
@TCIII Yes, wheelSymmetryError exists to compensate for the left and right wheels either not being of equal diameter or some other issue with the drive geometry.

The setting is the "distance per count" of the encoder, so voltage isn't a part of the issue. The wheel speed is read by the encoders, so the voltage output to wheel speed shouldn't matter, as the read wheel speed will dictate the final voltage for a given speed. Hence, the left/right side should only differ if the wheels are not the same diameter because of some abnormality like under or over inflation of one tire compared to the other, or one wheel just slipping badly and consistently.

Besides making sure it can go straight for a meter, also ensure that it can rotate 180, and 360 degrees both clockwise and counterclockwise.

Finally, you sad it would veer off to one side when under navigation control. Was the navigation clearly attempting to correct this, because it wasn't expected?
Or was this the navigation having strange ideas about the best way to get somewhere?
In other words, is this odometry failure or path planning failure?

TCIII, 2022-03-28:
@chrisl8,

Thanks for the response and detailed guidance.
 1) Both tires are inflated to the same pressure: 12 pounds atmospheric.
 2) There is no observable slip between the tires and driving surface.
 3) During the Rviz 2D Nav Goal the drawn path was initially straight ahead and ArloBot slowly veered to the left (began to turn 
     slowly left CCW) instead of moving straight ahead as per the drawn path.
 4) The slow drift to the left was observed when running ArloBot in a straight line for one meter using the `maneuver` function 
     in `~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh` with the default values for `wheelSymmetryError` and the 
     `distancePerCount`. Changing to the `wheelSymmetryError` and the `distancePerCount` values I published in my post above corrected the slow drift to the left and kept ArloBot moving straight ahead for one meter.  

When using the default `wheelSymmetryError` and the `distancePerCount` values ArloBot overshot the commanded angle of rotation. 
When I use my `wheelSymmetryError` and the `distancePerCount` values the ArloBot rotation to the commanded angle is virtually spot on.

I will have to make a disclaimer here: My Arlo robot chassis uses the Eddie chassis and the Eddie motor/ wheels, but is using the Arlo encoder and wheel disks. The distance between the center of the Eddie chassis wheels is within a 0.01 meter of the default `trackWidth` so I have left it at the default value of 0.403 meter.

How about if we move this issue to a new issue so that it will not preempt the docking station development enhancement issue? 

Comments?

Regards,
TCIII

chrisl8, 2022-03-28:
@TCIII  Yes, we should move this somewhere else. Also, it isn't really a code issue, just having to tweak your variable settings for your particular robot. The wheelSymmetryError is doing this in the code:

https://github.com/chrisl8/ArloBot/blob/458754b83dc41449e768d11a4682ed2e972d264a/PropellerCodeForArloBot/ROSInterfaceForArloBot/ROSInterfaceForArloBot.c#L542
```
        expectedRightSpeed =
            expectedRightSpeed / distancePerCount * wheelSymmetryError;
```

So it literally causes the robot to adjust how many "ticks" in the encoder per rotation are used between the right and left wheels.

I have never had to use it, but it exists for exactly the purpose you are using it for, so if it works, that is all that matters. Just keep in mind that you set this when you are working with the robot later.

TCIII, 2022-03-28:
@chrisl8,

You are right that it is not a code issue, but I as surprised at how far I had to deviate from the default `wheelSymmetryError` value.
I suspect that there may be a mechanical problem with the right Motor/Wheel assembly that is causing ArloBot to deviate from a straight path when using the default `wheelSymmetryError` value. Though the gear lash between the right and left Motor/Wheel assemblies seem to be the same.

Regards,
TCIII

TCIII, 2022-03-28:
@chrisl8

"dockingStation": {
    "_has_dockingStation": "Set this to true if you have a docking station",
    "has_dockingStation": true,
    "dockingIrReceiverUniqueString": "Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_019990C9",
    "dockingIrReceiverStringLocation": "ID_SERIAL",
    "angular_z": 0.1,
    "linear_x": -0.08
  },

I changed "linear_x": -0.08 to "linear_x": 0.08 otherwise ArloBot will backup once it has detected the IR Transmitter beam.

TCIII

[83 more replies]

closed 2023-09-15 by chrisl8