chrisl8/ArloBot #204: ArloBot will not drive in a straight line when using the default driveGeometry values

TCIII, 2022-03-28:
@chrisl8,

I found that ArloBot would veer off to the left when it should go straight when using 2D Nav Goal with Rviz.

So I ran `~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh` and used cut and try (instead of the "[Borenstein, J., & Feng, L. (1995). UMBmark: A benchmark test for measuring odometry errors in mobile robots"] method) to adjust the `wheelSymmetryError` and the `distancePerCount` until ArloBot would go straight for 1 meter. I converged on the correct values farily quickly though I found that adjustments in the `distancePerCount` affected the `wheelSymmetryError` and vice versa.
Here is what I found:
`wheelSymmetryError` IS: 1.0, S/B: 0.79

`distancePerCount` IS: 0.00338, S/B: 0.00350

In the `driveGeometry` tuning guide you state that the `wheelSymmetryError` "accounts for a difference in left/right wheel diameter." I have accurately measured both of the six inch diameter wheels and when inflated to the same pressure, they both are the same diameter. 
The only thing that I can think of that is causing ArloBot to veer off to the left of a straight line, when `wheelSymmetryError` is set to 1.0,  is that the right hand motor is turning faster than the left hand motor for the same drive voltage?

Comments?

Regards,
TCIII

TCIII, 2022-03-28:
https://github.com/chrisl8/ArloBot/issues/198#issuecomment-1080167112

TCIII, 2022-03-28:
https://github.com/chrisl8/ArloBot/issues/198#issuecomment-1080850767

TCIII, 2022-03-28:
@chrisl8,

Are you using the Arlo Motor/Wheel Assemblies or the Eddie Motor Wheel Assemblies?

Mine are the original Eddie Motor/Wheel Assemblies.

Parallax Tech Supports says that the  Arlo Motor/Wheel Assemblies can fit on the Eddie Robot Base and have significantly less gear train backlash which would be a real plus.

Comments?

TCIII

chrisl8, 2022-03-28:
I am using the Arlo Motor/Wheel Assemblies.

TCIII, 2022-04-03:
@chrisl8,

I purchased one of the remaining pairs of the Arlo Motor/Wheel assembly and it arrived on Saturday.

I quickly assembled the Motor/Wheel Assemblies in the morning and installed them on my ArloBot, which was equipped with the original Eddie Motor/Wheel Assemblies, but had been converted to the ArloBot Encoders, this afternoon which was not mean trick especially removing the battery plate with the batteries still on it.

After reassembling ArloBot I used the `~/catkin_ws/src/ArloBot/scripts/PropellerSerialTest.sh` maneuver function to test the Arlo Motor/Wheel assemblies after I had set the `distancePerCount:` and `wheelSymmetryError:` values back to their default values.

The Arlo Motor Wheel Assemblies performed to expectations without the startup clunk of the Eddied Motor/Wheel Assemblies due to the 1/2 inch of wheel circumference slop inherent in the Eddie gear train. They are very quite and were almost spot on when performing the 1 meter travel test.

Definitely a worthwhile upgrade to my ArloBot chassis irrespective of the cost.

Now to try out making a map.

TCIII

TCIII, 2022-04-06:
@chrisl8,

I am closing this issue since I was able to resolve the maneuvering issue by replacing the original Eddie Motor/Wheel assemblies with the Arlo Motor/Wheel assemblies.

Regards,
TCIII

closed 2022-04-06 by TCIII