chrisl8/ArloBot #207: /dev/fd/63: line 693: npm: command not found

akycdi, 2022-08-09:
I am getting this error while installing the ros on ubuntu in VM.
I have installed  the ros noetic version in the ubuntu 
[image]

akycdi, 2022-08-09:
presented with another error

chrisl8, 2022-08-16:
Did you get it sorted?

akycdi, 2022-08-22:
Yes, I have sorted but unfortunately i am unable to run the arlo on the both ubuntu 20.04 and 18lts versions 
I am getting the same error i have changed the noetic and the  melodic visa versa
Sorry for the late reply and can you help me with this error.

i am just trying to run the script.
[image]

chrisl8, 2022-09-12:
It seems to be having trouble finding `simpleide` which doesn't surprise me as that is basically deprecated by the creator. The most recent version uses another method of installing. In fact `setup-noetic.sh` doesn't ever try to run simpleide.

Can you try `setup-noetic.sh` again and let me know if it is still giving you errors?

chrisl8, 2022-12-26:
Closing due to lack of response for a while. If you are still having trouble feel free to reopen this issue.

closed 2022-12-26 by chrisl8

akycdi, 2023-01-05:
Yes, Thanks I have completed the project with the SimpleIDE . I haven't been on Git that much and I don't know how the notification got missed. Thanks again for the reply.